platformio/platform-atmelavr #115: Can't set board_vendor (USB manufacturer) in platformio.ini

lalten, 2018-11-08:
I want to override the `USB_PRODUCT` and `USB_MANUFACTURER` strings on my Sparkfun Pro Micro.

According to the docs (http://docs.platformio.org/en/latest/boards/atmelavr/sparkfun_promicro16.html#configuration), I can

> override default SparkFun Pro Micro 5V/16MHz settings per build environment using `board_***` option

The Pro Micro's [board settings json](https://github.com/platformio/platform-atmelavr/blob/master/boards/sparkfun_promicro16.json) have the `build.usb_product` and `vendor` fields (defaults are `SparkFun Pro Micro` and `SparkFun`).

I can override the `USB_PRODUCT` with this `platformio.ini`:
```
[env:sparkfun_promicro16]
platform = atmelavr
board = sparkfun_promicro16
framework = arduino
board_build.usb_product = "MyProduct"
board_vendor = "MyCompany"
```

But as `dmesg` shows, only `USB_PRODUCT` is applied, not `USB_MANUFACTURER`:
```
[119019.190230] usb 1-1: new full-speed USB device number 115 using xhci_hcd
[119019.344882] usb 1-1: New USB device found, idVendor=1b4f, idProduct=9203, bcdDevice= 1.00
[119019.344888] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[119019.344892] usb 1-1: Product: MyProduct
[119019.344896] usb 1-1: Manufacturer: SparkFun
[119019.346978] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```

The relevant code is at https://github.com/platformio/platform-atmelavr/blob/master/builder/frameworks/arduino.py#L40. It looks like the `build_vendor` setting should work?

What am I missing?

werty1st, 2023-04-18:
I found a solution: 
[https://stackoverflow.com/questions/53278298/cant-override-board-vendor-usb-manufacturer-via-platformio-ini/76049354#76049354](https://stackoverflow.com/questions/53278298/cant-override-board-vendor-usb-manufacturer-via-platformio-ini/76049354#76049354)

**platformio.ini**
```ini
[env:adafruit_neotrinkey_m0]
platform = atmelsam
board = adafruit_neotrinkey_m0
framework = arduino
extra_scripts = pre:setUSB.py
```
**setUSB.py**
```py
Import("env")

board_config = env.BoardConfig()
board_config.update("build.hwids", [
  ["0x27B8", "0x01ED"]
])
board_config.update("build.usb_product", "NeoPixel Trinkey M0-XXX")
board_config.update("vendor", "Adafruit-X")
```

**result:**
```bash
system_profiler SPUSBDataType

USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT6000USBXHCI

        NeoPixel Trinkey M0-XXX:

          Product ID: 0x01ed
          Vendor ID: 0x27b8
          Version: 1.00
          Serial Number:
          Speed: Up to 12 Mb/s
          Manufacturer: Adafruit-X
          Location ID: 0x02100000 / 1
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
```

rileyjshaw, 2023-08-02:
I agree with @lalten that it should be possible to override this from `platformio.ini`. From [the docs](https://docs.platformio.org/en/stable/projectconf/sections/env/options/platform/index.html#more-options):

> You can override any board option declared in manifest file using the next format board_{OBJECT.PATH}, where {OBJECT.PATH} is an object path in JSON manifest. Please navigate to “boards” folder of [PlatfomIO development platforms](https://github.com/topics/platformio-platform) and open JSON file to list all available options.

As noted above, the correct defines appear in `.vscode/c_cpp_properties.json`, but `USB_MANUFACTURER` is not applied on upload.