platformio/platform-atmelavr #117: Can't find atmel-toolchain in alpine docker container

TimVosch, 2018-11-20:
What kind of issue is this?

Possibly(?)

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Arch Linux x64

**PlatformIO Version** (`platformio --version`): 3.6.1

### Description of problem
I am running a docker container for my Continuous Integration, which I install platformio on. However when I try to build any project, PlatformIO reports that it can't find *avr-gcc* / *avr-g++*.

#### Steps to Reproduce

1. Create docker image from: https://pastebin.com/pNWXjSgb
2. Run docker container with name `jenkins-blueocean`
3. Shell in (or create a pipeline but this is easier):  `sudo docker exec -i -t jenkins-blueocean /bin/bash`
4. Create a project: ex. `pio init -b uno`
5. Insert something in main.cpp (see basic source below)
6. `pio run`

### Actual Results
https://pastebin.com/SqRkyrR5

### Expected Results
Succesful build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

**Source file to reproduce issue:**
```cpp
int main() {

}

void loop() {

}
```

### Additional info
I also tried different projects, non are working.

ivankravets, 2018-11-20:
Yes, it seems that existing AVR toolchain for ARM is not compatible with Alpine distributive. It should work with Ubuntu/Debian. Temporary solution is to remove everything from `~/.platformio/packages/toolchain-atmelavr` but keep `package.json`. PlatformIO will use the next binaries https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L93

Another solution is to fork https://github.com/platformio/platform-atmelavr and remove "toolchain" dependency.

Also, we have a feature request https://github.com/platformio/platformio-core/issues/1367 which will allow overriding platform dependencies from `platformio.ini` in PlatformIO Core 4.0

TimVosch, 2018-11-22:
I got it working (atleast the building) with the build_flags below. I don't think each is necessary, but it works :man_shrugging: Note that `/var/jenkins_home/` is usually your own home directory, however this is a jenkins docker container.

```
build_flags =
        -B/usr/lib/gcc/avr/6.1.0/
        -B/var/jenkins_home/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/5.4.0/
        -I/var/jenkins_home/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/5.4.0/include
        -I/var/jenkins_home/.platformio/packages/toolchain-atmelavr/avr/include/
        -I/var/jenkins_home/.platformio/packages/toolchain-atmelavr/libexec/gcc/avr/5.4.0/
```

I will try and create a new image which will install the alpine atmel-toolchain and see if I can use some symlink magic to make it work.