platformio/platform-atmelavr #120: Programming with non serial prgrammer broken

koppa, 2018-12-10:
Version: platform-atmelavr stable and master.

When using a avr programmer without serial interface like the dragon_isp, the following lines fail in `builder/main.py`:

```
        if not upload_options.get("disable_flushing", False) \
            and not env.get("UPLOAD_PORT", "").startswith("net:"):
                pass
            env.FlushSerialBuffer("$UPLOAD_PORT")
```

`FlushSerialBuffer` will be called and fails, due to `upload_port=usb` (in the project config):

```*** [upload] could not open port usb: [Errno 2] No such file or directory: 'usb'```

When disabling the FlushSerialBuffer for usb devices, everything works correctly.

ivankravets, 2018-12-20:
Do you use `pio run -t program` instead of `pio run -t upload`?  See https://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

closed 2018-12-20 by ivankravets

koppa, 2018-12-21:
No I used `pio run -t upload`.

The problem is the code for flushing the serial port is executed, and for user it is not obvious, that this part fails.  In my opinion flushing should only be done for programmers, where the build script is sure, that this is a serial device (maybe only do it explicitly?)

ivankravets, 2018-12-21:
Please use `pio run -t program`.