platformio/platform-atmelavr #127: Use micronucleus uploader for ATTiny

wschaeffer, 2019-01-13:
The use of micronucleus uploader is exclusive for digispark boards, see platform.py:
`disabled_tool = "tool-micronucleus"`

Therefore it is not possible to use the micronucleus uploader when using a regular ATTiny with the micronucleus bootloader.

When using:
`[env:attiny85]`
`platform = atmelavr`
`board = attiny85`
`framework = arduino`
`upload_protocol = micronucleus`
An error appears: 
> 'micronucleus' is not recognized as an internal or external command, operable program or batch file.

A solution might be to set 'board = digispark-tiny' but this prohibits the use of standard atmel libraries, like SPI.

When setting `disabled_tool = ""` everything seems to work fine. Is this a bug or an unwanted feature?

ivankravets, 2019-01-16:
Thanks! Resolved in upstream version https://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

Please re-test.

closed 2019-01-16 by ivankravets

wschaeffer, 2019-01-16:
It works now, thanks!

ivankravets, 2019-01-16:
Thanks! The final release is out. You can switch back to stable version.

commit f5a3138a (2019-01-16): Allow to use "micronucleus" tool/protocol for other boards // Resolve #127