platformio/platform-atmelavr #132: Invalid Board JSON for Sparkfun Pro Micro 5V

R0ll1ngSt0ne, 2019-02-21:
Sparkfun Pro Micro 5V board claims PIDs 0x9205 and 0x9206.

The board JSON included with platformio (sparkfun_promicro16.json) lists incorrect PIDs:

    "hwids": [
      [
        "0x1B4F",
        "**0x9203**"
      ],
      [
        "0x1B4F",
        "**0x9204**"
      ],

The above PIDs are for 3.3V version of the board, not 5V.

R0ll1ngSt0ne, 2019-05-13:
You are welcome!

R0ll1ngSt0ne, 2019-05-13:
I think there is still a problem with the JSON:

The PIDs claimed by Pro Micro 5V are PID: 0x9205 (**bootloader**); 0x9206 (**sketch**)

However, JSON now lists the PIDs in this order:

>     "hwids": [
>       [
>         "0x1B4F",
>         "0x9205"
>       ],
>       [
>         "0x1B4F",
>         "0x9206"
>       ],

It seems that when writing a sketch to the board, platformio writes the _first PID_ to to the board - causing the board to use  bootloader PID for sketch. I think the order of pids in the JSON should be reversed.

Thanks,

Michael

R0ll1ngSt0ne, 2019-05-13:
Is there a way to specify in the JSON which if HWIDs are for bootloader and which are for a sketch?

ivankravets, 2019-05-14:
Hm... But we don't support flashing of bootloader. So, we can use HWID/PID for sketch?

ivankravets, 2019-05-14:
Please re-test with the upstream version  http://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

commit b0a284c6 (2019-04-23): Fix HWIDs for SparkFun Pro Micro 5V/16MHz // Resolve #132
commit c4270b15 (2019-05-14): Swap PID/VID for SparkFun Pro Micro 5V/16MHz // Resolve #132
closed 2019-05-14 by ivankravets