platformio/platform-atmelavr #134: Error: Could not find a version that satisfies the requirement '~1.50400.0' for your system 'freebsd_amd64'

closedcube, 2019-03-10:
Hello,

Just updated a build server on FreeBSD 11.2 and got the following error:

$ pio upgrade
You're up-to-date!
PlatformIO 3.6.5 is currently the newest version available.

$ pio run
Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-atmelavr @ ~1.50400.0
Error: Could not find a version that satisfies the requirement '~1.50400.0' for your system 'freebsd_amd64'
--

Any chance to fix above?

Thanks

stonehippo, 2019-03-17:
@closedcube this means that the package toolchain-atmelavr for FreeBSD hasn’t been updated on Bintray (which is where PIO keeps its packages). Looking at the files page in Bintray, you can see that the last available packages was 1.40902.0 (
https://bintray.com/platformio/dl-packages/toolchain-atmelavr#files)

You could work around this by downgrading the package in your `~/.platformio/platforms/atmelavr/package.json`, or you can see if @ivankravets can get a newer toolchain into Bintray (I’ve been trying to understand how the packages are managed there, too).

ivankravets, 2019-03-20:
We upgrade GCC toolchain in https://github.com/platformio/platform-atmelavr/releases/tag/v1.12.0

So, you can use the previous version of dev/platform via `platform = atmelavr@<1.12.0` in `platformio.ini`.

Need to go into logs and check why we switched to GCC 5.4.0.

closedcube, 2019-03-20:
Many thanks for the update, just changed a release environment to the previous version.

Please let us know if you need any additional info.

trombik, 2019-10-11:
@ivankravets please consider to document the build and upload process to bintray so that users can build the tools. recent FreeBSD does not have `libelf.so.1`, which is required by `avrdude` provided by platformio. as such, even the old one does not work.

a workaround for users: install `devel/avrdude` from the ports or install the package. replace the binary from platformio with it.

```
pkg install devel/avrdude
cp /usr/local/bin/avrdude ~/.platformio/packages/tool-avrdude/avrdude_bin
```

trombik, 2020-06-19:
here is my solution: https://github.com/trombik/platformio-freebsd-toolchain-atmelavr