platformio/platform-atmelavr #136: Fuses (Part 3)

gcormier, 2019-04-04:
I realize there has been quite a bit of discussion in #128 as well as https://github.com/platformio/platformio-core/issues/865

I think simple support for fuses should be added in platformio.ini.

Changing a boards file for this should not be necessary. Fuses can be seen as parameters for the AVR, and many different boards can support quite a few configurations. The Arduino IDE hides most of this from beginners, which is probably good, but it doesn't take long to out-grow this.

For instance, a fuse can be set to change Brown Out Detection behaviour. Even boards such as the UNO could have different fuse values depending on the use of the board.

Another option would be to prevent any further flashing of the chip. Again, this is regardless of any board or variant of a platform - perhaps a user wants to do this for some reason. Defining a new board doesn't really make sense.

To share code that might need specific fuses set, it would be beneficial to have this contained in the platformio.ini for distribution, rather then having an extra_scripts defined to flash the fuses.

I am new to PIO, and perhaps the discussion is closed - if so, forgive me. I am otherwise really impressed at the whole suite, this was the only thing that stuck out as odd to me so I figured it could be an improvement. :)

ivankravets, 2019-04-23:
Have you seen this http://docs.platformio.org/en/latest/platforms/atmelavr.html#fuses?

Which board does not have fuses?

gcormier, 2019-04-24:
I have seen it, and its the solution I am using for now.

However, it seems like an extra step that could be avoided by just allowing a few overrides in the configuration file. :)

ivankravets, 2019-04-24:
See example https://github.com/platformio/platform-atmelavr/blob/develop/boards/uno.json#L31

The board manifests are here https://github.com/platformio/platform-atmelavr/tree/develop/boards

Please provide PR.

gcormier, 2020-03-04:
I believe this is now handled. 

https://docs.platformio.org/en/latest/platforms/atmelavr.html#custom-fuses

closed 2020-03-04 by gcormier