platformio/platform-atmelavr #137: MCUdude MightyCore boards package

MCUdude, 2019-04-14:
Hi!

As the maintainer of [MightyCore](https://github.com/MCUdude/MightyCore), I would like to provide better documentation on how to use PlatformIO with the Arduino core. If this turns out to be a success I'd like to add some of my other Arduino cores as well. I know fuse settings is difficult, so I will leave this out of the question for now.

First, the MightyCore version that's included here is very old and outdated. This will have to be updated to the latest version and preferably be automatically update whenever I push a commit or do a new release. How can this be done?

MightyCore has the additional option of turning on of off LTO (link time optimization). How can this feature be turned on or off in the platformio.ini file?

MightyCore contains three different [pinouts or variants](https://github.com/MCUdude/MightyCore/tree/master/avr/variants). Standard, Bobuino, and Sanguino. How can one of these pinouts be selected using the platformio.ini file?

If possible, I'd also like to re-structure the JSON file(s) a bit to make the platformio.ini file a little sleeker. Is possible to make one single JSON file for all MightyCore devices? Then all other settings will have to be provided in the ini file.

Is it also possible that some sort of platformio.ini template is loaded into the project after you've selected *MightyCore* as your board? If so I could add some additional settings (shown below).

Ideally, I'd like a working platformio.ini file to look like this:
``` ini
[env:MightyCore]
platform = atmelavr
framework = arduino

; Device
board = mightycore1284 ; Do we have to specify board here, or is MCU enough?
board_build.mcu = atmega1284p

; Build options
board_build.f_cpu = 160000000L
;board_build.lto = true
;board_build.pinout = standard ; (or bobuino or sanguino)

; Upload options
board_upload.speed = 115200
upload_port = /dev/cu.usbserial*
```

MCUdude, 2019-04-20:
@ivankravets is there any way I could contribute, so any of this could happen?

ivankravets, 2019-04-22:
Sorry, I missed this issue :( Sure, we would be thankful in any contribution.

Ideally, would be good if we have a separate framework package per each Arduino AVR core. However, to simplify the integration process, we created this repository https://github.com/platformio/platformio-pkg-framework-arduinoavr
So, you can actually make a PR and I'll merge and re-deploy to our CDN.

* Entry point of dev/platform builder => https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py
* Arduino core builder => https://github.com/platformio/platform-atmelavr/blob/develop/builder/frameworks/arduino.py

These files are located in your machine by `%HOME_DIR%/.platformio/platforms/atmelavr/***`. You can edit them on a local machine for testing purpose.

> MightyCore has the additional option of turning on of off LTO (link time optimization). How can this feature be turned on or off in the platformio.ini file?

See example:

* http://docs.platformio.org/en/latest/platforms/teensy.html#optimization
* https://github.com/platformio/platform-teensy/blob/develop/builder/frameworks/arduino.py#L169

> If possible, I'd also like to re-structure the JSON file(s) a bit to make the platformio.ini file a little sleeker.

Each known/generic board has own JSON manifest. If you need some extra configuration settings, you can extend it. See https://github.com/platformio/platform-atmelavr

> ;board_build.pinout = standard ; (or bobuino or sanguino)

```
[env:myenv]
...
board_build. variant = ...
```

If you have any questions, I would be glad to help.

MCUdude, 2019-04-22:
About updating the corefiles, How is it done? The version hosted over at platformio/platformio-pkg-framework-arduinoavr is very old, but the version present on my computer is much newer.
Will you guys take care of the updating, or do I have to submit a PR?

>;board_build.pinout = standard ; (or bobuino or sanguino)
>[env:myenv]
>...
>board_build. variant = ...

I tried this but no other variant than `standard` is preset at the moment. I'm not really sure where the "new" pinout files go. It would be awesome if you could add these files and update the corefiles. I promise to do the testing!

As for the LTO option, how about a generic LTO option for AVR targets such as `AVR_LTO`

``` ini
build_flags = -D AVR_LTO
```

It should cover the needs for most users.

I'm not really sure how it should be implemented though. [Here's how it's done in the boards.txt file.](https://github.com/MCUdude/MightyCore/blob/21c2af8ad7972c2a626bf6f539e9e2c35beccf5a/avr/boards.txt#L96-L107)

ivankravets, 2019-04-23:
See https://github.com/platformio/platformio-pkg-framework-arduinoavr/commit/bc854ca65e6faac834ca0062c9886cb124cab683

Could you re-test with the upstream version? See https://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

> build_flags = -D AVR_LTO

In this case, no need to reinvent wheels. This is actually an issue of Arduino IDE builder which is so primitive and does not allow to configure the build process. I see that LTO is enabled by default. If developers don't need LTO, they can remove flags 
 
```ini
[env:myenv]
...
build_unflags = -flto
```

> I tried this but no other variant than standard is preset at the moment. 

I've just added 3 variants:

* https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/variants/mightycore
* https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/variants/sanguino
* https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/variants/bobuino

Does it work for you?

MCUdude, 2019-04-23:
> Could you re-test with the upstream version? 

I'll do it right away!

> In this case, no need to reinvent wheels. This is actually an issue of Arduino IDE builder which is so primitive and does not allow to configure the build process. I see that LTO is enabled by default. If developers don't need LTO, they can remove flags

That's a pretty easy way of disabling LTO. I'll add it to my personal platformIO template 👍 

> I've just added 3 variants:
> ...
> Does it work for you?

Maybe we should add mightycore as a prefix? as in mightycore_standard, mightycore_bobuino and mightycore_sanguino? IMO this will be easier to maintain when my other AVR Arduino cores get added later.

ivankravets, 2019-04-23:
Done! Please re-run `pio update`.

https://github.com/platformio/platformio-pkg-framework-arduinoavr/commit/d5e84b37846d7e3386858e2f6a8182200e588e12

MCUdude, 2019-04-23:
Cool! Quick question. I installed PlatformIO through VS Code, and everything is working fine there. On the other hand my terminal app keeps telling me that the pio/platformio command isn't found.

Do you recommend installing platformIO _in_ my terminal too, or will the these two separate installs collide?

ivankravets, 2019-04-23:
Just click on "Terminal" icon on the bottom toolbar or from PlatformIO left sidebar. It will populate PATH with PlatformIO Core.

MCUdude, 2019-04-23:
Well. I ran `pio update`, and updated to the latest version (3.6.7). However it still hasn't fetched your latest commit:

```
> Executing task in folder Testproject_2019: platformio run <

Processing MightyCore (framework: arduino; platform: https://github.com/platformio/platform-atmelavr.git; board: mightycore32)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/mightycore32.html
PLATFORM: Atmel AVR > MightyCore ATmega32
HARDWARE: ATMEGA32 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 12 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/MightyCore/src/main.cpp.o
Archiving .pioenvs/MightyCore/libFrameworkArduinoVariant.a
Indexing .pioenvs/MightyCore/libFrameworkArduinoVariant.a
In file included from src/main.cpp:1:0:
/Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/Arduino.h:316:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
Compiling .pioenvs/MightyCore/FrameworkArduino/CDC.cpp.o
Compiling .pioenvs/MightyCore/FrameworkArduino/HardwareSerial.cpp.o
*** [.pioenvs/MightyCore/src/main.cpp.o] Error 1
In file included from /Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/USBAPI.h:33:0,
from /Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/CDC.cpp:19:
/Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/Arduino.h:316:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pioenvs/MightyCore/FrameworkArduino/CDC.cpp.o] Error 1
In file included from /Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/HardwareSerial.cpp:29:0:
/Users/hans/.platformio/packages/framework-arduinoavr/cores/MightyCore/Arduino.h:316:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pioenvs/MightyCore/FrameworkArduino/HardwareSerial.cpp.o] Error 1
=========================================================== [ERROR] Took 0.86 seconds ===========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

My current platformio.ini file looks like this
``` ini
[env:MightyCore]
platform = https://github.com/platformio/platform-atmelavr.git
board = mightycore32
framework = arduino
board_build.mcu = atmega32
board_build.f_cpu = 16000000L
;board_upload.maximum_size = 32768
board_upload.speed = 115200
board_build.variant = mightycore_sanguino
upload_port = /dev/cu.usbserial*

;build_unflags = -flto
```

Is there anything I'll have to do in order to "re-fetch"?

MCUdude, 2019-04-23:
Sorry, I actually ran `pio upgrade` instead of `pio update`. It's working fine now. I'll test some more and report back later. BTW what's up with the boards selection menu? ATmega1284 appears four times! Oh, and are you able to change my "signature" from Mcudude to MCUdude 😉?

[image]

ivankravets, 2019-04-23:
> ATmega1284 appears four times!

It seems that you have 4 dev/platforms of AtmelAVR with different Git revision. Please navigate to PIO Home > Platforms and remove duplicate.

>  "signature" from Mcudude to MCUdude 

Changed ;)

MCUdude, 2019-04-23:
After some more testing, I can confirm that everything seems to be working just fine. When this be available through `platform = atmelavr`? I'm currently writing a simple guide for MightyCore users, and I want to refer to the generic atmelavr platform rather than a specific git repo.

ivankravets, 2019-04-23:
I've just released a new version => https://github.com/platformio/platform-atmelavr/releases/tag/v1.13.0

Please uninstall all Git dev/platforms and use stable.

>  I'm currently writing a simple guide for MightyCore users

Great! 👍 We would be thankful for this guide. You can refer developers to this configuration file http://docs.platformio.org/en/latest/projectconf.html

closed 2019-04-23 by ivankravets

MCUdude, 2019-04-24:
Sorry for picking up the thread again, but avrdude.conf is missing some targets.
I'm not sure where this file is located, but could you add the following lines to avrdude.conf?
Even though some of these AVRs have the same signature it's still nice to be able to use the actual part number for `board_build.mcu`.

We're currently missing ATmega164A, ATmega164PA, ATmega324A, ATmega324PB, ATmega644A and ATmega644PA

```
#------------------------------------------------------------
# ATmega164A
#------------------------------------------------------------

# similar to ATmega164P

part parent "m164p"
    id               = "m164a";
    desc             = "ATmega164A";
    signature        = 0x1e 0x94 0x0f;

    ocdrev              = 3;
  ;

#------------------------------------------------------------
# ATmega164PA
#------------------------------------------------------------

# similar to ATmega164P

part parent "m164p"
    id               = "m164pa";
    desc             = "ATmega164PA";
    signature        = 0x1e 0x94 0x0a;

    ocdrev              = 3;
  ;

#------------------------------------------------------------
# ATmega324A
#------------------------------------------------------------

# similar to ATmega324P

part parent "m324p"
    id               = "m324a";
    desc             = "ATmega324A";
    signature        = 0x1e 0x95 0x15;

    ocdrev              = 3;
  ;

#------------------------------------------------------------
# ATmega324PB
#------------------------------------------------------------

# similar to ATmega324P
  
part parent "m324p"
    id               = "m324pb";
    desc             = "ATmega324PB";
    signature        = 0x1e 0x95 0x17;

    ocdrev              = 3;

    memory "efuse"
        size = 1;
        min_write_delay = 4500;
        max_write_delay = 4500;
        read = "0 1 0 1 0 0 0 0 0 0 0 0 1 0 0 0",
               "x x x x x x x x o o o o o o o o";

        write = "1 0 1 0 1 1 0 0 1 0 1 0 0 1 0 0",
                "x x x x x x x x x x x x i i i i";
    ;
;

#------------------------------------------------------------
# ATmega644A
#------------------------------------------------------------

# similar to ATmega164p

part parent "m644"
    id               = "m644a";
    desc             = "ATmega644A";
    signature        = 0x1e 0x96 0x09;

    ocdrev              = 3;
  ;

#------------------------------------------------------------
# ATmega644P
#------------------------------------------------------------

# similar to ATmega164p

part parent "m644"
    id               = "m644p";
    desc             = "ATmega644P";
    signature        = 0x1e 0x96 0x0a;

    ocdrev              = 3;
  ;

```

ivankravets, 2019-04-24:
Thanks! Please re-run `pio update` :)

MCUdude, 2019-04-24:
Cool! I may start planning for [MiniCore](https://github.com/MCUdude/MiniCore) support then! Many users have requested MiniCore PlatformIO support 😉

ivankravets, 2019-04-24:
Great! Please ping me if you need any help from our side 😊

MCUdude, 2019-04-24:
Great! I'll probably open a new issue here when the questions start to pile up 🙂