platformio/platform-atmelavr #138: PlatformIO MiniCore support

MCUdude, 2019-04-25:
Hi!

I'm starting to look into PlatformIO MiniCore support.
@ivankravets would you prefer if I issued some PRs, or do you want to add it yourself?

* First I need to add all the respective minicoreXXX.json files [here](https://github.com/platformio/platform-atmelavr/tree/develop/boards)
* Then I need to add `boards_minicore.txt` [here](https://github.com/platformio/platformio-pkg-framework-arduinoavr)
* Add `minicore_standard` pinout variant [here](https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/variants)
* Add MiniCore libraries [here](https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/libraries/__cores__)
* And then add the MiniCore core files [here](https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/cores)

Did I miss anything?

ivankravets, 2019-04-25:
We are going to refactor all our bord list and split them into:

- Dev KITs (class boards)
- Devices

We need that to allow developers to start their projects from DEVICES if they have a custom board and would like to use bare programming or SDK.

So, instead of duplicating boards by Arduino cores it is better to have 1 DEVICE and explain in PlatformIO docs for Atmel AVR dev/platform how to switch between Arduino cores and variants.

For example, a few devices:

* `ATMEGA168.json`
* `ATTINY88.json`
* ...

 What do you think?

MCUdude, 2019-04-25:
> So, instead of duplicating boards by Arduino cores it is better to have 1 DEVICE and explain in PlatformIO docs for Atmel AVR dev/platform how to switch between Arduino cores and variants.

This is a very good idea! My Arduino cores follow this exact philosophy and are chip oriented instead of board oriented. The user has a specific microcontroller and wants to use the Arduino framework to develop an application. Clock speed, Bootloaders, upload port, upload speed, brown-out detection, LTO (and so on) is all up to the user.

But even though the user has selected an ATmega328P as the target, he/she may want to program it as an Arduino UNO (locked to 16 MHz external oscillator), use it barebone (avrlibc only) or program it through MiniCore. One benefit with MiniCore is that if you use the internal oscillator you get two extra IO pins. How can the user choose what framework is used?

MCUdude, 2019-05-02:
@ivankravets any hints?

ivankravets, 2019-05-07:
Sorry for the delay, I had a small vacation.

Can you make PR with generic board manifests as described above? `ATMEGA168PA.json`?

MCUdude, 2019-05-07:
Sorry, I'm heading to Shenzhen, China for a week on a business trip. I can do it as soon as I get back. I'm not even sure Github is available in chine without the use of VPN.

Do you want _all_ variants to have a dedicated JSON file or just the ones that actually have different device signatures? ATmega168P and ATmega168PA have the same signature. ATmega168/A and ATmega168P/PA does not.

I just did a quick edit of an existing file. Is this how you want the files to look like? If not, please provide an example, and I'll make a PR as soon as I can.

```` JSON
{
  "build": {
    "core": "MiniCore",
    "extra_flags": "-DARDUINO_AVR_ATmega168",
    "f_cpu": "16000000L",
    "mcu": "atmega164p",
    "variant": "miniycore_standard"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "MiniCore ATmega168P/PA",
  "upload": {
    "maximum_ram_size": 1024,
    "maximum_size": 15872,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 115200
  },
  "url": "https://github.com/MCUdude/MiniCore",
  "vendor": "MCUdude"
}
```

ivankravets, 2019-05-14:
How about this `ATMEGA168P.json`?

```json
{
  "version": 2,
  "device": {
    "mcu": "ATmega168P",
    "flash": 16384, 
    "ram": 1024,
    "eeprom": 512,
    "f_cpu": 20000000
  },
  "build": {
    "frameworks": {
      "arduino": {
        "core": "MiniCore",
        "extra_flags": "-DARDUINO_AVR_ATMEGA168P",
        "f_cpu": "16000000L",
        "variant": "miniycore_standard"
      }      
    },
    "extr_flags": "... common flags for all frameworks"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATMEGA168P",
  "upload": {
    "maximum_ram_size": 1024,
    "maximum_size": 15872,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 115200
  },
  "url": "https://www.microchip.com/wwwproducts/en/ATmega168P",
  "vendor": "Microchip"
}
```

MCUdude, 2019-05-20:
Looks good to me, but I have a few questions.

First, what does "version": 2 means?
Let's say I want to change F_CPU. Would I still add `board_build.f_cpu = 8000000L` to platformio.ini?

This is the template platformio.ini file I use for my projects. What would it look like with the new JSON file?

``` ini

; PlatformIO Project Configuration File for MightyCore
; https://github.com/MCUdude/MightyCore/
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options
; https://github.com/MCUdude/MightyCore/PlatformIO.md
; https://docs.platformio.org/page/projectconf.html

; ENVIRONMENT SETTINGS
[env:MightyCore]
platform = atmelavr
;platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = mightycore32
; Actual target 
board_build.mcu = atmega32
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; BUILD OPTIONS
; Current pinout
board_build.variant = mightycore_standard
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
;build_flags = 

; UPLOAD SETTINGS
; Serial port. Use COMx on Windows
;upload_port = /dev/cu.usbserial*
; Upload baud rate
board_upload.speed = 115200

; Upload using programmer
;upload_protocol = usbasp
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600

```

ivankravets, 2019-05-21:
I've just thought... to avoid any delays you can make a PR with own boards in the old format. Later, we will reformat them.

As you described here => https://github.com/platformio/platform-atmelavr/issues/138#issuecomment-490010934

MCUdude, 2019-05-21:
That's fine! Is it OK that I do changes to existing JSON files as well (atmega328pb.json for instance)?

ivankravets, 2019-05-21:
I see that `atmega328pb.json` uses an official Arduino core. I propose to keep it without changes. Maybe, some developers use it in their projects.

As I understand correctly, they can switch to your core later without any issues.

MCUdude, 2019-05-22:
> I see that atmega328pb.json uses an official Arduino core.

The 328pb isn't officially supported by the official Arduino core. There's no pin mapping for the extra pins, and PWM generated by timer3 and 4 isn't available through analogWrite.

> As I understand correctly, they can switch to your core later without any issues.

Yes, this is true. It's 100% compatible with the official core but offers some extended functionality that the 328pb needs. It would be a pity if we can't have proper 328pb support in PlatformIO IMO. Please let me know how we can solve this.

ivankravets, 2019-05-22:
Ah, I got you. Yes, please update `atmega328pb.json` too. Thanks!

MCUdude, 2019-05-23:
Do you want the "name" fields to be:
"ATmega328P"
"Atmel ATmega328P"
"MCUdude ATmega328P"
"MiniCore ATmega328P"

Personally, I'd prefer the first one if it's possible to choose between the official Arduino framework and MiniCore in a separate menu when you start a new project. Is this option something that should be added to the JSON files?

MCUdude, 2019-05-23:
.. And how should I deal with devices that doesn't have a bootloader (and will have to be programmed with a program like USBasp or USBtinyISP). Like this?

``` json

{
  "build": {
    "core": "MiniCore",
    "extra_flags": "-DARDUINO_AVR_ATmega48P",
    "f_cpu": "16000000L",
    "mcu": "atmega48p",
    "variant": "minicore_standard"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATmega48P/PA",
  "upload": {
    "maximum_ram_size": 512,
    "maximum_size": 4096,
    "protocol": "arduino",
    "require_upload_port": false,
  },
  "url": "https://github.com/MCUdude/MiniCore",
  "vendor": "MCUdude"
}

```

ivankravets, 2019-05-23:
>   "url": "https://github.com/MCUdude/MiniCore",
>  "vendor": "MCUdude"

What is MCUdude? Do we have physical MCUdude boards?

MCUdude, 2019-05-23:
Well, I do for MightyCore, but not for MiniCore.

Should it say Microchip instead? How about the URL to the MiniCore repo? Lots of useful documentation for the user there.

ivankravets, 2019-05-23:
Oh, sorry :( Now I understand. `MigthyCore` this is software part, not hardware. In this case, I recommend using "device" notation for file name and provide documentation on how to switch between cores. We already did this for ST STM32 => http://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores

They also have a few Arduino cores.

I recommend the next format for `ATMEGA168P.json`:

```
{
  "build": {
    "core": "MiniCore",
    "extra_flags": "..",
    "f_cpu": "16000000L",
    "mcu": "...",
    "variant": "minicore_standard"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATMEGA168P",
  "upload": {
    "maximum_ram_size": 1024,
    "maximum_size": 15872,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 115200
  },
  "url": "https://www.microchip.com/wwwproducts/en/ATmega168P",
  "vendor": "Microchip"
}
```

Later, it will be easy for us to convert it to 2.0 format. The important thing here is a board identifier. It will be the same, so, end developer will not be affected later.

What do you think?

MCUdude, 2019-05-23:
> What do you think?

Looks good to me? But my previous question still remains unanswered. How should I handle ATmega48, which doesn't have a bootloader and require a dedicated programmer? Is it OK if I add `require_upload_port": false,` and remove the speed option?

ivankravets, 2019-05-23:
> Is it OK if I add require_upload_port": false, and remove the speed option?

Yes, please update `upload` section to the working format.

> Is it correct that extra_flags have two dots, while mcu have three dots?

You need to replace these `.....` with real data. `"mcu": "..."` => `"mcu": "atmega168p"`, and `"extra_flags": ".."` to `"extra_flags": "-DARDUINO_AVR_ATMEGA168"` or similar to that.

MCUdude, 2019-05-23:
> You need to replace these ..... with real data. "mcu": "..." => "mcu": "atmega168p", and "extra_flags": ".." to "extra_flags": "-DARDUINO_AVR_ATMEGA168" or similar to that.

Thanks! I found it a bit unclear since ATmega168 was already there under `"name"`

ivankravets, 2019-05-23:
> Thanks! I found it a bit unclear since ATmega168 was already there under "name"

We don't use `name` field in the build process and show only in UI and docs.

MCUdude, 2019-05-23:
Last question, where should I put the MiniCore repo URL?

ivankravets, 2019-05-23:
We will create a new section in our docs similar to STM32. Will it work for you? You can even propose your PR to this section. I'll show later where is this place.

MCUdude, 2019-05-23:
That's OK for me! Do you mean something like this? It would be nice if the repo URL was visible to the user for the user when MiniCore is selected as core.

```json
{
  "build": {
    "core": "MiniCore",
    "core_url": "https://github.com/MCUdude/MiniCore",
    "extra_flags": "-DARDUINO_AVR_ATmega168P",
    "f_cpu": "16000000L",
    "mcu": "atmega168p",
    "variant": "minicore_standard"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATmega168P/PA",
  "upload": {
    "maximum_ram_size": 1024,
    "maximum_size": 15872,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 115200
  },
  "url": "https://www.microchip.com/wwwproducts/en/ATmega168P",
  "vendor": "Microchip"
}
```

ivankravets, 2019-05-23:
This `core_url` will not be visible for and developer. We will put this link in docs.

MCUdude, 2019-05-23:
That's perfectly fine.

Many of the AVR chips have different suffixes but the same device signature. ATmega328P and ATmega328PA share the same signature. Is it OK if these two variants share the same JSON file too?

Filename: atmega48p.json

``` json
{
  "build": {
    "core": "MiniCore",
    "extra_flags": "-DARDUINO_AVR_ATmega48P",
    "f_cpu": "16000000L",
    "mcu": "atmega48p",
    "variant": "minicore_standard"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATmega48P/PA",
  "upload": {
    "maximum_ram_size": 512,
    "maximum_size": 4096,
    "protocol": "arduino",
    "require_upload_port": false,
  },
  "url": "https://www.microchip.com/wwwproducts/en/ATmega48P",
  "vendor": "Microchip"
}
```

ivankravets, 2019-05-23:
> Is it OK if these two variants share the same JSON file too?

Sure!

---

I like it! Just one thing, to keep non-board profiles (for devices) in the capital case => ATMEGA48P.json

closed 2019-06-01 by ivankravets

ivankravets, 2019-06-01:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

MCUdude, 2019-06-02:
I'm able to compile for ATmega328P and ATmega328PB when adding
`platform = https://github.com/platformio/platform-atmelavr.git` to platformio.ini.

`platform = atmelavr` does not work.

However, I'm not able to create a new project with VSCode. See [platformio #2606](https://github.com/platformio/platformio-core/issues/2606) for details.

commit c7d34ddc (2019-06-01): Add support for MiniCore 2.0.3 // Resolve #138