platformio/platform-atmelavr #139: Export HEX file with bootloader included

MCUdude, 2019-05-03:
Hi!
In Arduino IDE it's possible to export the compiled project hex file with the bootloader included in the same HEX file. Would this be possible with PlatformIO as well? As for MightyCore (and other MCUdude cores),  F_CPU, upload speed, and upload port have to be defined somehow. 

Thanks!

ivankravets, 2019-05-06:
How do you do that with Arduino IDE?

MCUdude, 2019-05-06:
The *Export compiled binary* option leaves two hex files in the sketch folder. One with and one without a bootloader included. Very useful!

[image]

pfeerick, 2019-05-07:
The binary that platformio creates is relatively easy to find... it's in the project folders .pioenvs/environment-name folder... i.e. for an project with an environment name of 'uno' it's in the `.pioenvs/uno` folder as `firmware.hex`... the giveaway being the last few lines of the compile... but that doesn't help with a version _with_ a bootloader. 

```
Linking .pioenvs/uno/firmware.elf
Checking size .pioenvs/uno/firmware.elf
Building .pioenvs/uno/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.4% (used 9 bytes from 2048 bytes)
PROGRAM: [          ]   1.4% (used 444 bytes from 32256 bytes)
```

If this information is correct, it should't be to hard to add building a bootloader as an extra_script.py, as it's 'just' a matter of adding the bootloader hex to the application hex, with one minor change. 

https://www.avrfreaks.net/forum/merging-bootloader-hex-file-and-application-hex-file-one

josefgull, 2021-02-25:
is there now a sultion i would also like to generaate an hexfirmware file with a bootloader for an mrkzero

deladriere, 2021-06-10:
@josefgull I use Bossac to download the firmware from the chip once programmed in a classic way (using the IDE- Plaftormio or Arduino)
It includes the bootloader and can be used with a binary programmer
Does it make sense ?