platformio/platform-atmelavr #140: Uploading using programmer doesn't work

MCUdude, 2019-05-03:
Hi!

I believe there's a bug here. When I've specified to use a dedicated programmer in platformio.ini (USBasp in this case) I'm not able to upload successfully by clicking the upload button in VS Code.

A little research reveals that this is caused by the -D flag in avrdude, which prevents the flash from being erased before upload. Is there any reason why the -D flag needs to be there in the first place? It's not used by Arduino IDE. I suggest the user may add -D to `upload_flags` if there's a special case where it's needed. For most users, it's only causing trouble.

ivankravets, 2019-05-06:
Do you use `Program` instead of `Upload`? See http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

MCUdude, 2019-05-06:
No, I'm using `Upload`. `Program` works just fine. But why does the -D flag needs to be there in the first place? It's just causing confusion like this? It would be much more elegant if we could use the same upload command no matter what upload method is used. It's already defined in platformio.ini.
Just my two cents 🙂

pfeerick, 2019-05-07:
> It's not used by Arduino IDE.

Not true. The avrdude upload pattern for the Arduino IDE is :
```
"{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} "-P{serial.port}" -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"
``` 

`-D` is right there just before the 'what to actually flash' `-U` specification. Hence, `Upload` is behaving exactly like the Arduino IDE in this respect.

MCUdude, 2019-05-07:
> -D is right there just before the 'what to actually flash' -U specification. Hence, Upload is behaving exactly like the Arduino IDE in this respect.

Sorry, I must have missed it. I looked through the platform.txt file, but I couldn't find it at the time. So why isn't an erase necessary when uploading using the bootloader. It's clearly needed when uploading using a dedicated programmer.

pfeerick, 2019-05-07:
Because erasing the flash would erase the bootloader... which you don't really want to do since you're about to use it! 😆

MCUdude, 2020-11-27:
A lot has changed since this issue was opened. The current implementation where we're using a custom upload command is much better IMO.

closed 2020-11-27 by MCUdude