platformio/platform-atmelavr #141: Compilation error for board `mightycore644` when including SPI.h in v1.13.0

frazar, 2019-05-04:
Broken build when updating to v1.13.0, with `mightycore644` board. 

I attach a minimal test sample to reproduce the issue: [arduino-blink.zip](https://github.com/platformio/platform-atmelavr/files/3144226/arduino-blink.zip). As you can see from the `platformio.ini` file, the basic program builds fine when using the previous version (`platform = atmelavr@1.12.5`).

Also note that, when commenting `#include <SPI.h>`, the compilation error disappears.

Here is the full compilation log for a clean build.

```
user@machine ~/arduino-blink> pio run
Processing mightycore644_atmelavr1.12.5 (framework: arduino; board: mightycore644; platform: atmelavr@1.12.5)
--------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/mightycore644.html
PLATFORM: Atmel AVR > MightyCore ATmega644
HARDWARE: ATMEGA644P 16MHz 4KB RAM (63KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI (MightyCore)> 1.0
Compiling .pioenvs/mightycore644_atmelavr1.12.5/src/Blink.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/lib8e1/SPI/SPI.cpp.o
Archiving .pioenvs/mightycore644_atmelavr1.12.5/libFrameworkArduinoVariant.a
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/CDC.cpp.o
Indexing .pioenvs/mightycore644_atmelavr1.12.5/libFrameworkArduinoVariant.a
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HID.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HardwareSerial.cpp.o
Archiving .pioenvs/mightycore644_atmelavr1.12.5/lib8e1/libSPI.a
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HardwareSerial0.cpp.o
Indexing .pioenvs/mightycore644_atmelavr1.12.5/lib8e1/libSPI.a
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HardwareSerial1.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HardwareSerial2.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/HardwareSerial3.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/PluggableUSB.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/Tone.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/USBCore.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/WInterrupts.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/WMath.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/WString.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/_wiring_pulse.S.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/abi.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/avr-libc/malloc.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/avr-libc/realloc.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/hooks.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/main.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/new.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring_analog.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring_digital.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring_extras.cpp.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/mightycore644_atmelavr1.12.5/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/mightycore644_atmelavr1.12.5/libFrameworkArduino.a
Indexing .pioenvs/mightycore644_atmelavr1.12.5/libFrameworkArduino.a
Linking .pioenvs/mightycore644_atmelavr1.12.5/firmware.elf
Checking size .pioenvs/mightycore644_atmelavr1.12.5/firmware.elf
Building .pioenvs/mightycore644_atmelavr1.12.5/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.2% (used 9 bytes from 4096 bytes)
PROGRAM: [          ]   1.5% (used 986 bytes from 64512 bytes)
================================== [SUCCESS] Took 1.93 seconds ==================================

Processing mightycore644_atmelavr1.13.0 (framework: arduino; board: mightycore644; platform: atmelavr@1.13.0)
--------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
[... 107 lines cut ...]
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp:45:13: error: 'SS1' was not declared in this scope
pinMode(SS1, OUTPUT);
^
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp:50:5: error: 'SPCR1' was not declared in this scope
SPCR1 |= _BV(MSTR);
^
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp:59:13: error: 'SCK1' was not declared in this scope
pinMode(SCK1, OUTPUT);
^
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp:60:13: error: 'MOSI1' was not declared in this scope
pinMode(MOSI1, OUTPUT);
^
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp: In static member function 'static void SPI1Class::end()':
/home/frazar/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/SPI1/src/SPI1.cpp:74:5: error: 'SPCR1' was not declared in this scope
SPCR1 &= ~_BV(SPE);
^
Compiling .pioenvs/mightycore644_atmelavr1.13.0/FrameworkArduino/CDC.cpp.o
*** [.pioenvs/mightycore644_atmelavr1.13.0/lib13b/SPI1/SPI1.cpp.o] Error 1
=================================== [ERROR] Took 1.01 seconds ===================================

=========================================== [SUMMARY] ===========================================
Environment mightycore644_atmelavr1.12.5	[SUCCESS]
Environment mightycore644_atmelavr1.13.0	[ERROR]
=================================== [ERROR] Took 2.95 seconds ===================================
```

MCUdude, 2019-05-06:
I know all about it. I did fix this some days ago, but it's not available through PlatformIO yet. It can be solved by either replacing the libraries with the upstream version. 

I suggest @ivankravets updates the MightyCore core files based on the upstream version and not the latest release. I'll issue a new MightyCore release very soon, but at the moment I'm waiting for a collaborator to figure out some cycle accurate code in order to achieve accurate delayMicroseconds() for odd "UART frequencies" such as 1.8432 MHz, 7.3728 MHz and 14.7456 MHz. This will make MightyCore even more versatile for PlatformIO users.

ivankravets, 2019-05-07:
@MCUdude please ping me when a release will be ready. We currently support stable software in our PlatformIO dev/platforms. You can create your own dev/platform and maintain beta software. However, I don't recommend to do that. It's very difficult to track issues.

MCUdude, 2019-05-07:
> MCUdude please ping me when a release will be ready.

I certainly will!

ivankravets, 2019-06-22:
Should be fixed with the latest release.

closed 2019-06-22 by ivankravets

gocamiot, 2019-08-18:
Is this fixed yet?

ivankravets, 2019-08-19:
See https://github.com/platformio/platformio-pkg-framework-arduinoavr/issues/20