platformio/platform-atmelavr #145: Mouse.h and Arduino Pro Micro doesn't work in PlatformIO

mrbaungartner, 2019-05-27:
I’m using a joystick and an Arduino Pro Micro to emulate the mouse, I use exactly the same code on both platforms, it compiles perfectly, with no errors, however it only works if I pass the code via Arduino IDE.

I downloaded the latest version of VS Code with PlatformIO, i also downloaded the library Mouse.h from PlatformIO Library Manager, and even so, after i upload the code to my Micro Pro the mouse does not respond to the joystick!
**But the same code works when i upload via Arduino IDE!**

Here is the code:

```
/* HID Joystick Mouse Example
   by: Jim Lindblom
   date: 1/12/2012
   license: MIT License - Feel free to use this code for any purpose.
   No restrictions. Just keep this license if you go on to use this
   code in your future endeavors! Reuse and share.

   This is very simplistic code that allows you to turn the
   SparkFun Thumb Joystick (http://www.sparkfun.com/products/9032)
   into an HID Mouse. The select button on the joystick is set up
   as the mouse left click.
 */

#include <Arduino.h>
#include <Mouse.h>
int horzPin = A0;  // Analog output of horizontal joystick pin
int vertPin = A1;  // Analog output of vertical joystick pin
int selPin = 9;  // select button pin of joystick

int vertZero, horzZero;  // Stores the initial value of each axis, usually around 512
int vertValue, horzValue;  // Stores current analog output of each axis
const int sensitivity = 200;  // Higher sensitivity value = slower mouse, should be <= about 500
int mouseClickFlag = 0;

void setup()
{
  pinMode(horzPin, INPUT);  // Set both analog pins as inputs
  pinMode(vertPin, INPUT);
  pinMode(selPin, INPUT);  // set button select pin as input
  digitalWrite(selPin, HIGH);  // Pull button select pin high
  delay(1000);  // short delay to let outputs settle
  vertZero = analogRead(vertPin);  // get the initial values
  horzZero = analogRead(horzPin);  // Joystick should be in neutral position when reading these
}

void loop()
{
  vertValue = analogRead(vertPin) - vertZero;  // read vertical offset
  horzValue = analogRead(horzPin) - horzZero;  // read horizontal offset
//delay(3000);

  if (vertValue != 0)
    Mouse.move(0, vertValue/sensitivity, 0);  // move mouse on y axis
  if (horzValue != 0)
    Mouse.move((horzValue/sensitivity) *-1, 0, 0);  // move mouse on x axis

  if ((digitalRead(selPin) == 0) && (!mouseClickFlag))  // if the joystick button is pressed
  {
    mouseClickFlag = 1;
    Mouse.press(MOUSE_LEFT);  // click the left button down
  }
  else if ((digitalRead(selPin))&&(mouseClickFlag)) // if the joystick button is not pressed
  {
    mouseClickFlag = 0;
    Mouse.release(MOUSE_LEFT);  // release the left button
  }
}
```

pfeerick, 2019-06-27:
Just linking this to the forum thread for completeness: https://community.platformio.org/t/the-code-compile-in-arduino-ide-and-platformio-but-only-works-if-using-arduino-ide/7744

Also, I have a Arduino Micro on the way so I can test this on my end...

zaggash, 2019-09-18:
Got the same issue here but with an Aduino Mega and another library (fastled)
Exact same symptoms.

mrbaungartner, 2019-09-18:
> Got the same issue here but with an Aduino Mega and another library (fastled)
> Exact same symptoms.

Sadly the Platformio is not reliable.

Nickjgniklu, 2024-01-21:
For Mouse you will need to first add `lib_deps = Mouse
` in platformio.ini then add `#include <Mouse.h>` in main.cpp or other files using mouse in your project