platformio/platform-atmelavr #147: 32-bit avrdude will cease to function on macOS 10.15

noisymime, 2019-06-05:
The current version of `tool-avrdude` (1.60300.190424) on macOS contains a 32-bit binary for `avrdude`. Under the upcoming version of macOS (10.15) 32-bit binaries will not function and I have confirmed under the developer preview that the packaged `avrdude` will not run. 

Version 6.3 of avrdude available at http://download.savannah.gnu.org/releases/avrdude/ appears to compile cleanly under macOS as a 64-bit binary and functions as expected

ivankravets, 2019-06-06:
Thanks! Please re-run `pio update`. Does it work now?

noisymime, 2019-06-06:
Thanks @ivankravets 
I've updated and the `avrdude` binary is indeed now the 64-bit version, however looks to be relying on some extra libs that aren't present for me:

`dyld: Library not loaded: /usr/local/opt/libftdi/lib/libftdi1.2.dylib
  Referenced from: /Users/josh/.platformio/packages/tool-avrdude/bin/./avrdude
  Reason: image not found
Abort trap: 6`
and
`dyld: Library not loaded: /usr/local/opt/libusb-compat/lib/libusb-0.1.4.dylib
  Referenced from: /Users/josh/.platformio/packages/tool-avrdude/bin/./avrdude
  Reason: image not found
Abort trap: 6`

Both of those are available through `brew` and installing them makes the binary work, but neither are standard on macOS. Was this binary one you've compiled yourself or gotten from `brew`?

ivankravets, 2019-06-07:
I compiled it. Do you use PlatformIO for flashing? This avrdude does not work as a separate binary. Just only from PlatformIO Core. Try to flash board with PlatformIO.

mrVanboy, 2019-06-15:
Installing the latest `avrdude` from brew and replacing with it preinstalled binary in PlatformIO directory works for me.

```bash
brew install avrdude
mv ~/.platformio/packages/tool-avrdude/bin/avrdude ~/.platformio/packages/tool-avrdude/bin/avrdude.bak
cp $(which avrdude) ~/.platformio/packages/tool-avrdude/bin/avrdude
```

Symlinking should probably work too.

noisymime, 2019-06-21:
@ivankravets With the latest update, I can't upload through PlatformIO at all. I get the same error as trying to run from the command line:
`
Looking for upload port...
Auto-detected: /dev/cu.usbmodem14201
Uploading .pioenvs/megaatmega2560/firmware.hex
dyld: Library not loaded: /usr/local/opt/libftdi/lib/libftdi1.2.dylib
Referenced from: /Users/josh/.platformio/packages/tool-avrdude/bin/avrdude
Reason: image not found
*** [upload] Error -6
`
Folder ` /usr/local/opt/libftdi` doesn't exist on this system and isn't a default one at all.

ivankravets, 2019-06-23:
@noisymime what is your macOS version?

noisymime, 2019-06-23:
This is on 10.14.5, but I can give it a try on my 10.15 dev image in a day or so.

pfeerick, 2019-06-27:
Is it worth trying the version that Arduino are using? I believe they rebuilt it in the last week or so for 64bit mac? From their [package.json](https://github.com/arduino/Arduino/blob/master/hardware/package_index_bundled.json): 

http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino14-i386-apple-darwin11.tar.bz2

ivankravets, 2019-06-28:
@pfeerick 
```
$ file arduino-cc/avrdude/bin/avrdude
arduino-cc//avrdude/bin/avrdude: Mach-O executable i386
```

noisymime, 2019-06-28:
The 64-bit binary from the Arduino IDE is at: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2

Note that the PR to actually switch over to this hasn't been merged yet, so it might still be under testing. It works fine on the 10.14 and 10.15 machines I've tried it on though. I tried it on a completely vanilla, fresh install 10.14.5 (ie nothing from brew) and it works.

ivankravets, 2019-06-28:
Thanks! Please re-run `pio update`. The package is deployed to our CDN.

closed 2019-06-28 by ivankravets

pfeerick, 2019-06-28:
@ivankravets Oops? As noisymime indicated... I jumped the gun... the PR hasn't been committed yet, so it's not live... hence the wrong binary. :-/ 

It looks like that arduino/Arduino/pull/8976 also pushes a new version of avr-gcc also to stop OSX 10.15 moaning about it also. Just have to keep an eye on this one as there appears to be some slowdown on compile and upload ... although it's limited to the 10.15 Catalina betas for now...

ivankravets, 2019-06-28:
@pfeerick it works for me on 10.13 and @noisymime reported the same for  10.14 and 10.15.