platformio/platform-atmelavr #152: Error compiling for ATmega324PB

MCUdude, 2019-06-25:
Hi!
I have a project where PlatformIO is used to program an ATmega324PB. After I upgraded the toolchain I now get this error:

```
avr-gcc: error: device-specs/specs-atmega324pb: No such file or directory
```

I looked in the device-specs folder, and the file is indeed missing. Can you fix this soon? 🙂

MCUdude, 2019-06-25:
turns out `iom324pb.h` is also missing.

It also seems like MightyCore v2.0.2 isn't available in my current installation. How can I add this version to PlatformIO? v2.0.2 brings printf support to the print class (Serial.printf is now a thing, at least in my Arduino cores), and I need this for the project I'm working on.

MCUdude, 2019-07-05:
@ivankravets could you look into this?

ivankravets, 2019-07-10:
Thanks! Please re-run `pio update`.

closed 2019-07-10 by ivankravets

MCUdude, 2019-07-10:
Cool! Everything seems to work fine now!

Thanks!

commit 574da9b8 (2019-07-10): Fix error compilation for ATmega324PB // Resolve #152