platformio/platform-atmelavr #153: Extra script for setting fuses and burning bootloader

MCUdude, 2019-07-11:
Hi!

As a developer working with AVRs I feel like the last thing missing from PlatformIO now is the ability to set the correct fuse bits and burn the correct bootloader based on the information given in platformio.ini.

[MightyCore](https://github.com/MCUdude/MightyCore),  [MiniCore](https://github.com/MCUdude/MiniCore), and [MegaCore](https://github.com/MCUdude/MegaCore) (all supported by PlatformIO) comes with a bunch of pre-compiled bootloaders where all shares the same file names rules.

As long as `board`, `board_build.f_cpu`, and `board_upload.speed` is present in platformio.ini Is it possible for a script to figure out the correct fuse bits and load the correct bootloader hex file.

Preferably there should be three flags, one to only set the fuse bits, one to only burn the bootloader and one for doing both. Since the name and path of the bootloader are so extensive, we can check if the user entered a valid `board_build.f_cpu`, and `board_upload.speed` by checking if a bootloader with these values exists.

A bootloader for a >=64kB device is named like this:
```
bootloaders/atmega1284p/16000000L/optiboot_flash_atmega1284p_UART0_1000000_16000000L_BIGBOOT.hex
```

A bootloader for a <=32kB device is named like this:
```
bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L.hex
```

By adding a new parameter to platformio.ini (`board_upload.uart_port` maybe?) we can specify if UART0, UART1 or UARTn should be used for uploading. Again, this decides what bootloader hex file is loaded.

Another fuse related feature is the ability to set the brownout detection level. This is possible in Arduino IDE through a separate menu option. Something like `board_fuses.bod = 2.7V`?

The last option should be to manually set fuses and override everything. My idea here would be to check if `board_fuses.high`, `board_fuses.low` and `board_fuses.extended is present`. If one or more is present, then this fuse is overridden by whatever value the particular boards_fuses fields holds

I have very little experience with python, but I think this should be doable, even for me if someone points me in the right direction. I also haven't worked with PlatformIO advanced scripting before, so forgive me if some of the following questions are silly to some.

* Can a script like this be ran separately, and not when "regular" code is compiled and uploaded?
* Is a command similar to these realistic?
  - `pio extra_script -f` or `pio extra_script --fuses`
  - `pio extra_script -b` or `pio extra_script --bootloader`
  - `pio extra_script -f -b` or `pio extra_script --fuses --bootloader`
* Does there exist any examples I can look to learn how this may me implemented?

ivankravets, 2019-07-11:
Hi! It should not be difficult. The right place is extending this target https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L247.

P.S: I added "Configuration" section to http://docs.platformio.org/en/latest/frameworks/arduino.html and references to your manuals on how to use PlatformIO + Cores.

MCUdude, 2019-07-11:
> P.S: I added "Configuration" section to http://docs.platformio.org/en/latest/frameworks/arduino.html and references to your manuals on how to use PlatformIO + Cores.

Awesome! Now it should be fairly easy for the average user to use my cores through PlatformIO 👍 

> It should not be difficult. The right place is extending this target https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L247.

OK, so this wouldn't be implemented in a separate script, but rather the main build script instead?
So the commands would instead be `pio target_fuses` and `pio target_bootloader`? I want these commands to be available if `board`,` board_build.f_cpu`, and `board_upload.speed`is present.
If they are, then we can start parsing the approperiate boards.txt files.

ivankravets, 2019-07-11:
I think 2 targets are enough:

* `fuses`, which means setting fuses
* `bootloader`, which means burn bootloader?

MCUdude, 2019-07-11:
> I think 2 targets are enough:
fuses, which means setting fuses
bootloader, which means burn bootloader?

Exactly. But we're not able to set the correct fuses and burn the correct bootloader file if we're missing information. Here's the idea:

#### Option 1: Only `boards` are present
defaults to `board_build.f_cpu=16000000L`
defaults to `board_upload.speed=115200`
defaults to `board_upload.uart_port=UART0`
defaults to `board_fuses.bod=2.7V`

#### Option 2: `boards`,`board_build.f_cpu`and `board_upload.speed` are present
defaults to `board_upload.uart_port=UART0`
defaults to `board_fuses.bod=2.7V`
Will not execute if bootloader file isn't found

#### Option 3: `boards`,`board_build.f_cpu`, `board_upload.speed` and `board_upload.uart_port`are present
defaults to `board_fuses.bod=2.7V`
Will not execute if bootloader file isn't found

**Option 4: `boards`,`board_build.f_cpu`, `board_upload.speed` and `board_fuses.bod`are present**

defaults to `board_upload.uart_port=UART0`
Will not execute if bootloader file isn't found

#### Option 5: `boards`,`board_build.f_cpu`, `board_upload.speed`, `board_fuses.bod` and `board_upload.uart_port`are present

Will not execute if bootloader file isn't found

MCUdude, 2019-07-11:
Maybe it's better if we're able to move everything fuse related into the manifest JSON files? Then there's no dependencies other then the bootloader files.

ivankravets, 2019-07-11:
We did this for some boards. Do you mean this https://github.com/platformio/platform-atmelavr/blob/develop/boards/uno.json#L31 ?

MCUdude, 2019-07-11:
Yes, but those fuses are fixed at BOD=2.7V, and external oscillator.
What I'd like to implement is something that somehow generates the correct fuse bits based on what platformio.ini contains. If you look at the boards.txt files for MightyCore for instance, this is solved by representing the fuses in binary. Because PlatformIO now has separate manifest files for every target (e.g one for 324A, 324P, and 324PB) we only need binary representation for those that don't have an extended fuse.

If you want I can come up with a better pseudo-code example to show you better what I mean.
But that would be later today because right now I'm on my way to the beach! 🏖 
Anyways I think this would be a valuable and useful contribution to the PlatformIO project if I succeed!

ivankravets, 2019-07-11:
If you can help with PR, it would be great! I hope you see how we are busy with generic PlatformIO Core, our IDE extensions. The time is limited for dev/platforms now :(

> But that would be later today because right now I'm on my way to the beach! 🏖

He-he :) Have a nice relaxing!

MCUdude, 2019-07-11:
> If you can help with PR, it would be great!

I'll see what I can do. It will take some time though. And I'll probably ask some questions on the way if that's fine?

> I hope you see how we are busy with generic PlatformIO Core, our IDE extensions. The time is limited for dev/platforms now :(

It's fully understandable, and I do not expect you do do this for me _at all_. It's just that I might need some input from you to have an OK starting point.

So for reference, how can I set fuses for the `uno` target?
`pio fuses` doesn't seem to work.

ivankravets, 2019-07-19:
There is a new task in PlatformIO IDE for VSCode. Or, you can use PlatformIO CLI:
“$ pio run -t fuses

MCUdude, 2019-08-29:
I've been busy doing other things recently, but I just started looking at this again. To start out simple with a proof of concept I decided to stick with a separate `extra_script` for now. HOwever, I can't really get it to parse anything from platformio.ini.

Here's my platformio.ini file:
``` ini
[env:env_custom_target]
platform = atmelavr
board = ATmega324P
framework = arduino

extra_scripts = extra_script.py
custom_message = "Hello World!"
```

And here's the script I try to run using this command: `pio run --target testscript`
``` python
try:
    import configparser
except ImportError:
    import ConfigParser as configparser

Import("env")

config = configparser.ConfigParser()
config.read("platformio.ini")
host = config.get("env_custom_target", "custom_message")

def mytarget_callback(*args, **kwargs):
    print("Hello PlatformIO!")
    print(host)

env.AlwaysBuild(env.Alias("testscript", None, mytarget_callback))
```

And here's the error I get:
```
NoSectionError: No section: 'env_custom_target':
  File "C:\Users\h.bull.LAUDM\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 135:
    env.SConscript(item, exports="env")
  File "U:\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "U:\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\h.bull.LAUDM\Documents\PlatformIO\Projects\ADV_SCRIPT\extra_script.py", line 10:
    host = config.get("env_custom_target", "custom_message")
  File "C:\Users\h.bull.LAUDM\.platformio\python27\Lib\ConfigParser.py", line 607:
    raise NoSectionError(section)
```

ivankravets, 2019-08-29:
See docs for ConfigParser https://docs.python.org/2/library/configparser.html

You need to do checking before `.get(..)`. For example,
```
if config. has_section("..."):
    # do something
```

MCUdude, 2019-08-29:
Thanks! However, it turned out that the section I had entered was incorrect. It was supposed to be `env:env_custom_target` instead of `env_custom_target`.

Not I have something I can work with. I'm a complete Python noob, as has to Google for even the simplest things. But that's my problem 😉

For a large project, the platformio.ini file may contain several environments or `[env:myEnv]`. Let's say one is `[env:atmega324p]` and the other one is [env:atmega1284p]. How can I invoke the script to handle only one of them, and not both, without physically modifying the script to match the current project? Somehow the environment I want to use has to somehow be passed to the script.

Something like this?
`pio run --target testscript -[someflag] env:atmega324p`

ivankravets, 2019-08-29:
If you need to get the only option from a current environment, just use this shortcut: https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/pioproject.py#L28

```python
print(env.GetProjectOption("env_custom_target", "custom_message"))
```

In this case, no need to use ConfigParser at all.

MCUdude, 2019-08-29:
Cool! Is it also possible to check if an option is present? This way I can make some options optional, like `uart` and ``bod`. As you already know, I want the script to be able to figure out the correct fuse bits and later set fuse bits + burn bootloader based on the contents in the platformio.ini file.

But let's say I have these two environments. I only want to run the extra_script for the first one, not the second.

```ini
[env:env_custom_target]
platform = atmelavr
board = ATmega324P
framework = arduino

extra_scripts = extra_script.py

[env:env_custom_target2]
platform = atmelavr
board = ATmega324PA
framework = arduino
```

```python

def fuses(*args, **kwargs):
    print(env.GetProjectOption("board"))
    
env.AlwaysBuild(env.Alias("testscript", None, fuses))
```

when I run `pio run --target testscript` I finally get this output.
```
Environment         Status    Duration
------------------  --------  ------------
env_custom_target   SUCCESS   00:00:12.942
env_custom_target2  FAILED    00:00:09.423
```

I don't want the script to run for the second environment. How can I prevent this?

ivankravets, 2019-08-30:
> Is it also possible to check if an option is present?

Just 

```python
if env.GetProjectOption("option_exists"):
    # do something

# or

if not env.GetProjectOption("option_does_not_exists"):
    pass 
```

> I don't want the script to run for the second environment. How can I prevent this?

A lot of ways:

- http://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-e
- http://docs.platformio.org/en/latest/projectconf/section_platformio.html#default-envs

P.S: `pio run -e env_custom_target -t testscript`

MCUdude, 2019-08-30:
Awesome! I'll see of I can implement this in a nice way.

At the moment I have a script that takes these parameters in the platformio.ini file:
```
board = ATmega1284P
board_build.f_cpu = 8000000L
hardware.oscillator = internal
hardware.bod = 2.7V
hardware.uart = uart0
```

And based on these parameters will calculate the correct `lfuse`, `hfuse` and `efuse` for all MightyCore, MiniCore, MegaCore, and MajorCore compatible microcontrollers. With all sub variants (P, PA, PB etc) that's 31 different targets!

MCUdude, 2019-08-30:
OK, so I now have the fuses (`lfuse`, `hfuse`, `efuse`) stored in a variable each. How do I invoke Avrdude?

I need to build a command similar to this:
```
avrdude -Cavrdude.conf -v -p{board} -c{upload_protocol} -P{upload_flags} -e -Ulock:w:0x3f:m -Uefuse:w:{hfuse}:m -Uhfuse:w:{hfuse}:m -Ulfuse:w:{lfuse}:m 
```

MCUdude, 2019-08-30:
For reference, here's the script and ini file

```ini
[env:env_custom_target]
platform = atmelavr
board = ATmega32
framework = arduino
board_build.f_cpu = 1000000L
hardware.oscillator = internal
hardware.bod = disabled
hardware.uart = disabled
extra_scripts = extra_script.py
```

``` python
Import("env")

# Default values
target = str(env.GetProjectOption("board"))
f_cpu = "16000000L"
oscillator = "external"
bod = "2.7V"
uart = "uart0"

def get_lfuse():
    global target
    global f_cpu
    global oscillator
    global bod

    if(target == "ATmega2561" or target == "ATmega2560"  or target == "ATmega1284"  or target == "ATmega1284P" or \
       target == "ATmega1281" or target == "ATmega1280"  or target == "ATmega644A"  or target == "ATmega644P"  or \
       target == "ATmega640"  or target == "ATmega328"   or target == "Atmega328P"  or target == "ATmega324A"  or \
       target == "ATmega324P" or target == "ATmega324PA" or target == "ATmega168"   or target == "ATmega168P"  or \
       target == "ATmega164A" or target == "ATmega164P"  or target == "ATmega88"    or target == "ATmega88P"   or \
       target == "ATmega48"   or target == "ATmega48P"):
        if(oscillator == "external"):
            return 0xf7
        else:
            if(f_cpu == "8000000L"):
                return 0xe2
            else:
                return 0x62
    
    elif(target == "ATmega328PB" or target == "ATmega324PB" or target == "ATmega168PB" or target == "ATmega162" or \
         target == "ATmega88PB"  or target == "ATmega48PB"):
        if(oscillator == "external"):
            return 0xff
        else:
            if(f_cpu == "8000000L"):
                return 0xe2
            else:
                return 0x62
    
    elif(target == "AT90CAN128" or target == "AT90CAN64" or target == "AT90CAN32"):
        if(oscillator == "external"):
            return 0xcf
        else:
            if(f_cpu == "8000000L"):
                return 0xe2
            else:
                return 0x62
    
    elif(target == "ATmega8535" or target == "ATmega8515" or target == "ATmega32" or target == "ATmega16" or  \
         target == "ATmega8"):
        if(bod == "4.0V"):
            bod_bits = 0b00
        elif(bod == "2.7V"):
            bod_bits = 0b10
        else:
[... 155 lines cut ...]
    # Define UART port
    if(str(env.GetProjectOption("hardware.uart")) == "uart0" or str(env.GetProjectOption("hardware.uart")) == "uart1" or str(env.GetProjectOption("hardware.uart")) == "uart2" or str(env.GetProjectOption("hardware.uart")) == "uart3"):
        uart = str(env.GetProjectOption("hardware.uart"))
        print("UART port specified\t\tUsing hardware.uart = %s" % uart)
    elif(str(env.GetProjectOption("hardware.uart")) != "None"):
        uart = "no_bootloader"
        print("UART not specified\t\tNo bootloader will be installed")
    else:
        print("UART port not specified\t\tDefault is hardware.uart = %s" %uart)
    

    low_fuse = hex(get_lfuse())
    high_fuse = hex(get_hfuse())
    ext_fuse = hex(get_efuse())

    print("\nCalculated low fuse:  %s" % low_fuse)
    print("Calculated high fuse: %s" % high_fuse)
    print("Calculated ext fuse:  %s" % ext_fuse)

    # Invoke Avrdude here!
    

env.AlwaysBuild(env.Alias("fuses", None, fuses))

```

MCUdude, 2019-09-02:
@ivankravets sorry for bothering you, but I'm just not skilled enough to figure out how to invoke Avrdude at this point., so I'm pretty much stuck 😞

ivankravets, 2019-09-02:
We would like to have this as a part of dev/platform. Is it ok for you? I think you provided a lot of valuable information. I propose to move this code to `builder/bootloader.py` and extend with `bootloader` target.

Could you post here a few commands how should look AVRdude command? How many of them we should call per 1 bootloader flashing?

MCUdude, 2019-09-02:
That would be great! What's important to me is that it's easy to use this functionality

I think it would be great if we had separate commands for setting fuses and burning bootloader.
For small ATtiny AVRs, a bootloader doesn't make sense, but fuses are still very important.

Here's a suggestion on how the fuses command could/should look like:
`pio target fuses`. 
This would invoke the following Avrdude command:
```
avrdude -Cavrdude.conf -v -p{board} -c{upload_protocol} -P{upload_flags} -e -Ulock:w:0x3f:m -Uefuse:w:{hfuse}:m -Uhfuse:w:{hfuse}:m -Ulfuse:w:{lfuse}:m 
```

Here's a suggestion on how the bootloader command could/should look like:
`pio target bootloader`.
This would invoke the following Avrdude command:
```
avrdude -Cavrdude.conf -v -p{board_build.mcu} -c{upload_protocol} -P{upload_flags -Uflash:w:/path/to/bootloaders/optiboot_flash/bootloaders/{board_build.mcu}/{board_build.f_cpu}/optiboot_flash_board_build.mcu_{hardware.uart}_{board_upload.speed}_{board_build.f_cpu}.hex:i -Ulock:w:0x0f:m 
```

Note that some chips like ATmega8515, ATmega8535, ATmega8, ATmega16, and ATmega32 doesn't have an efuse. In order to not have a special fuses command just for these, we can add a "fake" efuse option in the avrdude.conf file. I've done this with [MightyCore's avrdude.conf](https://github.com/MCUdude/MightyCore/blob/ae98efb6dca23ab56fa3e154cde9317cae658064/avr/avrdude.conf#L1651-L1653) for instance.

Have a look at my [Optiboot flash repo](https://github.com/MCUdude/optiboot_flash) on how different hex files for different targets are handled. This repo covers pretty much all AVR based boards out there, except the official Arduino Mega 2560 (uses an stk500v2 bootloader instead) and Arduino Nano (uses ATMEGABOOT instead).

If this makes into the official PlatformIO core it would make a BIG difference for AVR and Arduino developers who are working bare hardware and not just development boards. This is exciting!

MCUdude, 2019-09-02:
Another important feature would be to manually override the generated fuses. If you have a very specific need for a certain type of fuse setting, it should be possible to specify it in platformio.ini, like this. I can easily add this to the original script!

```ini
board_fuses.hfuse = 0x[nn]
board_fuses.lfuse = 0x[nn]
board_fuses.efuse = 0x[nn]
```

ivankravets, 2019-09-02:
@valeros will back soon here with the updates.

MCUdude, 2019-09-02:
Great! Looking forward to discussing these features with you. BTW I've updated the script I posted earlier. Not it supports overriding of lfuse, hfuse, and efuse in platformio.ini.

MCUdude, 2019-09-03:
OK, I've done some more work on the fuses script. 
* It's now possible to control the EESAVE fuse bit for those who need this (Used in ATTinyCore for instance). Use `hardware.eesave` to control this fuse bit.
* Fix issue with some bitshifted fuse settings that caused the wrong fuses to be generated.
* Added automatic correction of capital/non-capital letters.
* ATtiny13/A support (Will add support for [MicroCore](https://github.com/MCUdude/MicroCore) later)

@valeros when will I hear from you?
``` python
Import("env")

# Default values
target = str(env.GetProjectOption("board"))
f_cpu = "16000000L"
oscillator = "external"
bod = "2.7v"
eesave = "yes"
uart = "uart0"

def get_lfuse():
    global target
    global f_cpu
    global oscillator
    global bod

    # Return manually defined lfuse if present in platformio.ini
    if(str(env.GetProjectOption("board_fuses.lfuse")) != "None"):
        return int(env.GetProjectOption("board_fuses.lfuse"), 0)

    if(target == "ATmega2561" or target == "ATmega2560"  or target == "ATmega1284"  or target == "ATmega1284P" or \
       target == "ATmega1281" or target == "ATmega1280"  or target == "ATmega644A"  or target == "ATmega644P"  or \
       target == "ATmega640"  or target == "ATmega328"   or target == "Atmega328P"  or target == "ATmega324A"  or \
       target == "ATmega324P" or target == "ATmega324PA" or target == "ATmega168"   or target == "ATmega168P"  or \
       target == "ATmega164A" or target == "ATmega164P"  or target == "ATmega88"    or target == "ATmega88P"   or \
       target == "ATmega48"   or target == "ATmega48P"):
        if(oscillator == "external"):
            return 0xf7
        else:
            if(f_cpu == "8000000L"):
                return 0xe2
            else:
                return 0x62
    
    elif(target == "ATmega328PB" or target == "ATmega324PB" or target == "ATmega168PB" or target == "ATmega162" or \
         target == "ATmega88PB"  or target == "ATmega48PB"  or target == "AT90CAN128"  or target == "AT90CAN64" or \
         target == "AT90CAN32"):
        if(oscillator == "external"):
            return 0xff
        else:
            if(f_cpu == "8000000L"):
                return 0xe2
            else:
                return 0x62

    elif(target == "ATmega8535" or target == "ATmega8515" or target == "ATmega32" or target == "ATmega16" or  \
         target == "ATmega8"):
        if(bod == "4.0v"):
            bod_bits = 0b11
        elif(bod == "2.7v"):
            bod_bits = 0b01
        else:
            bod_bits = 0b00

        if(oscillator == "external"):
            return 0xff & ~(bod_bits << 6)
        else:
            if(f_cpu == "8000000L"):
                return 0xe4 & ~(bod_bits << 6)
            else:
                return 0xe1 & ~(bod_bits << 6)

[... 203 lines cut ...]
    
    # Define UART port
    if(str(env.GetProjectOption("hardware.uart")).lower() == "uart0" or str(env.GetProjectOption("hardware.uart")).lower() == "uart1" or str(env.GetProjectOption("hardware.uart")).lower() == "uart2" or str(env.GetProjectOption("hardware.uart")).lower() == "uart3"):
        uart = str(env.GetProjectOption("hardware.uart")).lower()
        print("UART port specified\t\tUsing hardware.uart = %s" % uart)
    elif(str(env.GetProjectOption("hardware.uart")) != "None"):
        uart = "no_bootloader"
        print("UART not specified\t\tNo bootloader will be installed")
    else:
        print("UART port not specified\t\tDefault is hardware.uart = %s" %uart)
    

    low_fuse = hex(get_lfuse())
    high_fuse = hex(get_hfuse())
    ext_fuse = hex(get_efuse())

    print("\nCalculated low fuse:  %s" % low_fuse)
    print("Calculated high fuse: %s" % high_fuse)
    print("Calculated ext fuse:  %s" % ext_fuse)

    # Invoke Avrdude here!

env.AlwaysBuild(env.Alias("fusess", None, fuses))

```

MCUdude, 2019-09-04:
@ivankravets would you like to show me how an Avrdude command in this script would look like?
While I have everything fresh in memory I'd like to continue working on this script while I'm waiting for @valeros to respond. The next step would be to add an Avrdude command to fuses() and start working on the bootloader part.

ivankravets, 2019-09-04:
Sorry for the delay :( @valeros is a little bit busy with something sweet 💣  for PlatformIO Core 😊 

He will back soon. I think this part of code should help https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L191:L212

MCUdude, 2019-09-04:
> I think this part of code should help https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L191:L212

I've tried this already but wasn't able to figure it out. Could you provide a somehow stripped-down version of this I can experiment with?

ivankravets, 2019-09-04:
> # Invoke Avrdude here!

```python
env.Execute("avrdude --version")
```

MCUdude, 2019-09-04:
Nice, that works. Last question (for now 😉); how to I refer to the avrdude.conf file PlatformIO comes with? Obviously this won't work:

```
env.Execute("avrdude -C avrdude.conf -cusbasp")
```

ivankravets, 2019-09-04:
```python
import os

platform = env.PioPlatform()
avrdude_dir = platform.get_package_dir("tool-avrdude")
avrdude_conf = os.path.join(avrdude_dir, "avrdude.conf")
cmd = [
  "-p", "$BOARD_MCU", 
  "-C", avrdude_conf
]
env.Execute(cmd)
```

P.S: I've not tested it :)

MCUdude, 2019-09-04:
It didn't work directly, but you provided enough information for me to figure it out.

Do you know why I'm getting single quotes and square brackets around the string when I'm trying to get `upload_flags` from platformio.ini?

if `upload_flags = -Pusb` is defined like this in platformio.ini, then `str(env.GetProjectOption("upload_flags"))` will return `['-Pusb']`, not `-Pusb` like I'm used to.

ivankravets, 2019-09-04:
`env.GetProjectOption("upload_flags")` - this is array. You can directly extend it with `cmd`.
```python
cmd.extend(env.GetProjectOption("upload_flags", []))
```

ivankravets, 2019-09-04:
Maybe this will help
```
print(" ".join(env.GetProjectOption("upload_flags", [])))
```

MCUdude, 2019-09-05:
@ivankravets thanks! With a little tweaking, I now have a functional script that will load the correct fuses to (almost) any AVR target based on parameters specified in platformio.ini. I decided to create a temporary repo where I host the script/project. I figured this would be easier for all of us. It's hosted here: https://github.com/MCUdude/pio-script

The next task would be to work on the bootloader command:
`pio run --target bootloader`. I will be using [Optiboot flash](https://github.com/MCUdude/optiboot_flash) since it is tested and proved, and is more or less identical to the official Optiboot code. The strict naming of the bootloader hex files makes it really easy to get the correct bootloader file based on `hardware.uart`, `board_build.f_cpu` and `board_upload.speed.`

ivankravets, 2019-09-05:
Great! You are now supe experienced Python dev 😊 

I see that Arduino Core keeps FUSES in boards.txt in `bootloader.***` section. Does it make sense for us to pre-fill boards/***.json with fuses data? Or, your dynamic way is better?

MCUdude, 2019-09-05:
> Great! You are now super experienced Python dev 😊

 Haha, not at all! 😁 I feel like I'm writing a C program with a weird syntax. But hey, it does work!

> I see that Arduino Core keeps FUSES in boards.txt in bootloader.*** section

bootloader.*** is to match the platform.txt file. The fuses are actually loaded as a separate command _before_ the actual bootloader, so it would actually make sense to rename it to fuses.*** in boards.txt and platform.txt. But it's just for looks, nothing critical.

> Does it make sense for us to pre-fill boards/***.json with fuses data? Or, your dynamic way is better?

If the script is watertight, there is no need to store any fuses in the json manifests. f nothing other than `board` is specified, the calculated fuses will be "safe" no matter what your hardware is.
If the user totally screws up the configuration to a point where it doesn't even make sense the script should write fuses that don't lock the user out. If the script is ran for a target that isn't supported the script should not do anything other than writing an error message to the screen.

Here are some rules I've implemented:
* If no oscillator type is specified, use the internal oscillator (to prevent lockout for hardware that doesn't have an external oscillator)
* If no f_cpu is specified, use 8 MHz. This is the maximum speed of the internal oscillator, and unlike the 1 MHz option it works with "fast" USBasp programmers without having to adjust the clock speed down (by using -B[divider] flag)
* If no uart is specified then use uart0. This lets you burn bootloader without specifying uart. Uart0 is used for uploading on all commercial AVR based boards I know of.
* BOD defaults to 2.7V of not specified. Arduino UNO also uses 2.7V, with most users not even knowing.
* EEPROM preserve is enabled by default. This is what most users who use a bootloader for uploading is used to.

The script still needs lots of testing, especially when the options in platformio.ini is randomly picked.

MCUdude, 2019-09-06:
@ivankravets @valeros So what will happen next? The script is pretty much finished. I'm sure you python gurus can improve the overall syntax to make it more elegant and easier to extend with support for other targets as well (@SpenceKonde's ATTinyCore is the first that comes to mind).

I'm not sure where you plan to place the Optiboot flash folder, so I've placed it under `/packages/framework-arduinoavr`

liebman, 2019-09-13:
Please also look at #157 - need to make sure that flash is not always erased when setting fuses.  If one is setting the reset pin to an IO pin then erasing flash removes firmware loaded before and device is no longer programmable (without a HV programmer)....

MCUdude, 2019-09-13:
The script I've made does not support RSTDISABLE as of today, to prevent users from messing up.
As of today, the script is mostly ATmega related since I mostly work with ATmegas. Other fuse settings for ATtinys could/should be added in the future to make this script even more versatile.

But yeah, it's probably better to erase the flash memory when the bootloader is burned instead. I'll look into it.

@ivankravets I haven't heard from you in a week. Just busy or abandoned the idea of implementing this into the pio core?

ivankravets, 2019-09-13:
@MCUdude It's in our TODO :)

valeros, 2019-09-17:
Hi @MCUdude ! Many thanks for preparing such a good basis for this feature. I've created [a special branch](https://github.com/platformio/platform-atmelavr/blob/feature/new-fuses-programming/builder/fuses.py) with the initial implementation based on the code snippet you posted above. So, most of the changes are stylistic, I also used the `mcu` field as the target (I hope that'll enable this feature for more boards). Also, would be great if you could give meaningful names for the sets of mcus used in calculations (Is there a reason why you grouped them in that way?).  Feel free comment on any issues you can find.

MCUdude, 2019-09-17:
Great cleanup! I didn't use the `mcu` field because I wasn't able to retrieve it fom the manifest files. Does this means the MCU paramter has to be defined in the ini file for this to work? Many that should't ne necessary most manifest files, as their board name acutually match the chip type. maybe you can make the mcu field optional by checking if the board name is a valid target for this script? Just a thought.

> Is there a reason why you grouped them in that way?

The reason why they're grouped the way they are under each fuse function is that their properties are identical, so the calculated value will be the same for the entire group. I used the [Engbedded fusecalc](http://www.engbedded.com/fusecalc) _a lot_ when I worked on this. Some chips have identical lfuse values, some have identical hfuse and some efuse. There isn't really any good patterns. It depends on flash size, chip families, "old" vs "new" AVRs (ATmega8 is old, ATmega88 is new for instance), And other varities between chips. Fuse-vise, the ATmega328P is for instance much closer to ATmega324A/P/PA than ATmega168/P, while Atmega168P/PA is not close to ATmega164A/P at all.

Bear in mind that ATmega8535, ATmega8515, ATmega8, ATmega16 and ATmega32 doesn't have an efuse. To me it looks like it terminates the script of efuse can't be calculated for.

Will you implement bootload burning in a different script?

valeros, 2019-09-18:
@MCUdude If there is no `mcu` field then we'll fall back to the board name, WDYT?
Earlier you mentioned that the `--t bootloader` and `--t fuses` targets can be used separately, but Arduino IDE executes these two steps as a single target, first it sets fuses and then uploads the bootloader. Can we use the fuses calculated by your script and then upload bootloader or we need to use the fuses specified in `boards.txt`?
At the moment, there are no bootloader binaries in our packages, so will it be OK if we make the field to the path to the bootloader a mandatory field in order to program bootloader? 
```
board_bootloader.path = bootloaders/optiboot_atmega328.hex
```

MCUdude, 2019-09-18:
> @MCUdude If there is no mcu field then we'll fall back to the board name, WDYT?

I assume we're talking about the contents of platformio.ini. It sounds like a great idea! if build_board.mcu is missing, the fall back to using board instead.

> Earlier you mentioned that the --t bootloader and --t fuses targets can be used separately, but Arduino IDE executes these two steps as a single target, first it sets fuses and then uploads the [...]

You can safely use the fuses calculated in the script. If the fields are correct in platformio.ini, the fuses will take the bootloader space into account when calculating the fuses. The reason why I suggested adding this to two separate commands is that some users don't need a bootloader for their project, and some chips don't support having a bootloader. Having two separate commands for this gives the user greater flexibility. PlatformIO is supposed to be more advanced than Arduino IDE? Maybe we can add a third command to both set fuses and then burn bootloader?

> At the moment, there are no bootloader binaries in our packages, so will it be OK if we make the field to the path to the bootloader a mandatory field in order to program bootloader?

I'm not sure this is a very good idea. What kind of bootloader that should be loaded highly depends on the configuration done by the user in platformio.ini. We have clock speed, baud rate and UART port for instance. It would IMO be much better if you included the optiboot_flash binaries instead because then the end-user doesn't have to care about what exact hex file is loaded. What's important is that the bootloader hex file matches the platformio.ini configuration.

valeros, 2019-09-18:
> Maybe we can add a third command to both set fuses and then burn bootloader?

Is there a reason why they set fuses before uploading the bootloader? Can we burn bootloader without setting the fuses?

> You can safely use the fuses calculated in the script. If the fields are correct in platformio.ini, the fuses will take the bootloader space into account when calculating the fuses.

Is it OK if calculated fuses for `uno` board differ from the fuses specified in `boards.txt`?

>  What kind of bootloader that should be loaded highly depends on the configuration done by the user in platformio.ini.

So binaries in your repository are fully compatible with bootloaders  for all boards shipped with Arduino IDE  (e.g `gemma`, `leonardo`, etc) ? In other words, will we be able to program boards from the Arduino IDE after programming your precompiled bootloader binary?

MCUdude, 2019-09-18:
> Is there a reason why they set fuses before uploading the bootloader? Can we burn bootloader without setting the fuses?

Let's say you want to want to change the upload speed. You can do this by only re-burn the bootloader. No need to re-set the same fuses again. But it's not a very big deal. It's to just have one command. It's also less confusing which parameter changes require new fuses and which require a new bootloader.

>So binaries in your repository are fully compatible with bootloaders for all boards shipped with Arduino IDE (e.g gemma, leonardo, etc) ? In other words, will we be able to program boards from the Arduino IDE after programming your precompiled bootloader binary?

No, the ATmega32U4 found on e.g Leonardo uses a completely different bootloader, and we can't play around with fuses the same way.

I have a suggestion though. Messing with fuses and various bootloaders isn't what you really need if you're working with a ready-made development board such as Arduino UNO. It's when you're working on custom hardware with a hack-picked microcontroller you need it. Maybe we could make this script only available to only some targets? Let's say you're playing around with an Arduino UNO. That's completely fine, and you select `uno` as your board. Let's now say you want to play with fuses and different upload speeds. Now you're technically not using the UNO as it was intended too, so instead you can switch out `board = uno` with `board = ATmega328P`. Now all "low-level" features are available. Is this a good idea? It's close to impossible to support custom fuse bits for all AVR-based boards out there.

valeros, 2019-09-19:
We'd like to allow users to program bootloaders even for development boards (at least users should be able to burn the bootloaders bundled with Arduino IDE).
So to sum up:
- `-t fuses` Since we don't want allow users to mess up with fuses, this target is only available for `MiniCore`, `MegaCore`, `MightyCore` (values calculated in your script)
- `-t bootloader` available for all boards. Fuses and the path to bootloader will be computed dynamically for `MiniCore`, `MegaCore`, `MightyCore` . In case of development boards, values for fuses and path to bootloader binary will be specified in a special section, e.g. for `uno` board:
```
board_bootloader.file = optiboot/optiboot_atmega328.hex
board_bootloader.low_fuses = 0xFF
board_bootloader.high_fuses = 0xDE
board_bootloader.extended_fuses = 0xFD
board_bootloader.unlock_bits = 0x3F
board_bootloader.lock_bits = 0x0F
```
- bootloader programming will be a single command that consists of two steps  `fuses` and `bootloader` that are executed sequentially.

Does this process sound good to you?  Also, are values for `unlock_bits` and `lock_bits` the same for your cores? Thanks!

MCUdude, 2019-09-19:
Sounds good to me! unlock_bits and lock_bits are the same for my cores as well.

A few other things:
* How would you handle devices that have no efuse? In this case, it's ATmega8, ATmega16, ATmega32, ATmega8515, and ATmega8535.
* Can we manually override the fuses for special scenarios? This may be useful to some (for instance enabling JTAG). My original script supported this. But it has to be all or none. You can't just specify one fuse. If you want to override, you have to override all. 

** To sum it up:**
So my cores (only) will support both the `fuses` and `bootloader` commands. And the correct Optiboot_flash hex file will be loaded based on the parameters from platformio.ini.

On "board" targets (uno, mega etc) the `fuses` command is not available. It is possible to burn bootloader, but this will load the default fuses and the default bootloader specified in the corresponding manifest file. Fuses will for now only be calculated if the target "belongs" to MightyCore, MiniCore, MegaCore (and perhaps [MajorCore](https://github.com/MCUdude/MajorCore) in the future?)

valeros, 2019-09-19:
I've updated scripts for fuses and bootloader. `efuse` is now optional (maybe other fuses should be also?). Here are some configuration examples:
```
; Only program fuses (MiniCore family)
[env: ATmega328P_with_custom_fuses]
platform = atmelavr
framework = arduino
board = ATmega328P
upload_protocol = stk500v1
upload_speed = 19200
; Custom fuses
board_fuses.low_fuses = 0xAA
board_fuses.high_fuses = 0xBB
board_fuses.extended_fuses = 0xCC

; Fuses and bootloader selected dynamically
[env: ATmega328P_dynamic_boot_fuses]
platform = atmelavr
framework = arduino
board = ATmega328P
upload_protocol = stk500v1
upload_speed = 19200

[env: uno_dev_board_with_custom_boot]
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_speed = 19200
; Bootloader configuration (bootloader image from framework package)
board_bootloader.file = optiboot/optiboot_atmega328.hex
board_bootloader.low_fuses = 0xAA
board_bootloader.high_fuses = 0xBB
board_bootloader.extended_fuses = 0xCC
```
At the moment, the logic assumes that your bootloaders are located here `framework-arduinoavr\bootloaders\optiboot_flash`

Am I missing something critical? Would be happy to hear your thoughts on this implementation. Thanks!

MCUdude, 2019-09-19:
>  `efuse` is now optional (maybe other fuses should be also?).

_All_ AVRs have a high fuse and a low fuse (well, except the new megaavr-0 series, such as ATmega4809, but that's another story 😉), so no need to make other exceptions.

I'd like to test it to see if the generated fuse values are correct. Is there an easy way to install this "version" of pio core? And how do I run the script? With `pio run -t fuses` and `pio run -t bootloader`?

MCUdude, 2019-09-19:
I think I managed to install this version. When I run `pio run -t fuses` I get the following error:

```
Scanning dependencies...
No dependencies
Target configuration:
Clock speed = 16000000L, Oscillator = external, BOD level = 2.7v, UART port = uart0, Save EEPROM = yes
Selected fuses: [lfuse = 0xf7, hfuse = 0xd4, efuse = 0xfd]
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [fuses] Explicit exit, status 1
```

upload_port is not needed when setting fuses and burning bootloader.

MCUdude, 2019-09-19:
Something isn't right. The script is able to get the specified F_CPU, but not `hardware.uart`
`board_upload.speed`, `hardware.oscillator`, `hardware.bod` and `hardware.eesave`.

No matter what I type in platformio.ini, is always uses the default values defined in the script.

valeros, 2019-09-19:
> Something isn't right. The script is able to get the specified F_CPU, but not hardware.uart
board_upload.speed, hardware.oscillator, hardware.bod and hardware.eesave

Strange, but I can't reproduce this behavior, this config works for me:
```
board_hardware.bod = 3.5V
board_hardware.uart = UART1
board_hardware.eesave = no
```
> upload_port is not needed when setting fuses and burning bootloader.

Wow, that's interesting, I just tried to reproduce the same logic as in Arduino IDE.
How does avrdude decide what device to program? What happens if two boards are connected simultaneously?

MCUdude, 2019-09-19:
aha! I forgot the `board_` prefix. It does work now!

> Wow, that's interesting, I just tried to reproduce the same logic as in Arduino IDE.
How does avrdude decide what device to program? What happens if two boards are connected simultaneously?

If you want to use a programmer that shows up as a serial port, you'll have to specify it under `upload_flags` otherwise, Avrdude just searches for a specific USB VID and PID. In my case it's the USBasp.

MCUdude, 2019-09-19:
Actually, this has been a problem before too. Not long ago I reported issue #158, and @ivankravets provided a fix. Maybe he can provide an elegant solution to this as well?

valeros, 2019-09-19:
Thanks, I removed the dependency on upload port. Does it work better now?

valeros, 2019-09-19:
I see where the issue is. What flags should be enabled by default? Will it be OK if we leave only `-C avrdude.conf -c protocol -p target` and everything else user'll need to specify in `upload_flags` (including `-e`)?
```
upload_flags = -PCOM5 -b19200 -e
```

MCUdude, 2019-09-19:
For reference, [here is the avrdude command](https://github.com/MCUdude/MightyCore/blob/ae98efb6dca23ab56fa3e154cde9317cae658064/avr/platform.txt#L117) used when setting fuses with MightyCore in Arduino IDE. I suggest having `-C{avrdude.conf}  -p{target} -c{protocol}`, and moving all extra flags to `upload_flags`. Is it OK for you if the fuses command doesn't have -e, while the bootloader command has it?

[42 more replies]

closed 2019-11-25 by ivankravets