platformio/platform-atmelavr #158: "Please specify `upload_port`" error even though I'm uploading using USBasp

MCUdude, 2019-08-12:
Hi!

I go this strange error and AFAIK I've never seen this before.
I'm trying to upload code to an ATmega324PB (MightyCore) and have the following platformio.ini file

```ini
; ENVIRONMENT SETTINGS
[env:MightyCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = ATmega324PB
; Clock frequency in [Hz]
board_build.f_cpu = 8000000L

; BUILD OPTIONS
; Current pinout
board_build.variant = mightycore_standard
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto

; UPLOAD SETTINGS
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = COM4
; Upload baud rate
;board_upload.speed = 57600

; Upload using programmer
upload_protocol = usbasp
; Aditional upload flags
upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = COM4
; Serial monitor baud rate
;monitor_speed = 250000
```

For uploading, I choose "Upload using Programmer" under env:MightyCore. After a short while I get this error:

```
Linking .pio\build\MightyCore\firmware.elf
Checking size .pio\build\MightyCore\firmware.elf
Building .pio\build\MightyCore\firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  54.0% (used 1105 bytes from 2048 bytes)
PROGRAM: [=====     ]  51.7% (used 16676 bytes from 32256 bytes)
Configuring upload protocol...
AVAILABLE: usbasp
CURRENT: upload_protocol = usbasp
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [program] Explicit exit, status 1
============================================================================================================ [ERROR] Took 5.72 seconds ============================================================================================================
The terminal process terminated with exit code: 1
```

Why do I have to specify `upload_port` when I'm using a USBasp programmer?

MCUdude, 2019-08-20:
@ivankravets maybe this issue should be moved to the [platformio-core repo](https://github.com/platformio/platformio-core) instead? Not sure if this is AVR related or not.

ivankravets, 2019-08-20:
Do use Program target?

MCUdude, 2019-08-20:
The output seen in my first post is when using "Upload using Programmer". When trying to upload by just clicking "Upload" I get this error:

```
Checking size .pio\build\MightyCore\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.2% (used 522 bytes from 16384 bytes)
PROGRAM: [          ]   3.4% (used 4392 bytes from 130048 bytes)
Configuring upload protocol...
AVAILABLE: usbasp
CURRENT: upload_protocol = usbasp
Looking for upload port...
Uploading .pio\build\MightyCore\firmware.hex

avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

*** [upload] Error 1
============================================================================================================= [FAILED] Took 16.41 seconds ============================================================================================================= 
The terminal process terminated with exit code: 1
```

So it seems like "Upload" accept a missing serial port, while "Upload using programmer" don't.

MCUdude, 2019-08-20:
Don't worry about the avrdude error above. I didn't have any hardware connected to my USBasp programmer, but that's not the issue 😉

ivankravets, 2019-08-20:
Wow, this is a bug! :( I've just fixed it. Please inform me when new releases of your cores will be available. We will do joined release.

closed 2019-08-20 by ivankravets

MCUdude, 2019-08-20:
Awesome! PlatformIO is getting better every day 👍 Thank you for being so available and spending time helping out users like me!

> Please inform me when new releases of your cores will be available.

AFAIK only MightyCore is affected by the SPI1 related bug we've discussed earlier. I'll keep you informed! It won't be long, promise 🙂

tablatronix, 2019-11-18:
How do I pull this change in, it is not yet release right?

tablatronix, 2019-11-18:
I think I found it
https://docs.platformio.org/en/latest/platforms/atmelavr.html#upstream

WiRail, 2020-10-26:
I'm trying to update my code, using PlatformIO, on ATMEGA 1284p, using arduinoISP and always gives me  this error:

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1

; ENVIRONMENT SETTINGS
[env:MightyCore]
platform = atmelavr
framework = arduino
lib_deps = arduino-libraries/LiquidCrystal@^1.0.7
; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = ATmega1284P
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; HARDWARE SETTINGS
; Oscillator option
board_hardware.oscillator = external
; Hardware UART for serial upload
board_hardware.uart = uart0
; Brown-out detection
board_hardware.bod = 1.8v
; EEPROM retain
board_hardware.eesave = yes

; UPLOAD SETTINGS
board_upload.speed = 9600
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = /dev/cu.usbserial*
upload_port = COM[50]
; BUILD OPTIONS
; Current pinout
board_build.variant = standard
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; Upload using programmer
upload_protocol = arduinoisp
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600

nogueira, 2021-05-12:
I still have the same problem.

nogueira, 2021-05-12:
it seems Platformio doesn't accept port over COM10, I was using COM11 and after I changed to COM5 it works

demostenes63, 2021-09-09:
That´s right. I changed COM13 to COM2 and it works!

HubKing, 2021-09-14:
> 
> 
> it seems Platformio doesn't accept port over COM10, I was using COM11 and after I changed to COM5 it works

Damn, this is so stupid... The COM port automatically kept increasing as I changed the port, and when it became COM11, PlatformIO did not recognise it. I opened Kitty and connected to COM11 and I could see the serial output so the port was working. After reading your comment, I changed the COM to a lower number like 5 (I had to remove all the "hidden" devices in the Device Manager), unplugged/re-plugged the cable, and now PlatformIO recognised the port.

ivankravets, 2021-09-14:
@HubKing , @demostenes63 ,

Could you revert back port to COM13 and try to run [pio device list](https://docs.platformio.org/en/latest/core/userguide/device/cmd_list.html)?

See how to open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli).

saintofinternet, 2021-11-20:
**My Env. settings are as follows** 

[env:168pa16m]
platform = atmelavr
board = pro8MHzatmega328
framework = arduino
lib_deps = locoduino/MemoryUsage@^2.21.1
upload_protocol = usbasp

**I am getting the following error after upload via USBASP programmer to an Arudino Pro Mini** 

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x20a9
         0x21 != 0xf1
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FD, H:DA, L:FF)

avrdude done.  Thank you.

*** [upload] Error 1

commit 003c769b (2019-08-20): Fix broken "program" target which requires "upload_port" // Resolve #158