platformio/platform-atmelavr #164: Cannot upload to Sparkfun arduino pro micro

MagixInTheAir, 2019-09-03:
Here is the full verbose upload log : https://pastebin.com/zgVdAzFf

It is a simple "LED blink" demonstration program. platformio.ini is as follows : 

    [env:sparkfun_promicro16]
    platform = atmelavr
    board = sparkfun_promicro16
    framework = arduino
    upload_port = /dev/ttyACM0

The board is detected in the device panel before trying to flash. After flashing, I have to physically unplug and then replug the board for it to be detected again. The result is the same with port auto-detection enabled and when the port is forced.

pfeerick, 2019-09-05:
Looks like [a ModemManager issue](https://community.platformio.org/t/pro-micro-ubuntu-errors/9378/4).

Long story short,[ you should be able to follow the FAQ guide about installing the udev rules](http://docs.platformio.org/en/latest/faq.html#platformio-udev-rules), but instead of the link given in the documentation, use the latest version that's in the develop branch, as that includes the rule to stop ModemManager poking it's nose where it's not wanted. It'll be available in the next release version. 

`curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/scripts/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules`

MagixInTheAir, 2019-09-05:
I had that ModemManager issue but probably used an old udev rule, will try thr solution ASAP. Anything could be done to make it work by default without needing to mess with udev rules ?

pfeerick, 2019-09-06:
For modem manager stuff, udev rules are the best solution. 

If you want to take it out of the equation by temporarily stoping it using `sudo systemctl stop ModemManager.service`, and if you don't need it at all (i.e. you don't use a 4G modem) disabling it from automatically starting at all using `sudo systemctl disable ModemManager.service`.

If you're using a clone or board that doesn't have a reset button, look for the RST pin, and tap that to GND to simulate the button press. A double press triggers the bootloader to briefly wait for programming.

MagixInTheAir, 2019-09-07:
I disabled the ModemManager and still cannot make it work, error message is the exact same. I can't do electronics right now so I can't check the reset solution. What log could I check to find more about the error ?

MagixInTheAir, 2019-09-18:
Bump : tried it under windows too, same problem. This doesn't seem to be an issue related to the Linux side of things.

pfeerick, 2019-09-19:
Did you have any luck with trying the double reset before programming? The other thing that might be worth trying if you have an ISCP programmer (or use the ArduinoasISP sketch on another arduino), is to replace the bootloader with the current Arduino Micro bootloader... It partly depends on if you can get the board to program normally with the Arduino IDE, but not PlatformIO... if that's the case, the PlatformIO upload code must be behaving slightly differently.