platformio/platform-atmelavr #165: Is there any plan to create a default, built-in target `burn_bootloader`?

doronbehar, 2019-09-04:
* https://community.platformio.org/t/new-target-to-upload-bootloader/908

I was wondering, after reading some posts on the forum about whether or not it is possible or not to use PlatformIO to install a bootloader (sometimes referred to as *firmware*) on Arduino boards, and since many users of the Arduino framework are used to the IDE and they don't have a choice but to use the IDE for these operations (using all kinds of `upload_tool`s), I was wondering: What if PlatformIO could replace Arduino's IDE even for this operation? Here are the links to the forum posts I found relevant:

1. https://community.platformio.org/t/is-it-possible-to-use-platformio-to-install-arduino-firmware/159/2
2. https://community.platformio.org/t/new-target-to-upload-bootloader/908
3. https://community.platformio.org/t/burn-bootloader-to-attiny85-via-arduinoasisp/662/16

It seems to me that although it is widely considered a '*one-time operation*' (according to [@ivankravets](https://community.platformio.org/u/ivankravets)'s [replay](https://community.platformio.org/t/new-target-to-upload-bootloader/908/2)), and therefor it may be considered that it "doesn't worth the effort", IMHO PlatformIO should have it as a default target, like `fuses`.

I've read the following solutions: [this](https://community.platformio.org/t/burn-bootloader-to-attiny85-via-arduinoasisp/662/9) and [this](https://community.platformio.org/t/new-target-to-upload-bootloader/908/3) and in the meantime they didn't work for me. Perhaps they worked for the latest version of PlatformIO at the time these replays were posted and perhaps I made a mistake when trying to put them together on my machine but that's not the point. Even if these solutions should work and they are good, they are not totally modular and generic for all projects. It's not the same experience as with using a built in target (like `fuses`).

I haven't posted this feature request/question on https://github.com/platformio/platform-atmelavr or https://github.com/platformio/platform-atmelsam because it is related to both of these platforms. Additionally, there may be other frameworks I don't know of, which may include a bootloader (or firmware) by default and I think PlatformIO should have a ready to use target like `burn_bootloader` for them as well.

What do you think?

pfeerick, 2020-01-09:
This was implemented in https://github.com/platformio/platform-atmelavr/releases/tag/v2.0.0

ivankravets, 2020-02-05:
https://github.com/platformio/platform-atmelavr/releases/tag/v2.0.0

closed 2020-02-05 by ivankravets