platformio/platform-atmelavr #169: Support for Industrial Shields Boards Ardbox/M-duino

Klasskampen, 2019-10-02:
Support for the Ardbox and M-duino boards from Industrial Shields would be appreciated.
[Industrial Shields web page](https://www.industrialshields.com/)

kevinboards, 2019-12-28:
Hey, looking to get the Industrial Shields boards going on platformio. I'm not sure where to start.
Based on Arduino Leonardo and Mega2560, also some references to esp32 depending on the board. 
Their website : https://www.industrialshields.com/
Their board reference for arduino IDE  http://apps.industrialshields.com/main/arduino/boards/package_industrialshields_index.json

The package that comes with this link is quite intricate and i'm not sure where to put all of the packages in the platformio folders.   Let me know if i can provide any extra information.  
There are 2 "types" of industrial shields platform M-duino (mega 2560) and Ardbox (leonardo), and each type has a few different board variations.

Jarovit, 2020-01-22:
As far as I know, only Arduino IDE is supported by Industrial Shields at the moment.

kevinboards, 2020-01-22:
> As far as I know, only Arduino IDE is supported by Industrial Shields at the moment.

That's right. I'm trying to figure out how to help in getting industrial shields  included in the platformio ide.

FlowlessRhetoric, 2020-03-03:
> > As far as I know, only Arduino IDE is supported by Industrial Shields at the moment.
> 
> That's right. I'm trying to figure out how to help in getting industrial shields included in the platformio ide.

If I may ask: any luck so far with that? I'm working with several and doing that through Platformio would be awesome.

kevinboards, 2020-03-03:
Unfortunately I have had no support. it turns out to be easier to just use platformio and just call the Industrial Shields what they really are (mega or uno)  Then I have a reference sheet between all the pins I got from the industrial shields website to reference the I/O to Pin numbers.  

I'm also not sure Industrial Shields is a real company . I can't seem to get any support from them, their courses don't actually exist, and the website and forum seem to be quite stale.  

I like the product and will keep using it but its a bit of a shame.  

If there were more Cat Memes involved in this issue/request it would be solved by now.

macpelos, 2020-06-15:
Not tested yet, but I found this: https://github.com/dgomes/industrial-shields that could maybe help

manusreload, 2020-09-01:
You can check the pins about every single board in https://www.industrialshields.com/es_ES/support/help

kevinboards, 2020-09-01:
The industrial shields boards are supposed to mimic PLC pin logic.   The idea is that instead of pins 1- 55  you have pin names that tell you what it performs  IO1 (digital input or output 1)  Q1 (output 1)  AQ 1 (analog output 1)   These boards have exceptional hardware built on them for circuit protection so going back and forth to the pinout sheets could be simplified if we could get the industrial shields board profiles into Platformio.

Bart-van-Ingen, 2023-04-04:
i know that this is a slightly old thread, but i will place it none the less for people that are currently looking to use the industrialshield arduino/esp32 in platformio. 
Together with Max Gerhardt, I was able to get a proof of concept to get the esp32 r19 plc to work with platformio! it still needs to get merged into the mainline branches of platformio, but for the rest it works just aswell as writing code in the arduino IDE.

The thread can be found here : [https://community.platformio.org/t/adding-industrialshields-support-from-arduino-ide-to-platformio/32808](https://community.platformio.org/t/adding-industrialshields-support-from-arduino-ide-to-platformio/32808 )

while this is currently only setup for the ESP32 boards, i believe it to be fairly trivial to get this to work for the atmelavr aswell!