platformio/platform-atmelavr #170: Support TTGO XI 8F328 dev board.  AVR on steroids.

exothink, 2019-10-02:
I would like PIO support for this board.

Kashouryo, 2019-10-25:
LGT8F328P is not a standardised AVR micro controller; it's more like a clone. Here is the unofficial Arduino support file https://github.com/LGTMCU/Larduino_HSP for the LGT8F328P.

MCUdude, 2020-05-15:
A while ago I got myself an 8F328P and 8F328D board because I was looking into adding support for them in [MiniCore](https://github.com/MCUdude/MiniCore). After some research and testing, I figured it just wasn't worth it. Here's why:
* Documentation is spare. The official datasheet only exists in Chinese. However, some enthusiasts have translated it into English, but you get my point.
* They are not direct clones, but rather variants. They have a different kind of memory mapping IIRC, where the EEPROM behaved differently. It doesn't have the same style of fuses that we're used to on regular AVRs.
* The programming interface isn't "standard" ISP, but rather some weird protocol. If you need to replace the bootloader you can't, because the Logic Green programmer is unobtainable, at least if you live outside China.
* You can't buy the bare chips!! All that's available on the internet we have access to is these development boards. This and the previous point is definitely the biggest deal-breaker for me.