platformio/platform-atmelavr #171: VSCode Extension compile error: incorrect attiny85 frequency

dontsovcmc, 2019-10-23:
- [ ] **PlatformIO IDE extension for VSCode**.

### Configuration

**Operating system**:
MacOS 

**PlatformIO Version** (`platformio --version`):
4.0.3 

### Description of problem

Failed to build project after update.. 

#### Steps to Reproduce

1. Open https://github.com/dontsovcmc/waterius/tree/master/Attiny85 in VSCode
2. Compile

### Actual Results
```
> Executing task: platformio run <
Processing attiny85 (platform: https://github.com/platformio/platform-atmelavr.git#v1.15.0; board: attiny85; framework: arduino; board_build.f_cpu: 1000000L  ;частота на которой работает микроконтроллер attiny85; upload_port: COM5; upload_protocol: usbasp; upload_flags: -Pusb, -B5; build_flags: -Wall)
-------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-avrdude @ ~1.60300.0
Downloading  [####################################]  100%
tool-avrdude @ 1.60300.190628 has been successfully installed!
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny85.html
PLATFORM: Atmel AVR 1.15.0 #299ca85 (git+https://github.com/platformio/platform-atmelavr.git#v1.15.0) > Generic ATtiny85
HARDWARE: ATTINY85 100MHz, 512B RAM, 8KB Flash

Processing attiny85 (platform: https://github.com/platformio/platform-atmelavr.git#v1.15.0; board: attiny85; framework: arduino)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny85.html
PLATFORM: Atmel AVR 1.15.0 #299ca85 > Generic ATtiny85
HARDWARE: ATTINY85 100MHz, 512B RAM, 8KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <EEPROM> 2.0
Compiling .pio/build/attiny85/src/Power.cpp.o
Compiling .pio/build/attiny85/src/SlaveI2C.cpp.o
Compiling .pio/build/attiny85/src/Storage.cpp.o
Compiling .pio/build/attiny85/src/main.cpp.o
<command-line>:0:16: error: token ";" is not valid in preprocessor expressions
/Users/dontsov/.platformio/packages/framework-arduinoavr/cores/tiny/Arduino.h:66:5: note: in expansion of macro 'F_CPU'
 #if F_CPU < 1000000L
     ^
<command-line>:0:16: error: token ";" is not valid in preprocessor expressions
/Users/dontsov/.platformio/packages/framework-arduinoavr/cores/tiny/Arduino.h:66:5: note: in expansion of macro 'F_CPU'
 #if F_CPU < 1000000L
     ^
<command-line>:0:16: error: token ";" is not valid in preprocessor expressions
/Users/dontsov/.platformio/packages/framework-arduinoavr/cores/tiny/Arduino.h:66:5: note: in expansion of macro 'F_CPU'
 #if F_CPU < 1000000L
     ^
<command-line>:0:16: error: token ";" is not valid in preprocessor expressions
/Users/dontsov/.platformio/packages/framework-arduinoavr/cores/tiny/Arduino.h:66:5: note: in expansion of macro 'F_CPU'
 #if F_CPU < 1000000L
     ^
Compiling .pio/build/attiny85/lib9ec/Wire/USI_TWI_Master/USI_TWI_Master.c.o
In file included from src/Storage.h:5:0,
                 from src/SlaveI2C.cpp:3:
/Users/dontsov/.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny/EEPROM/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^
Compiling .pio/build/attiny85/lib9ec/Wire/Wire.cpp.o
Compiling .pio/build/attiny85/lib9ec/Wire/WireS.cpp.o
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h: In function '_delay_ms':
<command-line>:0:16: error: expected ')' before ';' token
<command-line>:0:16: error: stray '\321' in program
<command-line>:0:16: error: stray '\207' in program
<command-line>:0:16: error: stray '\320' in program
<command-line>:0:16: error: stray '\260' in program
<command-line>:0:16: error: stray '\321' in program
<command-line>:0:16: error: stray '\201' in program
<command-line>:0:16: error: stray '\321' in program
<command-line>:0:16: error: stray '\202' in program
<command-line>:0:16: error: stray '\320' in program
<command-line>:0:16: error: stray '\276' in program
<command-line>:0:16: error: stray '\321' in program
<command-line>:0:16: error: stray '\202' in program
<command-line>:0:16: error: stray '\320' in program
<command-line>:0:16: error: stray '\260' in program
Compiling .pio/build/attiny85/lib9ec/Wire/twi.c.o
<command-line>:0:16: error: stray '\320' in program
........
<command-line>:0:16: error: stray '\200' in program
In file included from /Users/dontsov/.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny/Wire/src/USI_TWI_Master/USI_TWI_Master.h:35:0,
                 from /Users/dontsov/.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny/Wire/src/Wire.cpp:339:
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h:174:2: error: stray '\321' in program
  __tmp = ((F_CPU) / 1e3) * __ms;
  ^
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h:174:2: error: stray '\207' in program
....
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h:263:2: error: stray '\321' in program
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h: In function 'void _delay_ms(double)':
<command-line>:0:16: error: expected ')' before ';' token
<command-line>:0:16: error: expected ')' before ';' token
<command-line>:0:100: error: 'attiny85' was not declared in this scope
/Users/dontsov/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h: In function 'void _delay_us(double)':
<command-line>:0:16: error: expected ')' before ';' token
<command-line>:0:16: error: expected ')' before ';' token
<command-line>:0:100: error: 'attiny85' was not declared in this scope
*** [.pio/build/attiny85/lib9ec/Wire/USI_TWI_Master/USI_TWI_Master.c.o] Error 1
*** [.pio/build/attiny85/lib9ec/Wire/Wire.cpp.o] Error 1
<command-line>:0:16: error: token ";" is not valid in preprocessor expressions
/Users/dontsov/.platformio/packages/framework-arduinoavr/cores/tiny/Arduino.h:66:5: note: in expansion of macro 'F_CPU'
 #if F_CPU < 1000000L
     ^
*** [.pio/build/attiny85/lib9ec/Wire/twi.c.o] Error 1
============================================= [FAILED] Took 1.38 seconds =============================================
Процесс терминала завершен с кодом выхода: 1

Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.

```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env:attiny85]
platform = https://github.com/platformio/platform-atmelavr.git#v1.15.0
board = attiny85
framework = arduino
board_build.f_cpu  = 1000000L 
```

### Additional info
platformio from command line works good

Remove VSCode, remove /Users/dontsov/.platformio, install VSCode, Platformio Extension don't help.

pfeerick, 2019-11-12:
Sorry, can't reproduce that with either the code from that repo from either the 21st of Oct, or the current version of it, with either the stock or the platformio.ini configuration you supplied. Is this still not working for you @dontsovcmc? The first error `<command-line>:0:16: error: token ";" is not valid in preprocessor expressions` is hinting that there is a `;` where it shouldn't be, which is probably cascading through, causing all the subsequent errors. 

```
Processing attiny85 (platform: https://github.com/platformio/platform-atmelavr.git#v1.15.0; board: attiny85; framework: arduino)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny85.html
PLATFORM: Atmel AVR 1.15.0 #299ca85 > Generic ATtiny85
HARDWARE: ATTINY85 1MHz, 512B RAM, 8KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <EEPROM> 2.0
Building in release mode
Compiling .pio/build/attiny85/src/Power.cpp.o
Compiling .pio/build/attiny85/src/SlaveI2C.cpp.o
Compiling .pio/build/attiny85/src/Storage.cpp.o
Compiling .pio/build/attiny85/src/main.cpp.o
In file included from src/Storage.h:5:0,
                 from src/SlaveI2C.cpp:3:
/home/pfeerick/.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny/EEPROM/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^
Compiling .pio/build/attiny85/lib88b/Wire/USIWire.cpp.o
Compiling .pio/build/attiny85/lib88b/Wire/USI_TWI_Master/USI_TWI_Master.c.o
Compiling .pio/build/attiny85/lib88b/Wire/USI_TWI_Slave/USI_TWI_Slave.c.o
Compiling .pio/build/attiny85/lib88b/Wire/Wire.cpp.o
Compiling .pio/build/attiny85/lib88b/Wire/WireS.cpp.o
Compiling .pio/build/attiny85/lib88b/Wire/twi.c.o
Archiving .pio/build/attiny85/lib485/libEEPROM.a
Archiving .pio/build/attiny85/libFrameworkArduinoVariant.a
Indexing .pio/build/attiny85/lib485/libEEPROM.a
Indexing .pio/build/attiny85/libFrameworkArduinoVariant.a
Compiling .pio/build/attiny85/FrameworkArduino/HardwareSerial.cpp.o
Archiving .pio/build/attiny85/lib88b/libWire.a
Compiling .pio/build/attiny85/FrameworkArduino/Print.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/Stream.cpp.o
Indexing .pio/build/attiny85/lib88b/libWire.a
Compiling .pio/build/attiny85/FrameworkArduino/TinySoftwareSerial.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/WInterrupts.c.o
Compiling .pio/build/attiny85/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/WString.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/abi.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/main.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/new.cpp.o
Compiling .pio/build/attiny85/FrameworkArduino/wiring.c.o
Compiling .pio/build/attiny85/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/attiny85/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/attiny85/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/attiny85/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/attiny85/libFrameworkArduino.a
Indexing .pio/build/attiny85/libFrameworkArduino.a
Linking .pio/build/attiny85/firmware.elf
Checking size .pio/build/attiny85/firmware.elf
Building .pio/build/attiny85/firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=======   ]  73.6% (used 377 bytes from 512 bytes)
PROGRAM: [=====     ]  48.5% (used 3974 bytes from 8192 bytes)
=========================================== [SUCCESS] Took 2.39 seconds ===========================================
```

dontsovcmc, 2019-11-12:
@pfeerick thanks for remember!
It was this error: https://github.com/platformio/platformio-core/issues/3213
I wrote incorrect platformio.ini file in issuu: without comments which were a reason of error.

closed 2019-11-12 by dontsovcmc

pfeerick, 2019-11-12:
Oh, nasty one... I remember seeing that issue... nice to know it's fixed! Thanks! :)