platformio/platform-atmelavr #178: Difficulties when burning bootloader with Arduino as ISP (Mini/Mighty/MegaCore)

MCUdude, 2019-11-28:
See [MiniCore #115](https://github.com/MCUdude/MiniCore/issues/115) for more details.

It's impossible to set the upload baud rate for the bootloader when burning bootloader with Arduino as ISP. this is because ArduinoISP relies on `upload_speed = 19200` to work. However, when this is specified,  a bootloader with a baud rate of 19200 is always selected, no matter what `board_upload.speed` is defined to.

* `board_upload.speed` should be used when doing "regular" uploads using `-carduino` as upload protocol (using a bootloader).
* `upload_speed` should be used when the user is uploading using a programmer to upload, or setting the fuses/burning bootloader.

This is only an issue with programmers that uses a serial port and a baud rate to communicate with Avrdude. @ivankravets @valeros I guess it should be an easy fix.

valeros, 2020-05-06:
Sorry for the late reply.
Basically, `board_upload.speed` and `upload_speed` represent the same thing - the baudrate which uploader uses for sending firmware to a board. As I understand the problem, we cannot 
select a bootloader image since `upload_speed` is used as part of the bootloader image name.
What do you think if we introduce a new field to bootloader section, e.g.:
```
board_bootloader.speed = 115200
```
that can be used for choosing the right bootloader image?
If it's not specified then we will fall back to `upload_speed` value.

MCUdude, 2020-05-06:
> Sorry for the late reply.

No worries! I do understand that you have a lot to do. You should know that I really admire this project and all the work you and Ivan have put into this project 🎉

Maybe we could combine this with other improvements as well. Here's another thing that's a little annoying.

 Let's say I've created a project where I use a USB to serial adapter for uploading. However, if I want to bootload the board I've connected, I'll have to add my programmer to platformio.ini as well. Just like this:

```ini
; Upload using programmer
upload_protocol = usbtiny
; Aditional upload flags
upload_flags = -Pusb
```

I can now set fuses/burn bootloader. However, if I want to upload using my USB to serial adapter, I will have to uncomment these two lines to prevent them to "interfere" with regular UART uploads.

it would be great if we would use the `board_hardware.uart` field to determine if the standard upload command (I'm using the ➡ arrow button in VSCode) would trigger a UART upload or a programmer upload. If so, one could be able to do both without having to uncomment lines in the ini file. What do you think?

> As I understand the problem, we cannot
select a bootloader image since upload_speed is used as part of the bootloader image name.

That is correct. I like your idea, but could you make work no matter what is present in the ini file, like so?

```
ONLY board_upload.speed PRESENT
Use value for both bootloader and upload speed

---------------------------------------------------

ONLY board_bootloader.speed PRESENT
Use value for both bootloader and upload speed

---------------------------------------------------

BOTH PRESENT
Use board_bootloader.speed for bootloader speed and
board_upload.speed for upload speed
```

valeros, 2020-05-07:
> it would be great if we would use the board_hardware.uart field to determine if the standard upload command (I'm using the arrow_right arrow button in VSCode) would trigger a UART upload or a programmer upload. If so, one could be able to do both without having to uncomment lines in the ini file. What do you think?

Not sure about this, `board_hardware.uart` option is mainly used with your cores and `upload_flags` option might be used for UART upload as well. Besides, in most cases burning fuses/bootloader is one-time operation, right?

> ONLY board_bootloader.speed PRESENT
Use value for both bootloader and upload speed

In the most cases, it's not an option since `board_upload.speed` is always present (set by a user or default value from a board manifest).

MCUdude, 2020-05-07:
> Not sure about this, board_hardware.uart option is mainly used with your cores and upload_flags option might be used for UART upload as well. Besides, in most cases burning fuses/bootloader is one-time operation, right?

OK, at least my idea is not thought through yet. We can leave this for now and focus on the other thing.

> In the most cases, it's not an option since board_upload.speed is always present (set by a user or default value from a board manifest).

Yes, that is right, sorry. Then I think we agree. Use `board_bootloader.speed` is present to get the correct bootloader file. If not, use `board_upload.speed`. OK?

closed 2020-05-08 by valeros

MCUdude, 2020-05-08:
Thank you! Will give this a try very soon. I'll report back if I find any issues 🙂

commit 0477b234 (2020-05-08): Take into account new option for bootloader speed // Resolve #178
This will allow programming bootloaders with various baudrates