platformio/platform-atmelavr #192: Update MCUdude AVR cores

MCUdude, 2020-05-06:
Hi!

I've done some improvements to my AVR Arduino cores and would like the latest releases to be available through PlatformIO. I have also another Arduino core that I'd like to add. It's more or less identical to one I already have, and the targets (ATmega162 and ATmega8515) are even supported by the fuses and bootloader scripts.

The latest [MightyCore release is v2.0.5](https://github.com/MCUdude/MightyCore/releases/tag/v2.0.5)  
The latest [MiniCore release is v2.0.5](https://github.com/MCUdude/MiniCore/releases/tag/v2.0.5)  
The latest [MegaCore release is v2.0.3](https://github.com/MCUdude/MegaCore/releases/tag/v2.0.3)

@ivankravets Is there an easy way for me to just provide a PR to update these cores?

The new core I'd like to add to PlatformIO is [MajorCore](https://github.com/MCUdude/MajorCore), and the latest release for it is [v2.0.2](https://github.com/MCUdude/MajorCore/releases/tag/v2.0.2). Is there anything I can to to help out adding this core to PlatformIO?

valeros, 2020-05-06:
Hi @MCUdude ! Do we need additional entries for `ATmega162` and `ATmega8515` when calculating dynamic fuses [here](https://github.com/platformio/platform-atmelavr/blob/develop/builder/fuses.py#L10-L20) ?

Thanks!

closed 2020-05-06 by valeros

MCUdude, 2020-05-06:
> Hi @MCUdude ! Do we need additional entries for ATmega162 and ATmega8515 when calculating dynamic fuses here?

Nope, I took those into account when making the script 🙂

MCUdude, 2020-05-06:
@valeros I tried to create a new project from PlatformIO home in VSCode, but neither the ATmega8515 or ATmega162 shows up.

valeros, 2020-05-06:
The new version of the platform has not been released yet. Please try the [upstream version](https://docs.platformio.org/en/latest/platforms/atmelavr.html#stable-and-upstream-versions).

MCUdude, 2020-05-06:
Thanks.

Meanwhile I just tested the bootloader command for a board I had lying around that has an ATmega1284 in it. It seems like the script can't find the bootloader files:

```
$ pio run -t bootloader -v
Processing MightyCore (platform: atmelavr; framework: arduino; board: ATmega1284p; board_build.f_cpu: 16000000L; board_hardware.oscillator: external; board_hardware.uart: uart0; board_hardware.bod: 2.7v; board_hardware.eesave: yes; board_upload.speed: 115200; board_build.variant: standard; build_unflags: -flto; build_flags: ; upload_protocol: usbtiny; upload_flags: -Pusb; monitor_speed: 9600)
-------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284p.html
PLATFORM: Atmel AVR 2.0.0 > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
PACKAGES: 
 - framework-arduino-avr-mightycore 2.0.5 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Target configuration:
Target = atmega1284p, Clock speed = 16000000L, Oscillator = external, BOD level = 2.7v, UART port = uart0, Save EEPROM = yes
Selected fuses: [lfuse = 0xf7, hfuse = 0xd6, efuse = 0xfd]
Error: Couldn't find bootloader image
```

Note that the bootloader has gotten a new parameter in their name, the LED pin.

MCUdude, 2020-05-06:
Note that MiniCore, MegaCore and MajorCore has fixed bootloader LED pins, which means they will never change.

However, MightyCore (ATmega1284, 644, 324, 164, 32, 16, 8535) has the Bobuino pinout, which has the bootloader LED on a different pin to ensure maximum compatibility with the Arduino UNO.
The Bobuino pinout has its LED on pin B7 instead of B0.

MCUdude, 2020-05-06:
Update:
It turned out that the files I had locally weren't updated. The manifest files were missing the bootloader LED pin entry and the bootloader script was missing an argument. I guess you'll have to do a new release of the `platform-atmelavr` soon then.

However, there is still an issue. Is it possible to make sure that for all MightyCore compatible devices (ATmega1284, 644, 324, 164, 32, 16, 8535), if the `board_build.variant = bobuino` variant is selected, it selects led B7 instead?

valeros, 2020-05-06:
Could you please retest with the upstream version? `bobuino` variant now should use `B7`

MCUdude, 2020-05-06:
> Could you please retest with the upstream version?

Sorry, but I'm not sure how I do that. Do you have a `pio` command that I can execute?

valeros, 2020-05-06:
If you're already using the upstream version in your `platformio.ini`: 
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-atmelavr.git
board = ...
```
just run `platformio update` in the terminal window

MCUdude, 2020-05-06:
Of course! I've done this before, I've just forgotten it.

I just tried to burn bootloader to an ATmega1284P using the standard and bobuino pinout. Both work perfectly! Thank you so much!

MCUdude, 2020-05-06:
Oh, I discovered another tiny issue related to the fuses script. The ATmega162 has some weird efuse settings, and I actually forgot to add it to the efuse part initially. Its efuse is different from other AVRs, so it needs a dedicated entry. Would you be so nice and add this to the efuse part of fuses.py? 🙂 I've also tested this on actual hardware to make sure it works as it should.

```python
    # Add this to line 151
    mcus_6 = ("atmega162")

#....

    # Add this to line 203
    elif mcu in mcus_6:
        if bod == "4.3v":
            return 0xf9
        elif bod == "2.7v":
            return 0xfb
        elif bod == "1.8v":
            return 0xfd
        else:
            return 0xff
```

commit 1fc21c88 (2020-05-06): Update MCUdude AVR cores // Resolve #192