platformio/platform-atmelavr #196: Uploading to any ATMega 32u4 based board is still broken

Jkillelea, 2020-05-29:
The version of avrdude used by PlatformIO or some of the configuration must be wrong, because it's trivially easy to brick Leonardo and Pro Micro boards by just by uploading to them. I've done this with several different boards. The upload fails, and the board doesn't register as a USB device any more, and it takes an AVRISP to rescue it.

I'm unclear on why this happens, but it's a major issue since it has the potential to stop people who don't own separate programmers from using their boards afterwards, effectively destroying them.

MCUdude, 2020-06-03:
I've been able to "brick" my Leonardo several times through Arduino IDE too.

Can you post the complete Avrdude command that PlatformIO uses to upload? This would be the first step to figure out what's going on.

BTW how many times can you *usually* upload to it before the bootloader is toast and needs to be reloaded? One time? 20 times?

Sambo007, 2020-06-04:
Are you pushing the reset button right after you click upload? I remember having to do that. Search the net.

snodecoder, 2020-07-10:
I tried resetting my Pro Micro right after uploading, no effect. After many hours searching and trying proposed fixes, I ended up reinstalling VSCode and Platformio, no effect.
The workaround that finally worked for me was uploading my code via VSCode and Platformio through an Arduino Uno (configured as ISP).

GunZi200, 2020-07-16:
Here are my 2 cents. My setup is VScode 1.47.1, Windows 10, and am using Arduino Micro bootloader and configuration (Not MegaCore).

I've found on my custom ATmega32u4 board that "_Upload_" or "_Upload and Monitor_" can be unreliable for uploading a sketch, sometimes it works for me but most of times it doesn't.

What usually works for me is using "_Verbose Upload_". I don't know the details of why this would be different from the "_Upload_" button.

To open the Serial Monitor only the option "_Monitor_" ever works, "_Upload and Monitor_" never works for me.

If I can help any further with this issue, please let me know how. 👍

maxgerhardt, 2020-10-30:
Users are still reporting upload failures for Atmega32U4 (e.g. Leonardo) boards ([example](https://community.platformio.org/t/upload-error1-failure/16760/)). This has to be reinvestigated I think.

Fusseldieb, 2020-11-05:
Yep. Arduino Leonardo doesn't upload. Times out and requires manual reset while looking for the new upload port.

maxgerhardt, 2020-11-05:
Comment from [here](https://community.platformio.org/t/uploading-to-micro-leonardo/10935/20?u=maxgerhardt):

>I don’t know why exactly, but on SOME pcs it just waits far too long (until the Arduino goes out of the bootloader) and only THEN tries to upload it, causing it to avrdude: ser_open(): can’t open device “.\COM5”.
>
>A workaround is to manually reset the Arduino multiple times when it’s looking for the “New upload port” and if you do it in the right time, it recognizes it and uploads the firmware.
>
>It’s annoying and only happens on some PCs.

calumljohnston, 2020-11-11:
I have had this exact issue. Another workaround seems to be to use the "Verbose Upload" option from the side panel, which works perfectly for me.

[image]

calumljohnston, 2020-11-11:
I have had this issue and managed to resolve it by disabling the Bluetooth connection on my laptop, which was using two COM ports. With Bluetooth disabled the upload is as fast/faster than with the Arduino IDE.

Fusseldieb, 2020-11-11:
> I have had this issue and managed to resolve it by disabling the Bluetooth connection on my laptop, which was using two COM ports.

That's interesing! I definitively gonna check that and report back.