platformio/platform-atmelavr #207: Wrong flash sizes on 8KB flash AVR parts

ospilos, 2020-07-25:
There is an issue of incorretly staded flash size on bare AVR parts with 8KB of flash memory, namely:
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega8.json
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega88.json
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega88P.json
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega88PB.json

The value of "upload" -> "maximum_size": is 7680, while it should probably be 8192 as these are bare ICS, therefore without any bootloader in them.

On the other side 
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega48.json
seems to advertise the whole flash size in the json definitions.

In fact this is incosistent across most of the bare IC definitions, as ATMEGA parts mostly has the flash size liited, while ATTINY parts seem to advertise the whole flash size.

MCUdude, 2020-07-28:
By default, the PlatformIO/MiniCore implementation includes a 512-byte bootloader by default. If you don't want to use a bootloader you can just redefine the flash size in platformio.ini.

```ini
build_upload.maximum_size = 8192
```

The ATmega48 doesn't have a bootloader section. Thats why all 4096 bytes are "available" by default.

MCUdude, 2020-11-17:
@valeros would it be possible to automatically add 512 bytes to the `build_upload.maximum_size` somewhere if a bootloader is in use? I looked through the main.py file in this repo, but I couldn't find any references to this.

valeros, 2020-11-19:
I agree that all these manifests rather represent bare MCUs than full-fledged boards. IMO, the manifests should contain the precise memory size without any assumptions on bootloader presence. Anyone who want to use bootloader then should use the `build_upload.maximum_size` option.

MCUdude, 2020-11-20:
> Anyone who wants to use bootloader then should use the build_upload.maximum_size option.

@valeros is this something that could be handled automatically for bare MCUs that are supported by Mighty/Mega/Mini/Major/MicroCore? We know the bootloader sizes for the various targets (512B or 1024B). Mabe a field in the manifest file: `build_upload.bootloader_size = 512`?

valeros, 2020-11-24:
TBH, I'm a bit reluctant to pollute manifests with core-specific configurations. I guess users who know how to use bootloader should be aware of memory constraints it brings. Besides, how can we determine whether a target is using bootloader to reduce the memory size accordingly?

MCUdude, 2020-11-24:
For those who aren't using a bootloader, they are probably using the manifest files meant to be used with "my" Arduino cores. Since I've compiled all the bootloaders for all these boards, I know how big they are and how much space is allocated in every microcontroller.

Just for fun, the bootloader build scrip I'm using also generates a file that contains some information about every file, for example [atmega1284p_build_info.txt](https://github.com/MCUdude/MightyCore/blob/master/avr/bootloaders/optiboot_flash/bootloaders/atmega1284p/atmega1284p_build_info.txt). But what matters here is the actual allocated space set by the fuse bits.

Here's the rule.
* 1024 bytes are allocated if the microcontroller flash memory is greater or equal to 64kB
* 512 bytes are allocated if the microcontroller flash memory is less or equal to 32 kB.
* Exceptions:
  - ATmega48/P/PB does not have a bootloader section and can therefore not have a "regular" bootloader. Hence, 4096 bytes are available. This is true for all "classic"AVRs with 4kB or less.
  - AT90CAN32 has 32kB of flash, but 1024 bytes are allocated for the bootloader. This is because the compiled bootloader size is more than 512 bytes, and the next step is to allocate 1024 bytes

valeros, 2020-11-24:
Thanks for the info, but the question still remains, how can we detect that a bootloader is used and accordingly subtract the default bootloader size from the `maximum_size` field.

MCUdude, 2020-11-24:
> Thanks for the info, but the question still remains, how can we detect that a bootloader is used and accordingly subtract the default bootloader size from the maximum_size field.

My cores support `board_hardware.uart`. Subtract 512 or 1024 bytes if `board_hardware.uart` is not `no_bootloader`. `board_hardware.uart = uart0` is the default value for all these chips anyways at the moment. The user has to explicitly write in the platformio.ini file that no bootloader should be used: ``board_hardware.uart = no_bootloader`.

valeros, 2020-11-24:
@MCUdude  Could you please update the `maximum_size` field in your manifests so it will represent the whole flash size. Then I will be able to subtract the bootloader size if `no_bootloader` isn't set. Thanks!

MCUdude, 2020-11-24:
> MCUdude Could you please update the maximum_size field in your manifests so it will represent the whole flash size. Then I will be able to subtract the bootloader size if no_bootloader isn't set. Thanks!

Sure! I'll see if I can submit a PR tonight.

valeros, 2020-11-27:
@MCUdude Could you please take a look at https://github.com/platformio/platform-atmelavr/commit/216f574d0ca485570c6f8985574f229006c10e08. Did I miss anything in the calculation process?

closed 2020-11-27 by valeros

MCUdude, 2020-11-27:
@valeros looks good to me!

commit 216f574d (2020-11-27): Dynamically calculate bootloader size for MCUdude's cores
Resolve #207