platformio/platform-atmelavr #211: Could not find the package with tool-avrdude for linux_armv7l

mdevaev, 2020-09-03:
On Raspberry Pi 4 (armv7l) I can't upload AVR firmware:
```
Tool Manager: Installing tool-avrdude @ ~1.60300.0
Error: Could not find the package with 'tool-avrdude @ ~1.60300.0' requirements for your system 'linux_armv7l'
```
The error appeared in platformio 5.0.0. 4.3.1 works fine. It's possible that the problems are also showing up for another architecture (armv6h), but I haven't had a chance to check.

aberix, 2020-09-08:
I experienced the same issue, when I tried to upload to an target using a raspberry pi 3 as host.

Would it be a solution to downgrade platformio to 4.x?

mdevaev, 2020-09-08:
> Would it be a solution to downgrade platformio to 4.x?

Yes, 4.x is working.

aberix, 2020-09-12:
Thanks, that helped for the moment. I am waiting for a fix and will be ready to support testing the fix.

valeros, 2020-09-16:
Hi guys! Sorry for the issue, the package for `linux_armv7l` has been restored, should work now.

closed 2020-09-16 by valeros

gwendolynbrook, 2020-09-28:
I'm seeing the same issue today for armv7 

`toolchain-atmelavr @ 1.40902.0 has been successfully installed!
 PackageManager: Installing tool-avrdude @ ~1.60300.0
 Warning! Package Mirror: The 'sha1' sum '64f2bb7dc9a6c27747095694f2f3f270a9f34a50' of downloaded file 'tool-avrdude-linux_armv6l-1.60300.200527.tar.gz' is not equal to remote '75a72f9c9a17d6a96dabcd15f64e12e45c489b4f'
 Looking for another mirror...
 Error: Could not install 'tool-avrdude' with version requirements '~1.60300.0' for your system 'linux_armv7l'. `