platformio/platform-atmelavr #212: Unable to upload to Adafruit Feather328p

ChasDouglass, 2020-09-24:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

Ubuntu 20.04

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 5.0.1

### Description of problem

Unable to upload to Adafruit feather328p
The Arduino IDE succeeds.

#### Steps to Reproduce

Attempt to upload standard Blink sketch

### Actual Results
```
chas@oak ~/projects/Arduino/Blink> pio run --target upload -e feather328p -v
Processing feather328p (platform: atmelavr; board: feather328p; framework: arduino; upload_port: /dev/ttyUSB0)
------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/feather328p.html
PLATFORM: Atmel AVR (2.2.0) > Adafruit Feather 328P
HARDWARE: ATMEGA328P 8MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.0.0 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/feather328p/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.4% (used 212 bytes from 2048 bytes)
Flash: [=         ]   6.3% (used 2024 bytes from 32256 bytes)
.pio/build/feather328p/firmware.elf  :
section                    size      addr
.data                        46   8388864
.text                      1978         0
.bss                        166   8388910
.comment                     48         0
.note.gnu.avr.deviceinfo     64         0
.debug_info                1524         0
.debug_abbrev              1442         0
.debug_line                  26         0
.debug_str                  520         0
Total                      5814
<lambda>(["upload"], [".pio/build/feather328p/firmware.hex"])
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
BeforeUpload(["upload"], [".pio/build/feather328p/firmware.hex"])
Use manually specified: /dev/ttyUSB0
avrdude -v -p atmega328p -C /home/chas/.platformio/packages/tool-avrdude/avrdude.conf -c arduino -b 57600 -D -P "/dev/ttyUSB0" -U flash:w:.pio/build/feather328p/firmware.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/chas/.platformio/packages/tool-avrdude/avrdude.conf"
         User configuration file is "/home/chas/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Hardware Version: 4238654
         Firmware Version: 21231344.0
         Topcard         : Unknown
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Vtarget         : 214724426.9 V
         Varef           : 2126492.8 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_recv(): programmer is not responding

avrdude done.  Thank you.

*** [upload] Error 1
=============================================================== [FAILED] Took 66.69 seconds ===============================================================

Environment    Status    Duration
-------------  --------  ------------
uno            IGNORED
huzzah         IGNORED
feather328p    FAILED    00:01:06.692
========================================================== 1 failed, 0 succeeded in 00:01:06.692 ==========================================================
```

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = /dev/ttyUSB0

[env:huzzah]
platform = espressif8266
board = huzzah
framework = arduino
upload_port = /dev/ttyUSB0

[env:feather328p]
platform = atmelavr
board = feather328p
framework = arduino
upload_port = /dev/ttyUSB0

```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

int led = 0; // the pin the LED is connected to

void setup() {
    Serial.begin(9600);
    while(!Serial)
        ;
    pinMode(led, OUTPUT); // Declare the LED as an output
    Serial.println("Setup complete");
}

void loop() {
    Serial.println("Flip LED");
    digitalWrite(led, HIGH); // Turn the LED on
    delay(1000);
    digitalWrite(led, LOW); // Turn the LED on
    delay(1000);

}
```

### Additional info

thedanbob, 2020-09-25:
I'm experiencing the same issue on Archlinux but it's inconsistent. I tried several times on two different 328p feathers to upload via `PlatformIO: Build` in VSCode and it failed every time. However, just now I tried with `pio run --target upload -v` in the terminal and it worked on both. It fails again if I drop the `-v` flag, but keeping the flag doesn't always work either so it might be just coincidence.

ChasDouglass, 2020-09-25:
I normally run it through the CLion plugin which does not add the "-v" and it does not work there.  I used the command line version to post the problem hoping that the extra data from "-v" would give someone a clue as to what is wrong.

JoarGjersund, 2021-02-25:
I'm having the same issue. I also tried adding the -v flag then it worked the first time (not sure why), but after that it didn't work anymore. Tried this on two separate feather32p units, both had the same problem.

JoarGjersund, 2021-02-26:
update. Looks like reburning the bootloader with arduino IDE using USBTinyISP solved the problem. Now it works fine programming in platformio as well.