platformio/platform-atmelavr #213: tool-avrdude package too old ? Does not allow to program attiny with usbasp programmer

marcosjl31, 2020-10-15:
I've been facing some problems in order to upload a simple blink program using PlatformIO (see here : https://community.platformio.org/t/attiny85-usbasp-programming-works-on-arduino-ide-fails-on-platformio/16520 ).

Seems that avrdude tool provided within platformIO is not the one you get with ATTinyCore package in Arduino IDE : 

- Arduino : avrdude: Version 6.3-20190619
- PIO : avrdude: Version 6.3, compiled on Sep 12 2016 at 15:21:49

Specifically avrdude.conf seems to be the issue as when copying avrdude.conf from Arduino into PlatformIO "atmelavr" platform  directory, it solves the problem.

May be upgrading version of avrdude could be a solution

marcosjl31, 2020-10-16:
Some more information...
Issue is seen on both Linux and Windows 10 operating System.
I have used latest Arduino and PlatformIO versions.

This is the verbose output of Arduino Upload on Windows 10 : 

```
C:\Users\marcosjl\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\marcosjl\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.4.1/avrdude.conf -v -pattiny85 -cusbasp -Uflash:w:C:\Users\marcosjl\AppData\Local\Temp\arduino_build_249892/Attiny85.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\marcosjl\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.4.1/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny85
         Chip Erase delay              : 400000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    12     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65    12    32    0 yes      8192   64    128 30000 30000 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 187500 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: set SCK frequency to 187500 Hz
avrdude: reading input file "C:\Users\marcosjl\AppData\Local\Temp\arduino_build_249892/Attiny85.ino.hex"
avrdude: writing flash (636 bytes):

Writing | ################################################## | 100% 0.38s

avrdude: 636 bytes of flash written
avrdude: verifying flash memory against C:\Users\marcosjl\AppData\Local\Temp\arduino_build_249892/Attiny85.ino.hex:
avrdude: load data flash data from input file C:\Users\marcosjl\AppData\Local\Temp\arduino_build_249892/Attiny85.ino.hex:
avrdude: input file C:\Users\marcosjl\AppData\Local\Temp\arduino_build_249892/Attiny85.ino.hex contains 636 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.18s

avrdude: verifying ...
avrdude: 636 bytes of flash verified

avrdude done.  Thank you.

```

marcosjl31, 2020-10-16:
This is my blink project platformio.ini file : 

```
[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
board_build.f_cpu = 1000000L
upload_protocol = usbasp
upload_flags = 
    -Pusb
    -v 
```

And here is the verbose output of failing upload : 

```
<lambda>(["upload"], [".pio\build\attiny85\firmware.hex"])
AVAILABLE: usbasp
CURRENT: upload_protocol = usbasp
BeforeUpload(["upload"], [".pio\build\attiny85\firmware.hex"])
avrdude -Pusb -v -v -p attiny85 -C C:\Users\marcosjl\.platformio\packages\tool-avrdude\avrdude.conf -c usbasp -e -D -U flash:w:.pio\build\attiny85\firmware.hex:i

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\marcosjl\.platformio\packages\tool-avrdude\avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: seen device from vendor ->www.fischl.de<-
avrdude: seen product ->USBasp<-
         AVR Part                      : ATtiny85
         Chip Erase delay              : 400000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    12     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65     6    32    0 yes      8192   64    128 30000 30000 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

*** [upload] Error 1

```

marcosjl31, 2020-10-16:
Finally : on either Linux or Windows, copying the file avrdude.conf from Arduino ATTinyCore package to PlatformIO environment solves the issue.

marcosjl31, 2020-10-22:
Ok…
That’s a follow up message on this topic.

One of the differences in Compilation/Upload between the Arduino and PlatformIO IDE is this line (in verbose mode !) :
Setting bit clk period : 5.0
This seems to be atuomagically added in Arduino IDE…

When you set upload_flags to “-Pusb -B5” in platformIO: it solves the problem as well (no need to modify the avrdude.conf file)

This issue can be closed.

MCUdude, 2020-11-15:
This is a know USBasp "bug".

The default bit clock is a little bit too fast for microcontrollers running at 1 MHz. By dividing down the bit clock (using the -B flag) it is able to communicate. @marcosjl31 you can close this issue

marcosjl31, 2020-11-15:
Known Big...

closed 2020-11-15 by marcosjl31