platformio/platform-atmelavr #215: Arduino Micro Pro / Leonardo upload broken

Fusseldieb, 2020-11-05:
Hi,

I digged through numerous posts on the internet facing the same problem and found no solution.

Basically when trying to upload to an Arduino Micro Pro or Leonardo, it resets the board and waits for the new upload port, but waits _SOO_ long that the Arduino exits the bootloader and only then it thinks it's a good time to upload, which then obviously fails with:

```
...
Auto-detected: COM26
Forcing reset using 1200bps open/close on port COM26
Waiting for the new upload port...
...
avrdude: ser_open(): can't open device ...
```

Only workaround for now is to bridge "reset/RST" repeatedly on the Arduino when it's waiting for the new upload port and _eventually_ it uploads.

Would be nice to see a fix for this.

Funny thing is that it only happens on _some_ PCs.

MCUdude, 2020-11-30:
> Funny thing is that it only happens on some PCs.

That is true, and probably what makes it so difficult to fix. Just to confirm this, uploading to a Pro Micro or Leonardo works perfectly on both my Mac and my Windows PC. If you open the Windows device monitor, does the COM port (COM26 in your case) actually disappear and appear again when you try to upload?

Fusseldieb, 2020-12-01:
> > Funny thing is that it only happens on some PCs.
> 
> If you open the Windows device monitor, does the COM port (COM26 in your case) actually disappear and appear again when you try to upload?

Yes, the Arduino vanishes and appears again as Bootloader.

MCUdude, 2020-12-01:
> Yes, the Arduino vanishes and appears again as Bootloader.

And it re-appears as COM26? It's strange that PlatformIO isn't able to detect the new port.

What happens if you add
```
board_upload.use_1200bps_touch = false
```

... to your platformio.ini configuration and try to upload by pressing the reset button manually at the right time?