platformio/platform-atmelavr #223: Compilation error on Windows 10

snikul, 2020-12-23:
Hi, Guys!
I compile the simplest example for Arduino Uno within VSC Platformio on W10 Pro.
With the following errors. I need your friendly assistance to solve my problem.
I have already read tones of comments and found no straight answer.

platformio.ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = COM3
targets = upload

the result:

Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash   
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\uno\src\main.cpp.o
Archiving .pio\build\uno\libFrameworkArduinoVariant.a
Compiling .pio\build\uno\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o
avr-g++: error: src\main.cpp: No such file or directory
avr-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\uno\src\main.cpp.o] Error 1
c:\users\greenonion\.platformio\packages\toolchain-atmelavr\bin\../lib/gcc/avr/5.4.0/../../../../avr/bin/ar.exe: .pio\build\uno\libFrameworkArduinoVariant.a: No such file or directory
*** [.pio\build\uno\libFrameworkArduinoVariant.a] Error 1
Assembler messages:
Fatal error: can't create .pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio\build\uno\FrameworkArduino\CDC.cpp.o: No such file or directory
*** [.pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o] Error 1
*** [.pio\build\uno\FrameworkArduino\CDC.cpp.o] Error 1
================================================== [FAILED] Took 7.99 seconds ==================================================The terminal process "C:\Users\GreenOnion\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Hope to get your friendly reply.

Sergei Nikulin
Kaliningrad/Russia
serg.nikulin@gmail.com

matthewturner, 2021-02-28:
Hey @snikul I've just had exactly the same problem . The only way I've been able to resolve this is to disable Realtime Protection in Windows Defender. Everything seems to work now. Obviously this isn't ideal but it's a problem for another day.

snikul, 2021-02-28:
Hi, Matt,

Your way is understandable but I think the developers are not professionals
well enough.
They recommended to fix it the same way in many of their posts.
They have to find a couple of good Microsoft  system soft guys and pay them
!
If Windows Defender is not compatible they have to recommend the
alternative protection. Which is good and reliable....

My Best Regards
Sergei Nikulin

Kaliningrad/Russia