platformio/platform-atmelavr #224: Using toolchain-atmelavr v5.4.0 vs v7.3.0

GunZi200, 2020-12-30:
Hello everyone :)

I was simply trying to include `#include <cstdint>` for ATmega2561 which didn't work on v5.4.0 or v7.3.0 so maybe avr doesn't support it (curious about some C++ features). Anyway... I saw a big difference in program size: **468 byte difference to be exact!**

```ini
[env]
platform = atmelavr
framework = arduino
monitor_speed = 115200

board_build.f_cpu = 16000000L
board_hardware.uart = uart0
board_hardware.oscillator = external

upload_flags = -Pusb

; Use latest version (version 7.3.0)
platform_packages = 
    toolchain-atmelavr

[env:megaatmega2561]
board = ATmega2561
board_build.mcu = atmega2561
```

I am curious why v7.3.0 is not the default? Are there some compatability issues? Also is it possible to use an even newer version for avr-gcc? I noticed Arduino IDE uses v7.3.0 in their latest version v1.8.13

As I understand it v7.3.0 means PIO is using GNU v7.3.0 but aren't we already at v10.0.0+?

## toolchain-atmelavr v5.4.0
[image]

## toolchain-atmelavr v7.3.0
[image]

MCUdude, 2021-01-01:
I'm not sure why they aren't using v7.3.0 either. Arduino IDE have been doing this for a long time, and some of the gurus over at the Avrfreaks forum complains that MPLAB X and Atmel Studio 7 bundles v5.4.0 and not v7.3.0. BTW how did you switch between these two versions in PlatformIO?

@valeros any good reason to stay at v5.4.0?

GunZi200, 2021-01-01:
You can do it like this: 

```ini
; Use latest version (v7.3.0)
platform_packages = 
    toolchain-atmelavr@~1.70300.191015
```

EmperorArthur, 2021-01-28:
Thank you so much @GunZi200 I have been searching everywhere for how to include a newer toolchain.

How did you find what version mapped to "~1.70300.191015"?  I can't seem to find that information.

GunZi200, 2021-01-28:
@EmperorArthur I saw it here on the PlatformIO forum (from @maxgerhardt) https://community.platformio.org/t/code-not-building-in-platformio-but-building-in-arduino-ide-avr-core-issue/16618 😄

maxgerhardt, 2021-02-13:
Users also encounter actual compiler bugs in e.g. [here](https://community.platformio.org/t/undefined-reference-errors-when-the-code-base-became-too-big/19312) and some Arduino libraries like FastLED need the newer compiler too to be compilable ([see here](https://community.platformio.org/t/fastled-clockless-trinket-h24-error/11929)). If the Arduino IDE is using it currently, I'd say we should upgrade immediately.

technik, 2021-02-15:
As a workaround, I downloaded GCC 10 for AVR from here https://blog.zakkemble.net/avr-gcc-builds/ and manually copied it all to ~/.platformio/packages/toolchain-atmelavr. Now I can use the latest C++ in my builds :)
It would be nice to see the main repo keep up with latest gcc by default though.

EmperorArthur, 2021-02-17:
At the least the documentation around "toolchain-atmelavr" could be expanded.  Especially since this will probably happen again in the future.  Instead of stumbling around, trying to figure out where to get the latest version, there should be a page somewhere.

As an example:
* The [first google result](https://docs.platformio.org/en/v0.6.0/platforms/atmelavr.html), which is an old version of the docs, inaccurately says that "toolchain-atmelavr" contains multiple things, but has no links to the package itself
* The [latest](https://docs.platformio.org/en/latest/platforms/atmelavr.html) version of the docs, just link to GCC's latest page on the AVR. Not even the one for the compiler we're using!
* The [packages page](https://platformio.org/platforms/atmelavr/packages) on the main site, also just links to GCC!
* Searching Google for "toolchain-atmelavr 7.3" (without the quotes) now brings up fixes, but they don't explain why the version numbers correspond

**Heck, I have no idea where the "toolchain-atmelavr" repository is, where PlatfomIO downloads it from, or even how it is built!**

This might be another issue, but it certainly made something as easy as upgrading a dependency into a nightmare.

commit 9b317bd6 (2021-02-22): Switch platform to GCC v7.3.0 // Resolve #224
closed 2021-02-22 by valeros