platformio/platform-atmelavr #225: UnicodeDecodeError on upload.

Lhotse, 2020-12-30:
I'm getting the same issue while trying to upload to Arduino Micro, Pro Mini and Uno.

```
PlatformIO 5.0.4
VS code 1.52.1
Distribution: Linux Mint 19 Tara
Kernel: Linux 4.15.0-128-generic (x86_64)
Python 3.6.9
framework-arduino-avr 5.1.0 
tool-avrdude 1.60300.200527 (6.3.0) 
toolchain-atmelavr 1.50400.190710 (5.4.0)
```

```
Connecting to programmer: .
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 46, in run
    self.do_reading()
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 69, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 69, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 24: invalid continuation byte
```
I'm just trying to upload a blink sketch as a test, everything was working fine last time I used it to upload to Arduino Uno 3 months ago. It think there might be an invalid character that is not uff-8. I tried checking the proc.py but don't know enough about python to work out how to fix the issue. Also how can I find out what file is causing the UnicodeDecodeError. Thanks for any help.

valeros, 2021-01-04:
Hi @Lhotse ! Could you please run `pio device list` command directly in the IDE terminal?

Lhotse, 2021-01-04:
Hi Valeros, here is the output of the command.

```
$ pio device list
/dev/ttyS0
----------
Hardware ID: PNP0501
Description: ttyS0

/dev/ttyUSB0
------------
Hardware ID: USB VID:PID=1A86:7523 LOCATION=1-1.2
Description: USB2.0-Serial
```

Also, the log showing the upload port connecting...
```

> Executing task: pio run --target upload --environment micro <

Processing micro (platform: atmelavr; board: micro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/micro/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.8% (used 149 bytes from 2560 bytes)
Flash: [=         ]  14.4% (used 4132 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Forcing reset using 1200bps open/close on port /dev/ttyUSB0
Waiting for the new upload port...
Uploading .pio/build/micro/firmware.hex

Connecting to programmer: .
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 47, in run
    self.do_reading()
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 70, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/home/tristan/.platformio/penv/lib/python3.6/site-packages/platformio/proc.py", line 70, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 24: invalid continuation byte

====================================================================== [FAILED] Took 8.73 seconds ======================================================================
The terminal process "pio 'run', '--target', 'upload', '--environment', 'micro'" terminated with exit code: 1.
```

Lhotse, 2021-01-16:
I installed a clean copy of VSCode on a clean copy of Ubuntu 20.04. The issue no longer exists so I will close this.

closed 2021-01-16 by Lhotse