platformio/platform-atmelavr #226: Inconsistent free Flash memory between ATmega2560 and ATmega2561 when using a bootloader

GunZi200, 2020-12-31:
Hi I noticed this issue before v3.1.0

When I build/compile PIO shows total flash memory: 

* ATmega2561 shows: **261120 bytes**
* ATmega2560 shows: **253952 bytes** (8192 has been subtracted)

I'm not familiar where these numbers are calculated but I would think the flash storage should look exactly the same. Only difference between the chips is ATmega2561 has fewer physical pins.

Here is my platformio.ini settings:

```ini
[env]
platform = atmelavr
framework = arduino
monitor_speed = 115200

board_build.f_cpu = 16000000L
board_hardware.uart = uart0
board_hardware.oscillator = external

upload_flags = -Pusb

platform_packages = 
    toolchain-atmelavr

[env:megaatmega2561]
board = ATmega2561
board_build.mcu = atmega2561

[env:megaatmega2560]
board = megaatmega2560
board_build.mcu = atmega2560
```

GunZi200, 2020-12-31:
I looked into `arduino.py` and I think the problem is ATmega2560 is not considered to be any of `"MiniCore"`, `"MegaCore"`, `"MightyCore"`, `"MajorCore"` So maybe this isn't an issue at all? Why is there such a big difference in bootloader sizes? 🤔

MCUdude, 2021-01-01:
Hi!
`board = megaatmega2560` is probably the Arduino MEGA 2560 board. I'll assume you want to use the MegaCore implementation? Try using `board = ATmega2560` instead.

The MEGA2560 board uses a bootloader where 8192 bytes of flash are allocated for it. MegaCore does not allocate any flash memory for a bootloader if you haven't specified in the platformio.ini file that a bootloader is used.

GunZi200, 2021-01-03:
Thanks! I'm closing this issue since this isn't a bug 👍

closed 2021-01-03 by GunZi200