platformio/platform-atmelavr #232: Issue with 3rd party AVR core versioning

MCUdude, 2021-01-27:
Hi!

Even though MightyCore v2.0.6 is the latest version PlatformIO currently use, it still pulls the latest v2.0.8 when installing MightyCore. Is this right?

When I'm building for target `ATmega1284P`, this is the output:

```
Processing Upload_UART (platform: atmelavr; framework: arduino; board: ATmega1284P)
--------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduino-avr-mightycore @ ~2.0.6
Unpacking  [####################################]  100%          
Tool Manager: framework-arduino-avr-mightycore @ 2.0.8 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284P.html
PLATFORM: Atmel AVR (2.2.0+sha.41ab7b7) > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr-mightycore 2.0.8 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
Building in release mode

...

```

but when I go and look at the actual platform.txt file that's bundled with MightyCore, it reveals that v2.0.8 was downloaded.

It would be great if v2.0.6 actually downloaded v2.0.6.

And how do I manually specify what MightyCore version my project should use?

valeros, 2021-01-27:
That's how [Semantic Versioning](https://docs.npmjs.com/cli/v6/using-npm/semver#tilde-ranges-123-12-1) works. Even though we specified `2.0.6` as the minimal version, we also added the symbol `~` which allows patch-level changes that shouldn't break anything.

closed 2021-01-27 by valeros

MCUdude, 2021-01-27:
Ah! So it will actually pull newer versions than 2.0.6 due to the ~ in front.

> And how do I manually specify what MightyCore version my project should use?

I also figured out how to set this.

``` ini
; Force MightyCore v2.0.6
platform_packages =
  framework-arduino-avr-mightycore@2.0.6
```

commit 2abd70b7 (2021-01-27): Update default MightyCore package to v2.0.8
Resolve #232