platformio/platform-atmelavr #233: Digistump digispark ATTiny85 compile time noise

anno73, 2021-01-30:
Hi guys!

When compiling for a digispark, I get a lot of warning noise. I do not know how important this is but it makes me at least nervous every time I compile my code :-)

It boils down to 
- redefinition of BIN already mentioned in https://github.com/platformio/platform-atmelavr/issues/47 in 2017
- #warning "This file has been moved to <...>
- warning: initialization makes integer from pointer without a cast [-Wint-conversion]
- warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]

Any chance to get this fixed please?

Br, Alex.

github project: https://github.com/anno73/IoT-Universal-Socket-Strip/tree/master/software/relay

platformio.ini:
```
[platformio]
description = IoT Universal Socket Strip

[env:digispark-vusb]
platform = atmelavr
board = digispark-tiny
framework = arduino
debug_tool = simavr
lib_deps =
    ; nickcengel/TinyWireSio @ ^0.0.0-alpha+sha.722ada4382
    nickcengel/TinyWireSio 
    prosenb/EEPROMWearLevel @ ^2.0.0
lib_ldf_mode = chain+
```

Compile time output:
```
> Executing task in folder relay: C:\Users\alex\.platformio\penv\Scripts\pio.exe run <

Processing digispark-vusb (platform: atmelavr; board: digispark-tiny; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/digispark-tiny.html
PLATFORM: Atmel AVR (3.1.0) > Digispark USB     
HARDWARE: ATTINY85 16MHz, 512B RAM, 5.87KB Flash
DEBUG: Current (simavr) On-board (simavr)       
PACKAGES:
 - framework-arduino-avr-digistump 1.6.7        
 - toolchain-atmelavr 1.50400.190710 (5.4.0)    
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyWireSio> 0.0.0-alpha+sha.722ada4382
|-- <EEPROM> 2.0
|-- <EEPROMWearLevel> 2.0.0
|   |-- <EEPROM> 2.0
Building in release mode
Compiling .pio\build\digispark-vusb\src\main.cpp.o
Compiling .pio\build\digispark-vusb\lib639\TinyWireSio\TinyWireS.cpp.o
Compiling .pio\build\digispark-vusb\lib639\TinyWireSio\usiTwiSlave.c.o
Archiving .pio\build\digispark-vusb\lib1ac\libEEPROM.a
Compiling .pio\build\digispark-vusb\lib69f\EEPROMWearLevel\EEPROMWearLevel.cpp.o
Compiling .pio\build\digispark-vusb\FrameworkArduinoVariant\pins_arduino.c.o
Compiling .pio\build\digispark-vusb\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\digispark-vusb\FrameworkArduino\Print.cpp.o
In file included from C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Stream.h:24:0,
                 from C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/TinyDebugSerial.h:31,
                 from C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/WProgram.h:18,
                 from C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Arduino.h:4,
Compiling .pio\build\digispark-vusb\FrameworkArduino\TinyDebugSerial.cpp.o
Compiling .pio\build\digispark-vusb\FrameworkArduino\TinyDebugSerial115200.cpp.o
[... 312 lines cut ...]
        ^
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\wiring_digital.c: In function 'digitalWrite':
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\pins_arduino.h:80:33: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
 #define portOutputRegister(P) ( (volatile uint8_t *)( pgm_read_byte( port_to_output_PGM + (P))) )
                                 ^
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\wiring_digital.c:127:8: note: in expansion of macro 'portOutputRegister'
  out = portOutputRegister(port);
Linking .pio\build\digispark-vusb\firmware.elf
        ^
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\wiring_digital.c: In function 'digitalRead':
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\pins_arduino.h:81:32: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
 #define portInputRegister(P) ( (volatile uint8_t *)( pgm_read_byte( port_to_input_PGM + (P))) )
                                ^
C:\users\alex\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny\wiring_digital.c:153:7: note: in expansion of macro 'portInputRegister'
  if (*portInputRegister(port) & bit) return HIGH;
       ^
Building .pio\build\digispark-vusb\firmware.hex
Checking size .pio\build\digispark-vusb\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.1% (used 62 bytes from 512 bytes)
Flash: [===       ]  31.0% (used 1864 bytes from 6012 bytes)
============================================================================================================================================ [SUCCESS] Took 4.77 seconds ============================================================================================================================================
Terminal will be reused by tasks, press any key to close it.

```

maxgerhardt, 2021-02-06:
PlatformIO uses the core versions that are available for the board, they've not written that themselves.

To my knowledge the core is at https://github.com/digistump/DigistumpArduino/. You would need to ask them to fix the errors in their code.

Given that the last update was 4 years ago I don't think they'll respond though. Seems pretty abondend.

But I did stumble upon a forked & improved version of the core: https://github.com/ArminJo/DigistumpArduino. 

You can install that in your Arduino IDE and see if the code there works and doesn't have the warnings. Then you can raise a feature request to use that new core.

But after all, these are warnings that can be ignored I think.

anno73, 2021-02-06:
Thanks for the reply and thanks for the link. To be honest, I mistrust code that leaves warnings during compile time. Even if they can be safely ignored - possibly a bad habit of mine. I always fear to overlook something important.
Therefore I tried different cores like the trinket and the generic tinyx5 one. In the end I decided to go with a mix:
```
platform = atmelavr
board = digispark-tiny

board_build.core = tiny
board_build.variant = tinyx5
```
This works very well for me now. 

As of ArminJo's DigistumpArduino core, could you kindly point me to a location describing on how to install a custom core in VSCode/PlatformIO? I'll gladly give it a try then. 
I also pointed the maintainers to this project if they wanted to have their core added to platformio.

ArminJo, 2021-02-06:
> I also pointed the maintainers to this project if they wanted to have their core added to platformio.

@maxgerhardt 
Hi Max,
there is a [refurbished version of the digispark core](https://github.com/ArminJo/DigistumpArduino), can you tell me the steps to get it integrated in pio for common usage?

maxgerhardt, 2021-02-06:
Mainly works has to be done in this repository (platform-atmelavr) to add the board defnitions (in JSON format, `boards/`) that the core uses (if any new ones are in there), plus for the existing boards that should use this core, the `core:` attribute needs to be changed. Then a PlatformIO package has to be [packaged and published](https://docs.platformio.org/en/latest/core/userguide/package/cmd_pack.html) (e.g., the current board defnitions with `core:"digistump"` use the package `framework-arduino-avr-digistump`, a new one would have to be created).

A nice-to-have in the repository would just be that it has a `package.json` file that contains some metainfo (name, version, ..) so that the PlatformIO package can be created without the PlatformIO devs creating that file themselves (see e.g. https://github.com/esp8266/Arduino/blob/master/package.json) 

There's also a really hacky way to include a different core. Change what the used `core` value and then inject the package that PlatformIO will be looking for on the fly.

```
board_build.core = digistump-new
platform_packages = 
   framework-arduino-avr-digistump-new@<repository link that just has the digistump-avr subfolder of new core and platformio.ini>
```
I'll quickly see what changes need to be done here and check back then.

maxgerhardt, 2021-02-06:
Okay so it's actually much easier than expected to test it. Just use a `platform_packages` description to overwrite the `framework-arduino-avr-digistump` package.

Tested with

```
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
platform_packages = 
    framework-arduino-avr-digistump@https://github.com/ArminJo/DigistumpArduino/raw/master/framework-arduino-avr-digistump-1.7.0.zip
```

and it builds *without warnings*, uploads and runs on my Digispark USB device. Tested a simple Blinky.

Others should be able to test the core too that way. 

@ArminJo can you create a `framework-arduino-avr-digistump-1.7.1.zip` just like you did with 1.7.0? Then the PlatformIO developers should be able to integrate that package pretty easily here.

anno73, 2021-02-06:
Tested it on my source too and compiles w/o any issues. 
Final code is even a bit smaller: tiny/tinyx5: 5628 ArminJo: 5482
Also ram usage is slightly smaller by some bytes.
Deployed code does not show any issues on modules aswell.

ArminJo, 2021-02-07:
@maxgerhardt 
Thanks for the advice.
I created a new zip https://github.com/ArminJo/DigistumpArduino/blob/master/framework-arduino-avr-digistump-1.7.1.zip.

But what's about the new [micronucleus binaries](https://github.com/ArminJo/DigistumpArduino/tree/master/tools) for the different OS? 
How are they upgraded?

maxgerhardt, 2021-02-07:
>But what's about the new micronucleus binaries for the different OS?
> How are they upgraded?

For that you would have to create a package (see above links with `pio package pack`, again needs a `package.json`) file that has has the right binaries in it.

PlatformIO sticks the Micronucleus upload tool in a `tool-micronucleus` package with content (for Windows)

[image]

(Linux has only a folder with just the `micronucleus` binary).

See https://bintray.com/platformio/tool-packages and search for "tool-micronucleus". A reference `package.json` for that would e.g. be 

```json
{
  "name": "tool-micronucleus",
  "version": "1.200.4",
  "description": "ATTiny usb bootloader with a strong emphasis on bootloader compactness",
  "keywords": [
    "tools",
    "uploader"
  ],
  "license": "GPL-2.0-or-later",
  "system": [
    "linux_x86_64"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/micronucleus/micronucleus"
  }
}
```
Whereas the `200` encodes version 2.0.0 btw.

So the uploaded archives at https://github.com/ArminJo/DigistumpArduino/tree/master/tools are *almost* perfect, they're just missing the `package.json` in them with the meta-inofrmation for version and system updated.

ArminJo, 2021-02-07:
OK Thanks,
What is the exact semanics of the version field? micronucleus currently has version 2.5.

maxgerhardt, 2021-02-07:
It's using semver / semantic versioning. Here's [an example and explanation](https://community.platformio.org/t/find-the-mbed-version-which-was-used-before-a-backup/16159/4?u=maxgerhardt). For version 2.5 i'd encode it as `1.205.4` or 1.20500.4` (2.5.0). The last field is pretty much always used as a datecode though, e.g. 210207 encoding 2021-02 (Feb.)-07.

ArminJo, 2021-02-07:
@maxgerhardt Ok I prepared the [tools zip files](https://github.com/ArminJo/DigistumpArduino/tree/master/tools).
I am not quite sure if the windows USB driver installation files should be a part of the zip, since I have no idea how one would install the Digispark core for windows. Is there an autoexec for the old install.exe or so? Or is the user aware that there is an install.exe he must execute?

@anno73 what would you suggest?

As told, I use Eclipse and Sloeber, and have no experience with PIO.

maxgerhardt, 2021-02-07:
>Is there an autoexec for the old install.exe or so? Or is the user aware that there is an install.exe he must execute?

The drivers are not auto-installed, they are just referenced in documentation so that users can install it themselves.

ArminJo, 2021-02-07:
Thanks!
Then I will test it first on a fresh W10 system.

anno73, 2021-02-08:
@maxgerhardt

> I created a new zip https://github.com/ArminJo/DigistumpArduino/blob/master/framework-arduino-avr-digistump-1.7.1.zip.

I updated to v1.7.1. Upon first compile I get following errors:
~~~
> Executing task in folder relay: C:\Users\alex\.platformio\penv\Scripts\pio.exe run <

Processing DigistumpArduino (platform: atmelavr; board: digispark-tiny; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing https://github.com/ArminJo/DigistumpArduino/raw/master/framework-arduino-avr-digistump-1.7.1.zip
Unpacking  [####################################]  100%
Tool Manager: framework-arduino-avr-digistump @ 1.7.1 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/digispark-tiny.html
PLATFORM: Atmel AVR (3.1.0) > Digispark USB
HARDWARE: ATTINY85 16MHz, 512B RAM, 5.87KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
TypeError: _isdir: path should be string, bytes or os.PathLike, not NoneType:
  File "C:\users\alex\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\alex\.platformio\platforms\atmelavr\builder\main.py", line 162:
    target_elf = env.BuildProgram()
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\alex\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\alex\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\users\alex\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\alex\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\alex\.platformio\platforms\atmelavr\builder\frameworks\arduino.py", line 43:
    assert isdir(FRAMEWORK_DIR)
============================================================================================================================================ [FAILED] Took 7.95 seconds ============================================================================================================================================ 
The terminal process "C:\Users\alex\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
~~~

Second compile then works nice again.

Is this an issue of PIO or the v1.7.1 package? I think I had the same issue with initially changing to v1.7.0.

anno73, 2021-02-08:
> @anno73 what would you suggest?
> 
> As told, I use Eclipse and Sloeber, and have no experience with PIO.

@ArminJo I have no idea as I am not a DEV of PlatformIO but just a user ^.^ But I have seen you got an answer on this from maxgerhardt already :-)

bablokb, 2021-02-08:
So I followed the instructions from @maxgerhardt :
  - downloaded the tar.gz for my Linux-system from @ArminJo 
  - created a package.json
  - updated the tar.gz with the package json
  - added the `platform_packages` directive (pointing to my local tar.gz)

Now I'm almost there, there is only one error left:

    Configuring upload protocol...
    AVAILABLE: micronucleus
    CURRENT: upload_protocol = micronucleus
    Uploading .pio/build/digispark-tiny/firmware.hex
    Unrecognized option: -c
    *** [upload] Error 1

If I manually call the now unpacked and updated micronucleus to upload the hex, everything works fine. So how do I convince pio that micronucleus now uses different commandline options?

maxgerhardt, 2021-02-08:
>Unrecognized option: -c

Ah, that doesn't work anymore with `-c`? The current code does

https://github.com/platformio/platform-atmelavr/blob/dec3edd5f16c79a62bcc91fed7d04618b04721fe/builder/main.py#L184-L192

So to fix that there are two options: 
1. Correct the code to use the correct parameters (`C:\Users\<user>\.platformio\platforms\atmelavr\builder\main.py`)
2. Use a custom [`upload_cmd`](https://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer) in the `platformio.ini`

In general you would add

```
upload_protocol = custom
upload_flags =
    -flag
    value1
    --flag-2
    value 2
upload_command = upload-executable $UPLOAD_FLAGS $SOURCE
```
the `$SOURCE` will automatically evaluate to teh uploaded firmware, and `$UPLOAD_FLAGS` will evaluate what you will have written in `upload_flags`. Each flag / value needs to be newline-separated and indented.

If I read [docs](https://github.com/ArminJo/micronucleus-firmware/tree/master/commandline) correctly it should need a 

```
upload_protocol = custom
upload_flags =
    --run
  upload_command = $PROJECT_PACKAGES_DIR/tool-micronucleus/micronucleus $UPLOAD_FLAGS $SOURCE
```
so that will call `micronucleos --run <firmware.hex>`. 

Check that the installed micronucleos tool is in `C:\Users\Max\.platformio\packages\` though, if you have an older and a newer version installed there might be `tool-micronucleos@xyz` folders too.

ArminJo, 2021-02-08:
@maxgerhardt 
for the new versions, the correct commandline would be:
`--no-ansi --run --timeout 60 {build.path}/{build.project_name}.hex`
see https://github.com/ArminJo/DigistumpArduino/blob/c41f52111c7f3cd0b3f961be3f890955cae4ef8c/digistump-avr/platform.txt#L100
The old one of the so called "launcher" was : `-cdigispark --timeout 60 -Uflash:w:{build.path}/{build.project_name}.hex:i`
How can we address this? It is the same upload_protocol name.

maxgerhardt, 2021-02-09:
I'll open a proper issue or PR for full-integration of the new core, then that will be directly addressed in the platform code.

bablokb, 2021-02-09:
Works perfect. I'm looking forward to the integration into pio, but this is a usable workaround. Thank you all.

maxgerhardt, 2021-02-10:
Feature request opened as linked above this comment.