platformio/platform-atmelavr #235: Bootloader programming for digispark-tiny

bablokb, 2021-02-06:
I have problems uploading a program to the digistump-attiny85 because the bootloader is not supported. Digging into some threads I found out that platform.io does support uploading a new bootloader. [The documentation](https://docs.platformio.org/en/latest/platforms/atmelavr.html#bootloader-programming) states _"The default bootloader image and corresponding fuse bits are predefined in the board manifest file in the bootloader section, ..."_ but this does not seem the case for the digispark-tiny, since when I try to run the bootloader target I get the error-message

    Error: missing bootloader configuration!

So I assume that either the documentation is  not fully correct, or the board manifest file does not define the necessary configuration. Is there a workaround for the problem?

maxgerhardt, 2021-02-06:
The used core (https://github.com/digistump/DigistumpArduino/tree/master/digistump-avr in latest-released version 1.6.7.. from 5 years ago..) does not contain a bootloader file. 

Where do you get the bootloader from?

bablokb, 2021-02-07:
Yes, I also noticed that the core uses a very old version of micronucleus as the upload-program. The micronucleus github-repo has newer versions and also newer bootloaders.

It could turn out though that my root problem is not the bootloader, but the old version of micronucleus itself. The bootloader on my device seems to be too new for the old version of platformio. I will investigate this in the next days.

Nevertheless it would be a good idea to fix the manifest-file.

maxgerhardt, 2021-02-07:
You should look at the latest replies in https://github.com/platformio/platform-atmelavr/issues/233 because it shows how you can very easily use a new, forked & improved version of the core.

If you are missing the bootloader though your device is bricked and you need to have / build an ISP to reflash the bootlaoder, per https://github.com/ArminJo/micronucleus-firmware. I've flashed that bootloader on my device (via a normal USB upgrade as they describe) and I can still upload programs using PlatformIO.

[image]

bablokb, 2021-02-08:
I added a comment to https://github.com/platformio/platform-atmelavr/issues/233

The new version of micronucleous fixed the problem that I could not upload the program, although using the program from within pio fails because of incompatibilities of commandline options. But that is a different problem.

So I actually don't have to reprogram the bootloader at all, thanks for pointing me in the right direction. The question is if it makes sense to close this issue, because independent of my problem the board manifest-file is incomplete after all.