platformio/platform-atmelavr #242: [Critical] Latest toolchain-atmelavr package does not contain avr-gdb

maxgerhardt, 2021-02-22:
I've marked this issue as "critical" in the title because per closure of #224 the default toolchain-atmelavr version will soon be the latest 7.4.0 package version. However, when attempting to use the latest version by doing a 

```
platform_packages =
  toolchain-atmelavr@>=1.70300.0
```
through which one gets a package with version code `2.70300.201015`, debugging with avr-stub breaks because that toolchain package, for whatever reason, does not contain the `avr-gdb` executable. Starting the debugger simply loads forever. Refer issue https://github.com/jdolinay/avr_debug/issues/24 for reproduction.

Using an older version of the package, 

```
platform_packages =
  toolchain-atmelavr@1.70300.191015
```
*works*. 

If the package version `2.70300.201015` is PlatformIO provided (and I assume that is what will be used starting with the next release), please double check that it contains `avr-gdb`.

valeros, 2021-02-23:
Thanks for looking out. This platform is set to use `~1.70300.0` that contains GDB.

closed 2021-02-23 by valeros

maxgerhardt, 2021-02-23:
So the `2.70300.201015` was user uploaded? Can it still be ammended to contain `avr-gdb` for safety? 

Because I see it, there a few Google hits now where I (and other people) tell people to use the above `>=..` expression to get the new compiler, which will leave them unable to debug for reasons above.

Examples:
* https://community.platformio.org/t/fastled-clockless-trinket-h24-error/11929/6
* https://www.reddit.com/r/FastLED/comments/k25zea/trying_to_use_platform_io_with_fastled_library/gdsei28/
* https://stackoverflow.com/questions/61057914/error-compiling-for-board-arduino-genuino-mega-or-mega-2560-with-fastled-library
* http://www.quabr.com/61057914/error-compiling-for-board-arduino-genuino-mega-or-mega-2560-with-fastled-library

I know it's only 4 hits and one of the posts is mine which I can edit, but I can't edit the content on the other sites. There may be situations in which people find and use these posts. Of course, there should be less people looking for these posts links above because if the compiler is upgraded in general, the e.g. FastLED library works directly without a compiler error so there's no reason to look up anything to change it.