platformio/platform-atmelavr #247: Outdated ATTinyCore with i2c error

dontsovcmc, 2021-03-14:
Current version 1.5.2: https://github.com/SpenceKonde/ATTinyCore/releases

I see in output:
```
PACKAGES: 
 - framework-arduinoavr 4.1.2 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
```
It's hidden core version. Your version is 1.2.0: 
```
.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny/ATTinyCore/library.properties

name=ATTinyCore
version=1.2.0
author=Spence Konde
maintainer=Spence Konde <spencekonde@gmail.com>
```

1. where framework-arduinoavr downloads 1.2.0 ATTinyCore?

2. How I can use another ATTinyCore version safety?

Remark:
I love platformio, opensourse and library management. BUT I can't understand how Platformio framework management works! It wastes a lot of time for debugging framework errors!

valeros, 2021-03-15:
Hi @dontsovcmc ! Your log seems a bit outdated, the latest `framework-arduinoavr` package is no longer contains external cores, please run `pio platform update atmelavr`.

dontsovcmc, 2021-03-15:
@valeros I use platformio extension for VSCode. 
VSCode: Core 5.1.0·Home 3.3.4
platformio.ini platform = atmelavr@3.1.0

How I can understand what source files platformio use when compiling?

valeros, 2021-03-15:
You locked your platform version specifically to v3.1.0. It means that you won't get any package updates except patch fixes. If you want to pull the latest changes, you need to the delete version requirement `@3.1.0` and update your platform.

dontsovcmc, 2021-03-15:
Of course! atmelavr 3.1.0 is the latest. So, I can't understand where I can find you dependencies to ATTinyCore

valeros, 2021-03-15:
> atmelavr 3.1.0 is the latest

The latest is 3.2.0, but anyway, it looks better now. Does that i2c error still bother you taking into account hat the package version is now `1.3.2`.

dontsovcmc, 2021-03-15:
Wow, it's good.
1. So, if I remove version from platformio.ini, i need upgrade library manually before compiling, yes?
2. Where I can find what framework commit is used?

valeros, 2021-03-15:
> So, if I remove version from platformio.ini, i need upgrade library manually before compiling, yes?

Do you mean library or framework package? When you remove a version requirement then PlatformIO will update the package automatically if the `auto_update_platforms` setting is not disabled.

> Where I can find what framework commit is used?

The version of the package it the same as in the upstream repository.

dontsovcmc, 2021-03-15:
I know where is atmelavr releases https://github.com/platformio/platform-atmelavr/releases/tag/v3.2.0
But, I can't find tinycore repo... Where is it?

valeros, 2021-03-15:
https://github.com/SpenceKonde/ATTinyCore/releases

dontsovcmc, 2021-03-15:
Where I can find "what ATTinyCore version is inside platform-atmelavr 3.2.0"?

valeros, 2021-03-15:
https://github.com/platformio/platform-atmelavr/blob/develop/platform.json#L48

dontsovcmc, 2021-03-15:
Ok "~1.3.2", where is a link to ATTinyCore?

valeros, 2021-03-15:
There's no direct link, the package is downloaded to PlatformIO Registry and then delivered to users via our CDN.

dontsovcmc, 2021-03-15:
Ok, I'm nubbie, I know nothing about ATTinyCore. How can I understand, that platform.json downloads ATTinyCore for framework-arduino-avr-attiny?

valeros, 2021-03-15:
> Ok, I'm nubbie, I know nothing about ATTinyCore. How can I understand, that platform.json downloads ATTinyCore for framework-arduino-avr-attiny?

I'm not sure I understand the question. The package is downloaded when you first run a project that uses one of the boards supported by ATTinyCore.

> In what local folder I can find any property file from ATTinyCore?

The package is located here `~/.platformio/packages/framework-arduino-avr-attiny`

dontsovcmc, 2021-03-15:
I see framework-arduino-avr-attiny version:
/Users/dontsov/.platformio/packages/framework-arduino-avr-attiny/package.json - 1.3.2, ok.
But current SpenceKonde version is 1.5.2. Where is this number?

valeros, 2021-03-15:
It means that the latest supported version is 1.3.2 and there is no package for 1.5.2 at the moment.

dontsovcmc, 2021-03-15:
WOW! I'm confused. You have hidden relation with SpenceKonde/ATTinyCore repo without any information in the source code. You have 1.3.2 version of somewhere in platform.json.

p.s. I spend some days debugging code and can't understand with source downloads.

valeros, 2021-03-15:
> You have hidden relation with SpenceKonde/ATTinyCore

It's not hidden, you can open `~/.platformio/packages/framework-arduino-avr-attiny/package.json` and see what version and repository are used.

> p.s. I spend some days debugging code and can't understand what source downloaded.

What do you mean? It's explicitly printed when you compile your project:
```bash
PACKAGES:
...
 -  framework-arduino-avr-attiny 1.3.2
...
```

dontsovcmc, 2021-03-15:
Sorry, I don't see
```
"repository": {
    "type": "git",
    "url": "https://github.com/SpenceKonde/ATTinyCore"
  }
```

Is it possible to use latest ATTinyCore without framework-arduino-avr-attiny?

dontsovcmc, 2021-03-16:
> Does that i2c error still bother you taking into account hat the package version is now `1.3.2`?
The bug was solved in 1.5.0.
https://github.com/SpenceKonde/ATTinyCore/commit/4b9d184d5f1a75cdcc3d87aa9c412058e3ac5e4c

Is it possible to upgrade ATTinyCore version or I can link last ATTinyCore version manually?

Thanks and sorry for long talk!

valeros, 2021-03-18:
The ATTinyCore has been updated to the latest v1.5.2 in the dev branch, please test it with the upstream platform:
```ini
[env:envname]
platform = https://github.com/platformio/platform-atmelavr.git
framework = ...
board = ...
```

closed 2021-03-18 by valeros

dontsovcmc, 2021-04-16:
Thanks a lot!

commit 97cb11fe (2021-03-18): Update ATTinyCore to v1.5.2 // Resolve #247