platformio/platform-atmelavr #248: Cannot build for Atmega32u4 (Leonardo).  Bugs in 32u416m framework (USBCore)?

MakerMatrix, 2021-03-21:
I am trying to build a fairly simple code (which works fine in the ArduinoIDE).  But when I try to build it, there are many "not defined in this scope) errors coming from the Framework itself, specifically USBCore.  Screen scrape below.

> Executing task: C:\Users\jsmith\.platformio\penv\Scripts\platformio.exe debug <

Processing 32u416m (platform: atmelavr; board: 32u416m; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/32u416m.html
PLATFORM: Atmel AVR (3.2.0) > Microduino Core USB (ATmega32U4@16M,5V)
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit ILI9341> 1.5.6
|   |-- <Adafruit GFX Library> 1.10.6
|   |   |-- <SPI> 1.0
|   |   |-- <Adafruit BusIO> 1.7.2
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <HX711_ADC> 1.2.7
|-- <Adafruit BusIO> 1.7.2
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <SPI> 1.0
Building in debug mode
Compiling .pio\build\32u416m\FrameworkArduino\USBCore.cpp.o
In file included from C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBAPI.h:44:0,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.cpp:20:
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.cpp:73:29: error: 'USB_VID' was not declared in this scope
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                             ^
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.h:270:61: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                             ^~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.cpp:73:29: note: suggested alternative: 'USB_Recv'
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                             ^
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.h:270:61: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                             ^~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.cpp:73:37: error: 'USB_PID' was not declared in this scope
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                                     ^
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.h:270:66: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                  ^~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.cpp:73:37: note: suggested alternative: 'USB_Recv'
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                                     ^
C:\Users\jsmith\.platformio\packages\framework-arduino-avr\cores\arduino\USBCore.h:270:66: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                  ^~~~
*** [.pio\build\32u416m\FrameworkArduino\USBCore.cpp.o] Error 1
================================================================= [FAILED] Took 1.10 seconds =================================================================
The terminal process "C:\Users\jsmith\.platformio\penv\Scripts\platformio.exe 'debug'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

MakerMatrix, 2021-03-22:
My platformio.ini:

----------------------------------------------------
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:32u416m]
platform = atmelavr
board = 32u416m
framework = arduino
lib_deps = 
	adafruit/Adafruit ILI9341@^1.5.6
	olkal/HX711_ADC@^1.2.7
	adafruit/Adafruit BusIO@^1.7.2
	Wire
upload_port = COM5
--------------------------------------------

If I change env to "uno" and board to "leonardo", the software builds, no problem.

valeros, 2021-03-22:
Hi @MakerMatrix ! You need to specify the `usb_product` field similar to the Leonardo board, e.g.:
```ini
[env:32u416m]
platform = atmelavr
board = 32u416m
framework = arduino
lib_deps = 
	adafruit/Adafruit ILI9341@^1.5.6
	olkal/HX711_ADC@^1.2.7
	adafruit/Adafruit BusIO@^1.7.2
	Wire
upload_port = COM5

board_build.usb_product = "32u416m"
```

closed 2021-03-22 by valeros