platformio/platform-atmelavr #250: *** [upload] Error 309 at digispark attiny85

Kamilake, 2021-03-29:
[image]
RAM:   [          ]   1.8% (used 9 bytes from 512 bytes)
Flash: [=         ]   5.9% (used 352 bytes from 6012 bytes)
Configuring upload protocol...
AVAILABLE: micronucleus
CURRENT: upload_protocol = micronucleus
Uploading .pio\build\digispark-tiny\firmware.hex
*** [upload] Error 309

Whenever I hit the upload button to upload code, I get the following message: This is not a driver issue, the same project is programmed normally in different environments.
Two different environments have the same version of Windows and the same version of pio,code.
What makes this difference? I couldn't upload the code in any way.
[image]

The code I'm trying to upload is blink.

valeros, 2021-03-29:
https://community.platformio.org/t/how-to-program-attiny-85-with-platformio-vscode/18957/8 might contain possible solutions.

Kamilake, 2021-03-30:
> https://community.platformio.org/t/how-to-program-attiny-85-with-platformio-vscode/18957/8 might contain possible solutions.
Thanks for the quick advice. I will be testing Verbose mode soon. And I will give you the results.
[image]

Kamilake, 2021-03-30:
[image]

> https://community.platformio.org/t/how-to-program-attiny-85-with-platformio-vscode/18957/8 might contain possible solutions.
I still haven't got any information to fix the problem even in Verbose mode.. and Reinstalling the PIO didn't work for me.

`*** [upload] Error 309`

This is a problem before the USB driver.
There seems to be a problem with the micronucleus itself.
Please ask me for any additional information you need. I will answer immediately.
Thanks in advance.

Kamilake, 2021-04-01:
[image]

Just before, I saw a message on my console that seems to be downloading a library of something.
It soon disappeared and I couldn't find it on all tabs.(I saw links like adafruit etc.)
When I hit the upload button again everything worked fine.
My logs appear to have been deleted automatically, so I can't tell you how to fix it.
I will soon leave the unstable platform and return to the legacy Arduino IDE. (I've already found more than 10 bugs, and I have to be uncomfortable because I don't have the ability to fix them.)
Finally, please feel free to tell me if you need anything.

closed 2021-04-01 by Kamilake