platformio/platform-atmelavr #251: Building project with MegaCore no longer works

GunZi200, 2021-04-09:
I have tried with both ATmega2560 and ATmega2561.

This has never been a problem before.

[image]

GunZi200, 2021-04-09:
Here is my `.ini` file for one of my projects. It can be used to reproduce.

```ini
[env]
platform = atmelavr
framework = arduino
monitor_speed = 115200

board_build.f_cpu = 16000000L        ; 16MHz CPU frequency
board_hardware.uart = uart0          ; For Bootloader
board_hardware.oscillator = external ; External 16MHz crystal

upload_flags = -Pusb

build_flags =
    -Wunused-parameter ; Turn on compiler warning when functions have unused parameters

[env:megaatmega2561]
board = ATmega2561
board_build.mcu = atmega2561

[env:megaatmega2560]
board = ATmega2560
board_build.mcu = atmega2560
```

GunZi200, 2021-04-09:
I can reproduce the problem:

`platform = atmelavr@3.2.0` does not work

`platform = atmelavr@3.1.0` works

GunZi200, 2021-04-09:
If I revert line 78 here https://github.com/platformio/platform-atmelavr/commit/5077ecb2b1c1958fc469e24f2f5c00162924ec1b#diff-0184adedc292a93218525209d37dabe7330a013e6ff8fa46fd6328c5eb4598faR78 it works. Maybe something was missing when MegaCore was updated?

GunZi200, 2021-04-11:
@valeros @MCUdude I'm thinking that maybe its best to include MegaCore specific MCUs in the example.yml to catch the build errors earlier?

GunZi200, 2021-04-11:
Ok I looked more into the Github pipeline. There is one MegaCore MCU there and compiles fine. I installed Python 3.7.9 (pipeline uses 3.7.10) and I reinstalled PlatformIO... then the build worked fine. I had Python 3.9.1 before.

Now I've reinstalled 3.9.1 and I am getting this VS Code or SCons error:

[image]

But then when I try to reproduce this more by reinstalling PIO again without changing the Python installation I get no error. I'm so confused 😅 Will close this for now and reopen if I learn more.

GunZi200, 2021-04-19:
The issue appeared again on another machine. Adding `SPI` to `platformio.ini` does not work as show in the image below... 
[image]

valeros, 2021-04-19:
Have you tried to explicitly specify the `SPI` library as a dependency in `lib_deps`?

GunZi200, 2021-04-19:
> Have you tried to explicitly specify the `SPI` library as a dependency in `lib_deps`?

Yes I get the same results. You can see the Library Manager trying to install `SPI` in the image above. If I try to install SPI with a fresh platformio install I get a similar yellow text which says SPI library cannot be found for `windows-amd64`. Seems to install OK with the atmelavr framework though.

For reference I did it like this:

```ini
lib_deps =
     SPI
     ; ... more libraries below
```

valeros, 2021-04-19:
Could you please put together a simple project to reproduce the issue?

GunZi200, 2021-04-19:
@valeros here: https://github.com/GunZi200/MegaCore-ATmega2560-err

I get this result on my end. I created an Arduino Mega project and only changed the `platformio.ini` file. 

[image]

If you can't reproduce this try one thing I did:

1. Install Python 3.9.1 and add it to PATH
2. delete `.platformio` folder and open vscode.
3. try to build project

I seem to be able to bypass this error when my machine has Python 3.7.9 installed.

valeros, 2021-04-20:
It's strange, but I cannot reproduce the issue, tried your project with Python 3.9.1 and everything works just fine. Could you please run `pio system info` in your terminal?

GunZi200, 2021-04-20:
Result from two different computers:

Computer 1:
[image]

Will edit this comment on the other computer to add picture

valeros, 2021-04-20:
That's super weird. Let's focus on machines where it doesn't work. Do you see the `Wire` library somewhere here  `c:\Users\Gudni\.platformio\packages\framework-arduino-avr-megacore\libraries\`?

GunZi200, 2021-04-20:
Hmm the folder on the machines where it doesn't work just looks like this: 
[image]

valeros, 2021-04-20:
Apparently, the `framework-arduino-avr-megacore` package is broken for some reason. Does it always look like that even when you delete the `c:\Users\Gudni\.platformio` folder and try again?

GunZi200, 2021-04-20:
Hmm I get it to work sometimes but sometimes not. I found one way to trigger the issue when it was working.

I deleted the folder `framework-arduino-avr-megacore` then hit `Clean` and PIO will show me the progress of installing `framework-arduino-avr-megacore` again. I then hit `Build` and issue returns.

valeros, 2021-04-20:
Do you have an antivirus by any chance?

GunZi200, 2021-04-20:
No only Windows Defender, its never been a problem for me.

valeros, 2021-04-20:
Something is messing with PlatformIO packages on your machines and I'm guessing it's `Windows Defender`. Try to create [an exclusion](https://support.microsoft.com/en-us/windows/add-an-exclusion-to-windows-security-811816c0-4dfd-af4a-47e4-c301afe13b26) for `c:\Users\Gudni\.platformio` and then delete this folder to force PlatformIO redownload all packages.

GunZi200, 2021-04-20:
@valeros I noticed that PlatformIO plugins installs `framework-arduino-avr-megacore` folder when reinstalling the `.platformio` folder. I see not installation progress and it seems to occur while this issue takes places: platformio/platformio-vscode-ide#2506. If I restart VScode while this is happening, that seems to be enough to trigger the issue.

GunZi200, 2021-04-20:
Looking at the Console... could this be an issue?
[image]

MCUdude, 2021-04-20:
I'm not sure if it's relevant or not, but I was not able to get the example project @GunZi200 provided when platformio.ini looked like this. Identical error to this one: https://github.com/platformio/platform-atmelavr/issues/251#issuecomment-817303556

``` ini
[env:ATmega2560]
platform = atmelavr
board = ATmega2560
framework = arduino

; Libraries
lib_deps =
     Wire
     https://github.com/adafruit/RTClib ; Real-time clock library

lib_ignore = TinyWireM
```

However, if I changed the library dependent to this, it worked:

```ini
[env:ATmega2560]
platform = atmelavr
board = ATmega2560
framework = arduino

; Libraries
lib_deps =
     Wire
     adafruit/RTClib @ 1.13.0

lib_ignore = TinyWireM
```

valeros, 2021-04-20:
@GunZi200 I believe it has nothing to do with this issue, as your system contains broken framework package.

GunZi200, 2021-04-20:
I think I have a solution...

Reproduction steps to make it work:

1. Open VScode
2. Find the folder `.platformio\packages\framework-arduino-avr-megacore` and delete it
3. Keep the folder `.platformio\packages` open
4. Now reload VScode window `Ctrl + Shift + P` -> `Developer: Reload Window`
5. `PlatformIO: Rebuilding IntelliSense Index` starts.
6. Notice how the `framework-arduino-avr-megacore` starts with content:

[image]

7. Now wait for `PlatformIO: Rebuilding IntelliSense Index` to finish in VScode an notice how the remaining content in folder `framework-arduino-avr-megacore`  appears ~30 seconds later.

[image]

Build from this point works for me on all machines.

If I for some reason reload VScode in step 6 the issue appears and the package is broken. I can consistently reproduce this.

valeros, 2021-04-20:
@GunZi200 

> If I for some reason reload VScode in step 6 the issue appears and the package is broken. I can consistently reproduce this.

If you do that, you will kill the `platformio-core` process that's running in background, which can lead to unpredictable side effects. In your case, I believe, you interrupted the process when PlatformIO was unpacking the `framework-arduino-avr-megacore` package and leaving it in corrupted state.