platformio/platform-atmelavr #255: Change CPU frequency on attiny85 fails to compile in 3.3.0

5e3, 2021-04-24:
In 3.2.0 it was possible to set 
board_build.f_cpu = 16000000L

In 3.3.0 compiling fails with:
Compiling .pio\build\attiny85\FrameworkArduino\wiring_shift.c.o
C:\Users\xxx\.platformio\packages\framework-arduino-avr-attiny\cores\tiny\wiring.c: In function 'init':
C:\Users\xxx\.platformio\packages\framework-arduino-avr-attiny\cores\tiny\wiring.c:1095:8: error: #error "Frequency requested from internal oscillator that cannot be generated by prescaling"
       #error "Frequency requested from internal oscillator that cannot be generated by prescaling"
        ^~~~~
C:\Users\xxx\.platformio\packages\framework-arduino-avr-attiny\cores\tiny\wiring.c:1096:8: error: #error "Custom tuning is not supported in the current version of ATTinyCore"
       #error "Custom tuning is not supported in the current version of ATTinyCore"
        ^~~~~

The code I'm compiling depends on the 16MHz PLL clock.

valeros, 2021-04-26:
Hi @5e3 ! In the latest release the ATTinyCore was updated to the latest v1.5.2, so something may have changed in the internal implementation. Anyway, the error message seems quite descriptive. Does it work in the Arduino IDE?

5e3, 2021-04-26:
Hi @valeros 

Don't know - I didn't try. But if I set

build_flags = -DCLOCK_SOURCE=6

in platformio.ini it seems to resolve the issue...

r4wabr, 2021-08-14:
define the F_CPU: beetwine  #include <Arduino.h>
`#define F_CPU 16000000UL`

commit 012de336 (2021-08-25): Fix default clock source for several boards from ATTinyCore

Resolve #255
closed 2021-08-25 by valeros