platformio/platform-atmelavr #256: avr-gdb depends on libncurses.so.5 and libtinfo.so.5

neildarlow, 2021-05-01:
Hi,

I am running PlatformIO from VSCodium in flatpak format. There is an issue with avr-gdb requiring libncurses and libtinfo with major version number of 5 as follows:

neil@satellite:~$ ldd .platformio/packages/toolchain-atmelavr/bin/avr-gdb
linux-vdso.so.1 (0x00007ffd5605a000)
libncurses.so.5 => not found
libtinfo.so.5 => not found
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f8f18297000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f8f180a5000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f8f1809f000)
/lib64/ld-linux-x86-64.so.2 (0x00007f8f183ff000)

These libraries are marked legacy or deprecated in current Linux distributions and I see that the flatpak runtime environment  includes them at major version 6 also;

[📦 com.vscodium.codium ~]$ find /usr/lib -name 'libncurses*'
/usr/lib/x86_64-linux-gnu/libncurses.so
/usr/lib/x86_64-linux-gnu/libncurses.so.6
/usr/lib/x86_64-linux-gnu/libncurses.so.6.2
/usr/lib/x86_64-linux-gnu/libncursesw.so
/usr/lib/x86_64-linux-gnu/libncursesw.so.6
/usr/lib/x86_64-linux-gnu/libncursesw.so.6.2

[📦 com.vscodium.codium ~]$ find /usr/lib -name 'libtinfo*'
/usr/lib/x86_64-linux-gnu/libtinfo.so
/usr/lib/x86_64-linux-gnu/libtinfo.so.6
/usr/lib/x86_64-linux-gnu/libtinfo.so.6.2
/usr/lib/x86_64-linux-gnu/libtinfow.so
/usr/lib/x86_64-linux-gnu/libtinfow.so.6
/usr/lib/x86_64-linux-gnu/libtinfow.so.6.2

Could avr-gdb be updated to link against libncurses.so.6 and libtinfo.so.6 please.

Regards,
Neil Darlow

valeros, 2021-05-03:
Hi @neildarlow ! The GDB binary is part of the toolchain, so it's not possible to update it at the moment. I'm not sure whether it will help, but you can try this package:
```
platform_packages =
   toolchain-atmelavr @ ~2.70300.0
```

closed 2021-05-03 by valeros

neildarlow, 2021-05-03:
Hi @valeros,

Thank you but that caused more issues. For now I've switched from VSCodium in flatpak to AppImage and installed those legacy libraries.

Not an ideal solution but it works.

ATB,
Neil Darlow

Saganesque, 2022-05-19:
I am running into the same issue using PlatformIO on VSCode