platformio/platform-atmelavr #263: Suppport for ATTINY1624

farfarfromhome, 2021-09-24:
Please add support for ATTINY1624.

cutephoton, 2021-10-17:
Seconded. Support for Tiny-0 and Tiny-1 would be really helpful.

zebra1993, 2021-11-12:
What files or Infos would be needed to get the Tiny1624 added?
Is there a guide/list somewhere to start on the requirements?

marco-117, 2022-01-14:
I need this too, and had some email contact to the platformIO team. It seems taht they have currently to much projects to work on and no time to support this chip.

Only solution so far is to use the ARduino IDE and the megaTinyCore for the compilation and programming.
The supplier is also emailed, perhaps they can provide something.

And i posted it in the [https://community.platformio.org/t/attiny-two-hw-uart-integration-attiny-1624/25423](https://community.platformio.org/t/attiny-two-hw-uart-integration-attiny-1624/25423)

br74649, 2022-01-14:
any news on 1624 support?

maxgerhardt, 2022-01-14:
1. This issue is in the completely wrong repository, the ATTiny1624 is one of the newer chips with UPDI support, for which https://github.com/platformio/platform-atmelmegaavr/ is responsible
2. I've added support in a minimal way in the project https://github.com/maxgerhardt/pio-attiny1624-test. Can someone please test that on real hardware? (Upload is the PlatformIO default, avrdude + jtag2updi, see https://docs.platformio.org/en/latest/platforms/atmelmegaavr.html#upload-using-programmer if your uploader differs)

marco-117, 2022-01-14:
Thank you very much.
I will test it and give you response

farfarfromhome, 2022-01-14:
Max

I have hardware to test this on.  I previously was able to get the ATtiny1624 running with both serial ports.  I reconfirmed that it is still running before testing your changes.  I then swapped out my platformio.ini with yours, added my pymcuprog upload_commands and was able to successfully compile and upload.  Unfortunately there is no output from either serial port.  I am a novice at this but please let me know what I can do to help?  I include my platformio.ini file if it helps in any way.

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ATtiny1624]
platform = atmelmegaavr
framework = arduino
board = ATtiny1624

build_flags = 
	-std=c++17
	-D ATtiny1624

lib_deps = 
	jfturcot/SimpleTimer@0.0.0-alpha+sha.b30890b8f7

board_build.f_cpu = 16000000L  ; used for code compile but doesn't link to upload!
monitor_speed = 115200
monitor_port = COM4
upload_speed = 115200
upload_port = COM5

upload_command =
	- pymcuprog erase -t uart -u $UPLOAD_PORT -c $UPLOAD_SPEED -d attiny1624
	- pymcuprog write -t uart -u $UPLOAD_PORT -c $UPLOAD_SPEED -d attiny1624 -m fuses -o 1 -l 0xC4	; BOD 4V & Enabled 
	- pymcuprog write -t uart -u $UPLOAD_PORT -c $UPLOAD_SPEED -d attiny1624 -m fuses -o 2 -l 1		; l = 1 - 16MHz , l = 2 - 20Mhz
	- pymcuprog write -t uart -u $UPLOAD_PORT -c $UPLOAD_SPEED -d attiny1624 -f .pio\build\ATtiny1624\firmware.hex --verify

`

maxgerhardt, 2022-01-14:
In the Arduino IDE "Tools" menu, what settings for clock source have you selected?

farfarfromhome, 2022-01-14:
OK maybe I see it.  I am compiling to 20MHz and programming to 16MHz!!  Let me try again.

farfarfromhome, 2022-01-14:
No, still not working.  Let me find the file, where does it reside.  I worked on this several months ago and my recollection is very hazy.

maxgerhardt, 2022-01-14:
It should be in your project's `boards/` folder or in the `C:\Users\<user>\.platformio\platforms\atmelavr\boards`.

farfarfromhome, 2022-01-14:
`{
  "build": {
    "core": "megatinycore",
    "extra_flags": "-DARDUINO_attinyxy4 -DMILLIS_USE_TIMERB1 -DUARTBAUD5V",
    "f_cpu": "16000000L",
    "mcu": "attiny1624",
    "variant": "txy4"
  },
  "hardware": {
    "oscillator": "internal"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "ATtiny1624",
  "upload": {
    "maximum_ram_size": 2048,
    "maximum_size": 16384,
    "protocol": "jtag2updi",
    "speed": 115200
  },
  "url": "https://www.microchip.com/wwwproducts/en/ATTINY1614",
  "vendor": "Microchip"
}
`

farfarfromhome, 2022-01-14:
I note is says 16MHz.  So I changed
 board_build.f_cpu = 16000000L
 and 
pymcuprog write -t uart -u $UPLOAD_PORT -c $UPLOAD_SPEED -d attiny1624 -m fuses -o 2 -l 1		; l = 1 - 16MHz , l = 2 - 20Mhz
and still no joy.

maxgerhardt, 2022-01-14:
Can you upload your whole project and the used board file verbatim? Also you did not update the megatinycore framework right?

farfarfromhome, 2022-01-14:
I went back to my ini configuration and now I am getting a compile error about the DACREF0.  I now recollect this was one of the first problems that I had to resolve. 

I haven't updated the framework, the only changes came from the
platform_packages = 
   toolchain-atmelavr@https://github.com/maxgerhardt/pio-newer-atmel-toolchains/raw/main/toolchain-atmelavr-windows-1.70300.200526.tar.gz
   framework-arduino-megaavr-megatinycore@https://github.com/maxgerhardt/megaTinyCore.git

Sorry but where/how to upload?

farfarfromhome, 2022-01-14:
OK the project is there.  Where do I find the used board file?

maxgerhardt, 2022-01-14:
Have you looked at https://github.com/platformio/platform-atmelavr/issues/263#issuecomment-1013472782?

farfarfromhome, 2022-01-14:
Not sure what you mean by looking at #263.  I originally made the request for 1624 support.

maxgerhardt, 2022-01-14:
> It should be in your project's boards/ folder or in the C:\Users\<user>\.platformio\platforms\atmelmegaavr\boards.

farfarfromhome, 2022-01-14:
Sorry the json file.  Its now uploaded.

br74649, 2022-01-14:
hi max

I just compiled your test project and uploaded to a 1624

ive been toying with the ftdi upload rather than jtag2udpi and my upload command is not very elegant but I confirm that when im getting serial outputs on both uarts

my platform.ini is currently this!

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:attiny1624]
platform = atmelmegaavr
board = attiny1624
framework = arduino
; use updated toolchain and core
; link to toolchain is OS specific. default Windows.
; see https://github.com/maxgerhardt/pio-newer-atmel-toolchains.
; for Linux, use https://github.com/maxgerhardt/pio-newer-atmel-toolchains/raw/main/toolchain-atmelavr-linux_x86_64-1.70300.200526.tar.gz
platform_packages = 
   toolchain-atmelavr@https://github.com/maxgerhardt/pio-newer-atmel-toolchains/raw/main/toolchain-atmelavr-windows-1.70300.200526.tar.gz
   framework-arduino-megaavr-megatinycore@https://github.com/maxgerhardt/megaTinyCore.git

; set clock to 20Mhz
board_build.f_cpu = 16000000L
; remove old GNU+11 standard
build_unflags = -std=gnu++11
; set GNU++17 standard and add in configuration macros
build_flags = 
   -std=gnu++17
   -flto
   -mrelax 
   -DCLOCK_SOURCE=0
   -DTWI_MORS
   -DCORE_ATTACH_ALL
   -DMEGATINYCORE_MAJOR=2UL
   -DMEGATINYCORE_MINOR=5UL
   -DMEGATINYCORE_PATCH=6UL
   -DMEGATINYCORE_RELEASED=1
upload_protocol = custom 
upload_command = C:\Users\tony_\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Users\tony_\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py -t uart -u COM18 -b 230400 -d attiny1624 --fuses 2:0x01 6:0x04 8:0x00 -f$SOURCE -a write

maxgerhardt, 2022-01-14:
@farfarfromhome  I can't your project to compile with default settings. Once I add my override for the toolchain I get a successful compilation. (`
platform_packages = toolchain-atmelavr@https://github.com/maxgerhardt/pio-newer-atmel-toolchains/raw/main/toolchain-atmelavr-windows-1.70300.200526.tar.gz`). Further I see that in your `platformio.ini.new` you set `monitor_speed = 115200` which is wrong in regards to my source code, it does output on 9600 baud. 

@br74649 Thanks for confirming it's working for you!

br74649, 2022-01-14:
you sir are a star!!!!!!

i have loads of 1624's as i cant get hold of 1614's anymore and was porting my stuff back to arduino ide, you saved my weekend!

yes i stole the commandline from the arduino ide and those fuses are different between the 20mhz and 16mhz

farfarfromhome, 2022-01-14:
Max

Ok I changed your baud rate to 115200 and now everything is working.  I was using a separate terminal program and it was set to 115200!

I recollect to originally get this working I changed a pin definition file.  It will be good to use the official files now so that I don't have to worry about software updates breaking my compile!

Hopefully I didn't waste too much of your time on this.

Thanks

br74649, 2022-01-17:
Further testing on the tiny1624 I was using attachinterrupt to measure a pwm signal and the chip just reboots all the time when the interrupt fires, not sure if it's timers or what right now, the code works fine using the arduino ide so possibly some further challenges to resolve...

closed 2022-01-18 by valeros