platformio/platform-atmelavr #268: Add support for tinyAVR 2 Series MCU like attiny824

woodymgh, 2021-11-26:
Please add support for the Attiny Boards like:
attiny424
attiny426
attiny427
attiny824
attiny826
attiny827
attiny1624
attiny1626
attiny1627
attiny3224
attiny3226
attiny3227

This are upcoming great processors with many Hardware support like Timers, UARTS, 12Bit ADC, TWI, SPI and low Cost.
https://www.microchip.com/en-us/product/ATtiny824

closed 2022-01-18 by valeros

grst, 2022-01-25:
When I'm trying to compile for Attiny424 I am getting the following error. 
Is there still something missing or am I doing something wrong? 

```
PACKAGES: 
 - framework-arduino-megaavr-megatinycore 2.5.8 
 - toolchain-atmelavr 3.70300.0 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/ATtiny424/src/main.c.o
avr-gcc: error: device-specs/specs-attiny424: No such file or directory
Compiling .pio/build/ATtiny424/FrameworkArduino/Tone.cpp.o
avr-g++: error: device-specs/specs-attiny424: No such file or directory
Compiling .pio/build/ATtiny424/FrameworkArduino/UART.cpp.o
Compiling .pio/build/ATtiny424/FrameworkArduino/UART0.cpp.o
avr-g++: error: device-specs/specs-attiny424: No such file or directory
avr-g++: error: device-specs/specs-attiny424: No such file or directory
Compiling .pio/build/ATtiny424/FrameworkArduino/UART1.cpp.o
avr-g++: error: device-specs/specs-attiny424: No such file or directory
Compiling .pio/build/ATtiny424/FrameworkArduino/WInterrupts.c.o
Compiling .pio/build/ATtiny424/FrameworkArduino/WInterrupts_PA.c.o
avr-gcc: error: device-specs/specs-attiny424: No such file or directory
*** [.pio/build/ATtiny424/src/main.c.o] Error 1
Compiling .pio/build/ATtiny424/FrameworkArduino/WInterrupts_PB.c.o
avr-gcc: error: device-specs/specs-attiny424: No such file or directory
*** [.pio/build/ATtiny424/FrameworkArduino/Tone.cpp.o] Error 1
*** [.pio/build/ATtiny424/FrameworkArduino/UART.cpp.o] Error 1
*** [.pio/build/ATtiny424/FrameworkArduino/UART0.cpp.o] Error 1
*** [.pio/build/ATtiny424/FrameworkArduino/UART1.cpp.o] Error 1
*** [.pio/build/ATtiny424/FrameworkArduino/WInterrupts.c.o] Error 1
*** [.pio/build/ATtiny424/FrameworkArduino/WInterrupts_PA.c.o] Error 1
avr-gcc: error: device-specs/specs-attiny424: No such file or directory
*** [.pio/build/ATtiny424/FrameworkArduino/WInterrupts_PB.c.o] Error 1
```

my `platformio.ini` looks like this, and I removed the `.platformio/{.cache,packages}` directories before building: 
```ini
[env:ATtiny424]
platform = https://github.com/platformio/platform-atmelmegaavr.git
; platform = atmelmegaavr
board = ATtiny424
framework = arduino
```

woodymgh, 2022-01-25:
Hello,

the same issue here for attiny824 without arduino framework:

Processing ATtiny824 (platform: https://github.com/platformio/platform-atmelmegaavr.git; board: ATtiny824)
-------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATtiny824.html
PLATFORM: Atmel megaAVR (1.5.0+sha.de9877b) > ATtiny824
HARDWARE: ATTINY824 16MHz, 1KB RAM, 8KB Flash
PACKAGES: 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/ATtiny824/src/main.o
avr-gcc: error: device-specs/specs-attiny824: No such file or directory
*** [.pio/build/ATtiny824/src/main.o] Error 1
======================================== [FAILED] Took 0.91 seconds ========================================
The terminal process "platformio 'run'" terminated with exit code: 1.

is this the wrong toolchain ?

platformio.ini:

[env:ATtiny824]
; platform = atmelavr
platform = https://github.com/platformio/platform-atmelmegaavr.git
; board = attiny45
board = ATtiny824
; framework = arduino

woodymgh, 2022-01-26:
Hello,

i got it's working (to compile, i can't test if its wright)

i have changed the platformio.ini to:

[env:ATtiny824]
; platform = atmelavr
platform = https://github.com/platformio/platform-atmelmegaavr.git
; board = attiny45
board = ATtiny824
; framework = arduino

BUILD_FLAGS =
     -Battiny_dfp/gcc/dev/attiny824
     -Iattiny_dfp/include/

extra_scripts = extra_script.py

in the dir where platformio.ini is located i create extra_script.py with 
this content:

Import("env")

#
# Dump build environment (for debug)
# print(env.Dump())
#

env.Append(
   LINKFLAGS=[
       "-Battiny_dfp/gcc/dev/attiny824"
   ]
)

i go to https://packs.download.microchip.com/ and download the pack for  
Microchip ATtiny Series Device Support

i rename the extension from ".atpack" to ".zip"
now i unzip the pack.
i rename the dir with the pack to attiny_dfp
i move this dir (attiny_dfp) to the dir where platformio.ini is located

now i can compile

Processing ATtiny824 (platform: 
https://github.com/platformio/platform-atmelmegaavr.git; board: 
ATtiny824)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose 
mode can be enabled via `-v, --verbose` option
CONFIGURATION: 
https://docs.platformio.org/page/boards/atmelmegaavr/ATtiny824.html
PLATFORM: Atmel megaAVR (1.5.0+sha.de9877b) > ATtiny824
HARDWARE: ATTINY824 16MHz, 1KB RAM, 8KB Flash
PACKAGES:
  - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\ATtiny824\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project 
Inspect"
RAM:   [          ]   0.4% (used 4 bytes from 1024 bytes)
Flash: [          ]   3.1% (used 254 bytes from 8192 bytes)
======================================================================= 
[SUCCESS] Took 1.23 seconds 
=======================================================================

Series MCU like attiny824 (Issue #268)

woodymgh, 2022-01-26:
I have forgot:

i have changed the ldscript (because it's wrong)
in the file .platformio/packages/toolchain-atmelavr(version what you 
use)/avr/lib/ldscripts/avrxmega3.xn

replace
    data   (rw!x) : ORIGIN = 0x802000, LENGTH = __DATA_REGION_LENGTH__

with
   data   (rw!x) : ORIGIN = __DATA_REGION_ORIGIN__, LENGTH = 
__DATA_REGION_LENGTH__

***@***.***>