platformio/platform-atmelavr #269: KeyError: 'tool-avrdude' : fuses.py cannot find tool-avrdude

Madour, 2021-12-10:
Hello, I am trying to burn a bootloader to my DigiSpark board (ATTINY85).

When running the following command : `platformio.exe run --target bootloader --environment digispark-tiny`, I get a `KeyError : 'tool-avrdude'`

The error comes from _packages.py file the `self.packages`member dictionary does not contain the key "tool-avrdude': 

```python
def get_package_spec(self, name, version=None):
        return PackageSpec(
            owner=self.packages[name].get("owner"),
            name=name,
            requirements=version or self.packages[name].get("version"),
        )
```

Which is called in fuse.py line 479:

```python
        % join(env.PioPlatform().get_package_dir("tool-avrdude") or "", "avrdude.conf"),
```

Any help is wellcome thanks

MrSurly, 2023-03-25:
@Madour: I just came across this problem myself (#306) -- did you ever find a solution?

Madour, 2023-03-25:
Unfortunately not. In the end, I gave up as it was not very crucial for my project.

Did you fix the issue, or fond a workaround?

MrSurly, 2023-03-25:
`pio pkg install -t tool-avrdude` solved it for me. Evidently atmelmegaavr hard-codes this value somewhere, and if you're using a chip that _doesn't_ required avrdude, it errors out.  Installing the tool fixes that error.

commit e3d80f27 (2023-03-27): Install avrdude package when bootloader target is selected

Resolves #269
closed 2023-03-27 by valeros