platformio/platform-atmelavr #277: Upload to ATTiny167 fails

hcccs, 2022-03-06:
Uploading to an ATTiny167 fails with "Could not find USBtiny device (0x1781/0xc9f)". It doesn't prompt for a device connect.

`  
`Processing attiny167 (platform: atmelavr; board: attiny167; framework: arduino)
-----------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny167.html
PLATFORM: Atmel AVR (3.4.0) > Generic ATtiny167
HARDWARE: ATTINY167 8MHz, 512B RAM, 16KB Flash
PACKAGES:
 - framework-arduino-avr-attiny 1.5.2
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit SSD1306> 2.5.1
|   |-- <Adafruit GFX Library> 1.10.13
|   |   |-- <Adafruit BusIO> 1.11.2
|   |   |   |-- <Wire> 2.0.0
|   |   |   |-- <SPI> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|-- <Adafruit GFX Library> 1.10.13
|   |-- <Adafruit BusIO> 1.11.2
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|-- <Adafruit BusIO> 1.11.2
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <OneWire> 2.3.6
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.6
Building in release mode
Checking size .pio\build\attiny167\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  49.8% (used 255 bytes from 512 bytes)
Flash: [========  ]  84.8% (used 13886 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: usbtiny
CURRENT: upload_protocol = usbtiny
Looking for upload port...
Uploading .pio\build\attiny167\firmware.hex
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.

*** [upload] Error 1
=============================== [FAILED] Took 5.75 seconds ======`
`

hcccs, 2022-03-06:
[SOLVED
I added "upload_protocol = micronucleus" to platformio.ini and that worked.

Fusseldieb, 2023-02-14:
> upload_protocol = micronucleus

This fixed the upload issue on my ATtiny85.