platformio/platform-atmelavr #283: Platformio Arduino debug in CLion with simavr error "No module named gdb"

rubencart, 2022-05-22:
When I try to debug the following main.cpp in CLion with the platformio.ini listed below, I get the error below, and the debugger stops.
main.cpp:
```
#include <Arduino.h>

static int solenoidPin = 9;

void setup() {
    // write your initialization code here
    Serial.begin(115200);

    Serial.print("Setup");
    pinMode(solenoidPin, OUTPUT);          //Sets that pin as an output
}

void loop() {
    // write your code here
    Serial.print("Loop");
    digitalWrite(solenoidPin, HIGH);      //Switch Solenoid ON
    delay(1000);                          //Wait 1 Second
    digitalWrite(solenoidPin, LOW);       //Switch Solenoid OFF
    delay(1000);                          //Wait 1 Second
}
```
platformio.ini:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
debug_tool = simavr
```

Error:
```
Failed to process MI record:
>
 No module named gdb:
Reading symbols from /Users/rubencartuyvels/Documents/audiokunst/arduino/arduino-pianist/piano-fsm/.pio/build/uno/firmware.elf...done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = simavr
PlatformIO: Initializing remote target...
0x00000000 in __vectors ()
PlatformIO: Initialization completed
```
Any ideas on how to solve this? 
I am on macOS 10.15. As suggested [here](https://community.platformio.org/t/error-no-module-named-gdb-debug-start-problem/25763), I tried running `brew install avr-gdb` after having installed platformio and simavr, but this does not change anything.

dlschmidt, 2023-01-04:
I can confirm that I get similar errors without defining the debug_tool (using the default parameter = unified debugger).

No matter if I try to use the debugger under macOS ARM or Ubuntu ARM, I will get a similar error.
https://community.platformio.org/t/platformio-failes-to-start-debugger-on-arm/31272/2
https://github.com/platformio/platformio-core/issues/4506

Maybe we could try to configure PlatformIO to use own avr-gdb packages instead of preinstalled binaries?

dlschmidt, 2023-01-05:
@rubencart You can try to compile gcc, gdb and binutils yourself. Maybe start with an ubuntu VM, which should be easier.
Probably you should not compile the latest version of all tools because then gdb will fail because some functions are not implemented yet.
But I think this solution is a bit inappropriate because you want to use CLion instead of VSCode, or does CLion have an alternative to Remote SSH?