platformio/platform-atmelavr #292: Missing python 2.7 installation on MacOS Monterey 12.6 when trying to start debug session

Metamogul, 2022-10-23:
**OS**: Macos 12.6
**PlatformIO Core Version**: 6.1.4

**Actions:**
Starting a debug session from PlatformIO project tasks for a MightyCore project.

**Expected behaviour:**
The program is compiled, transmitted and the debugger connects to the session, halting at the first breakpoint.

**What happens instead:**
When trying to start the debug session , the debugger won't start because of a missing python library that has been removed from MacOS. This is the full output from the Debug Console tab:

    undefineddyld[84361]: Library not loaded: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python'
      Referenced from: '/Users/jan/.platformio/packages/toolchain-atmelavr/bin/avr-gdb'
      Reason: tried: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file), 
    '/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file)

The program is built and transfered to the developer board, but the debugger doesn't become active.

**Fix:**
I haven't found a fix yet. The simplest option seems to be to use a different version of avr-gdb that's not linked against this specific path to a python installation.

Metamogul, 2022-10-23:
When switching to a newer version of toolchain-atmelavr by setting `platform_packages = toolchain-atmelavr@3.70300.220127` in `platformio.ini`, I'm greated with the following error in the debugging console:

```
Unpacking
undefinedError: Traceback (most recent call last):
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/debug/cli.py", line 84, in cli
    return _debug_in_project_dir(
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/debug/cli.py", line 193, in _debug_in_project_dir
    loop.run_until_complete(coro)
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/debug/process/gdb.py", line 58, in run
    await self.spawn(*args, cwd=self.project_dir, wait_until_exit=True)
  File "/Users/jan/.platformio/penv/lib/python3.9/site-packages/platformio/debug/process/base.py", line 77, in spawn
    await loop.subprocess_exec(
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 1676, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/opt/homebrew/Cellar/python@3.9/3.9.15/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'avr-gdb'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Metamogul, 2022-10-23:
The reason is simple: Neither is there a `avr-gdb` binary included in `/Users/<user>/.platformio/packages/toolchain-atmelavr/bin` when installing the newer 3.70300.220127 of the atmelavr toolchain, nor does the script even seem to be looking there. I've created a temporary workaround by installing avr-gdb from `https://github.com/osx-cross/homebrew-avr/` (after fixing a bug; really doesn't seem to be my day). Nevertheless, this should be fixed and the toolchain package should be updated to contain the binary and to use it. Let me know what you think!

hraftery, 2022-10-31:
Same result (both the initial Python 2.7 error with the default toolchain-atmelavr@1.70300.191015, and the missing `avr-gdb` after updated to toolchain-atmelavr@3.70300.220127) for me with a fresh install of PlatformIO on:
- MacBook M1 Pro running macOS Monterey 12.6, and
- x86 iMac running macOS Big Sur 11.7

So I ran:

```
brew install brew tap osx-cross/avr
brew tap osx-cross/avr # currently installs 9.4.0, takes >10 minutes
brew install avr-gdb # eh, but avr-gdb is not included. This seems to do the trick
cd ~/.platformio/packages/toolchain-atmelavr/bin/
mv avr-gdb avr-gdb.orig
ln -s `which avr-gdb` avr-gdb
```

Which solves that issue, but alas, gets stuck on the next one:

```
Reading symbols from .../.pio/build/uno/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = avr-stub
PlatformIO: Initializing remote target...
.pioinit:13: Error in sourced command file:
:3333: Operation timed out.
```

hraftery, 2022-10-31:
Eh, my last error was because I needed `debug_port = /dev/cu.usbmodem21301` in `platformio.ini`, despite the documentation saying it is detected automatically. With that set, debugging "just works"! Boy, but if it doesn't, holy moly it's murky.

VitaliyGaliy, 2022-11-06:
after setting debug_port = /dev/cu.usbserial-1420
I've got - 
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = avr-stub
PlatformIO: Initializing remote target...
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
.pioinit:13: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout

=======================
my platformio.ini config is

[env:uno]
platform = atmelavr
board = uno
framework = arduino

debug_tool = avr-stub
debug_port = /dev/cu.usbserial-1420

hraftery, 2022-11-14:
@VitaliyGaliy having seen the error you describe in practice now, I'm confident it's not related to this issue. Instead, one trigger is not calling `debug_init()` during `setup()` as described [here](https://docs.platformio.org/en/latest/plus/debug-tools/avr-stub.html#configuration). If you still can't resolve it, I'd suggest searching for the error text, or opening a new issue.

markko1, 2022-12-26:
@VitaliyGaliy indeed, I had the same issue that I was getting a timeout on /dev/cu.usbserial-1420 on my M1 Mac. It turned out that I needed proper drivers for my USB-to-Serial adapter.
For me this helped: https://github.com/WCHSoftGroup/ch34xser_macos
It's somewhat daunting to download and install something from a website almost fully in Chinese (I think), but it fixed my issue. After installing the port needed to be changed to /dev/cu.wchusbserial410

dlschmidt, 2023-01-04:
Unfortunately my USBasp adapter has no serial port and setting debug_port would not work. 

I will try later if debugging works with my hardware under linux.

hraftery, 2024-11-21:
FYI, 2 years later I made a fresh start... and found this is all still required, verbatim.

Collated here for posterity.

`platformio.ini` must have as minimum:

```
[env:uno]
platform = atmelavr
board = uno
framework = arduino

platform_packages = 
	toolchain-atmelavr@~3.70300.220127

debug_tool = avr-stub
debug_port = /dev/cu.usbmodemXXXXXXX

lib_deps = 
	jdolinay/avr-debugger @ ~1.5
```

And then install `avr-gdb`:

```
brew tap osx-cross/avr
brew install avr-gcc # currently installs 9.4.0, takes >10 minutes
brew install avr-gdb # eh, but avr-gdb is not included. This seems to do the trick
cd ~/.platformio/packages/toolchain-atmelavr/bin/
mv avr-gdb avr-gdb.orig
ln -s `which avr-gdb` avr-gdb
```